stuartpittaway/diyBMSv4ESP32 #3: Add Russian language

sdqf-20, 2021-02-28:
Hello from Russia! I ordered circuit boards to repeat the project. I can translate the web interface and instructions into Russian, but I don’t see where the language data is stored. Sorry for my English if there are any errors.

Best Regards 
Evgeny

stuartpittaway, 2021-02-28:
Hi Martin, did you make any progress with this?  I may want to change the language settings into run-time Javascript rather than at compile time.  I can probably copy + paste any progress that you have made so far.

stuartpittaway, 2021-07-06:
I've now generated lots of language files, but not Russian.

Happy to add, if you/anyone wants to take on the challenge

sdqf-20, 2021-07-06:
Please show me where to find the files to edit. Ready to add Russian

stuartpittaway, 2021-07-06:
Here is the file in a new branch.

https://github.com/stuartpittaway/diyBMSv4ESP32/blob/RussianLangSupport/ESPController/web_src/lang_ru.js

At the moment, all the words are in English, so just need to edit that file to add the translations.

stuartpittaway, 2021-08-02:
Hello @sdqf-20 did you try and edit the Russian language files?

closed 2021-10-05 by stuartpittaway