stuartpittaway/diyBMSv4ESP32 #6: TFT: Add option to disable touch

atanisoft, 2021-03-09:
Using a TFT that does not have touch capability results in the TOUCH event being fired constantly. It would be good to have an option to disable this behavior.

stuartpittaway, 2021-03-09:
Ah, I'm not thinking of leaving the screen on all the time, so I was using touch to "wake" up the screen for a few minutes.

atanisoft, 2021-03-09:
Perhaps have a timer to wake up in the case of touch events not working?

stuartpittaway, 2021-03-09:
> Perhaps have a timer to wake up in the case of touch events not working?

I could use one of the I/O pins to wake it as well, so a simply push button would work.

atanisoft, 2021-03-09:
> Can you try the code from the branch - https://github.com/stuartpittaway/diyBMSv4ESP32/tree/get-tft-to-do-something

Working on it, will post in a short bit.

> I could use one of the I/O pins to wake it as well, so a simply push button would work. Could be a use for J13

That's also a great option, where is J13?

stuartpittaway, 2021-03-09:
> That's also a great option, where is J13?

Buried under the TFT screen, bottom left near the pin header

atanisoft, 2021-03-09:
The "get-tft-to-do-something" branch looks pretty good on the controller. I still have the loopback from the test app connected on RX1/TX1 and the console shows failed to send (likely due to no response from modules). I'm not worried about that just yet though.

The display looks like it went to sleep after a few minutes of inactivity but before it was displaying a large "button" for "configure access point".

> Buried under the TFT screen, bottom left near the pin header

Looks like a great candidate for another 90deg 2pin connector with the wires going out to the right or up to the top between the relays.

stuartpittaway, 2021-03-09:
Whilst I've got you.....

I'm working on the AVR programming code (it works!) but in that I want to disable tasks which may trigger whilst programming is in operation and affect the SPI bus, like SD card logging or touch.

I'm using a MUTEX on the SPI bus, which is fine and blocks the read/write, but I would prefer the task to be suspended so it didn't even try.

I suspended the tasks using 
```
vTaskSuspend(updatetftdisplay_task_handle);
```

but this still appears to trigger when a notification (from xTaskNotify) is triggered.

Any idea why or how I can stop the task completely, so it also ignores notifications?

stuartpittaway, 2021-03-23:
This is implemented on pin 1 of J13, wiring this to a push button switch wakes up the display - for when people are using cases or non-touch TFT display

Instructions - https://www.patreon.com/posts/49099220

closed 2021-03-23 by stuartpittaway