stuartpittaway/diyBMSv4ESP32 #11: ESP32 reboots intermitted when screen is touched

AlbertWeterings, 2021-04-13:
When the controller is running for 8 to 9 hours without being touched the ESP reboots upon touch of the screen.
If there is a way to write logging to SD card I'm happy to try and log this behavior.

stuartpittaway, 2021-04-14:
Hi Albert, thanks for the feedback.

I've a test controller on my bench which has been running this version https://github.com/stuartpittaway/diyBMSv4ESP32/commit/0601d004fcb29a2b32b0082fb274dfd0b0648eb4 for a number of weeks now without reboot, so its likely its a problem introduced in the newer version of code.

I suspect its a "stack canary" error - when one of the internal tasks runs out of stack space.  Unfortunately without capturing the debug logs (via serial/usb console) I can't see which task causes this problem.

AlbertWeterings, 2021-04-14:
The about page looks ok, I will keep an eye on it and will try to capture logging of it, can only do this when at home, as this is intermittent it might take some time to capture log data

AlbertWeterings, 2021-04-16:
[2021-04-16.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/6326256/2021-04-16.log)

It took me almost a lifetime to log the error. Look at line 348 and up.

stuartpittaway, 2021-04-19:
Thanks for the report Albert, unfortunately I'm not very well at the moment so may take a few days before I get chance to take a look

AlbertWeterings, 2021-04-20:
Just take your time. Get well soon

goferthat, 2021-05-12:
I'm seeing this behaviour too ... I'd assumed it was an intermittent connection (bad solder joint maybe) but was unable to isolate it. I'll see if I can snag a debug of it too.

stuartpittaway, 2021-05-13:
Hello, 

Thanks for the log file.

The key part is this..
```
[D][main.cpp:626] tca6408_isr_task(Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
```

This function "void tca6408_isr_task(void *param)" is reading from i2c the state of the TCA6408 chip.

I think this may be a mutex issue and the timeout stems from that.

@goferthat it would be great if you can get a debug dump similar to the one below so we can see if its the same issue.

I've got a test BMS running for over 30 days without a reboot now, and every day I jab the screen to wake it up - so its strange how we get these different problems.

Likely we can tweak some settings or add some additional error checking to fix this problem.

```
[D][tft.cpp:53] TFTScreenTouchInterrupt(): Ignored touch, screen already on
[D][tft.cpp:53] TFTScreenTouchInterrupt(): Ignored touch, screen already on
[D][main.cpp:626] tca6408_isr_task(Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40095958  PS      : 0x00060634  A0      : 0x8009477f  A1      : 0x3ffc04f0
A2      : 0x3ffc1d3c  A3      : 0x3ffb5308  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060623  A7      : 0x00000000  A8      : 0x3ffb5308  A9      : 0x3ffb5308
A10     : 0x00000003  A11     : 0x00000003  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060621  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa
Core 1 was running in ISR context:
EPC1    : 0x4000beb4  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40095958

ELF file SHA256: 0000000000000000

Backtrace: 0x40095958:0x3ffc04f0 0x4009477c:0x3ffc0510 0x40092ab3:0x3ffc0530 0x400fa3cd:0x3ffc0570 0x40081212:0x3ffc05d0 0x40081281:0x3ffc05f0 0x4008dcc9:0x3ffc0610 0x40009217:0x3ffb50e0 0x40007d13:0x3ffb5100 0x40007c69:0x3ffb5120 0x40008106:0x3ffb5140 0x400fa3d7:0x3ffb51d0 0x400db4c5:0x3ffb5230 0x40092c76:0x3ffb5250

Core 0 register dump:
PC      : 0x40093c62  PS      : 0x00060034  A0      : 0x80094445  A1      : 0x3ffbff90
A2      : 0x3ffbf5e8  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffc5970
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000001
A14     : 0x0000002a  A15     : 0x3ffd9dd0  SAR     : 0x00000019  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x40093c62:0x3ffbff90 0x40094442:0x3ffbffc0 0x40092d63:0x3ffbffe0 0x400957d5:0x3ffc0000 0x4008dcd2:0x3ffc0010 0x4018fa8b:0x3ffbc960 0x4011c3a2:0x3ffbc980 0x4009440d:0x3ffbc9a0 0x40092c76:0x3ffbc9c0

Rebooting...
ets Jun  8 2016 00:22:57

```

stuartpittaway, 2021-05-13:
@AlbertWeterings just reviewing the code and the version you are running in this issue is a little old now.

Can I suggest upgrading to the latest version, following the instructions from here https://github.com/stuartpittaway/diyBMSv4ESP32/blob/master/ProgrammingHardware.md

Before the upgrade, if you click on the "Save WIFI" button in the Storage page, the WIFI configuration will be written to the SD CARD so you won't have to reconfigure that again.

[image]

goferthat, 2021-05-13:
Ok.... console output from 2 different controllers (c1/c2) that were simply swapped in/out of an existing 4cell test setup. I think the c2 failure is identical(surmising) to what you've already seen in Albert's example. The C1 failure is different but I can't tell if it's related beyond the fact that it occurred while drumming my fingers on the screen.

Side note ... the C2 failure (my #2 of 4 working controllers) happens either immediately upon touching the screen after it's been idle awhile or within a dozen or fewer presses if it's just reloaded. The C1 failure is less frequent in occurrence but will eventually happen. Although I'm not keeping a log of such things I'm pretty sure I've seen this to varying degrees in all 4 controllers.

I wish I was conversant in this machine code ... after all ... who doesn't like mucking about in bits/bytes/nibbles/and words?!

I'll try the new code this morning ... currently running the 4/7 code so will go to the 5/11 on the C2 board and see if there's a change.

dan

[
[diybmsC1-1-fail_PC0x..5952.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/6472940/diybmsC1-1-fail_PC0x.5952.log)
[diybmsC2-2-fail_PC0x..5958.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/6472941/diybmsC2-2-fail_PC0x.5958.log)

stuartpittaway, 2021-05-13:
Thanks for the log file, similar error but looks in a different place in the code.

Try the newest version and see how that behaves

goferthat, 2021-05-13:
Controller board 0511 code upgrade ... preliminary results as relates to touch screen instability. ... Mixed bag? TBD

Initial results... my Controller #2 board is no better. I'm including the failure (attached) here. I'm beginning to think there's something flaky with it (hardware ... FYI ... C2 has been my problem child for awhile now although it seem to work fine if left alone). 
I also upgraded my Controller #1 board and have been unable (so far) to make it fail. I've not seen one go this long with me drumming my fingers on the screen. 

I'll use the C1 board today to finish top balancing a small 12V pack and report any additional screen issues I see over the next few days.

Dan

[diybmsC2-3_2-Fails_0511-load.log](https://github.com/stuartpittaway/diyBMSv4ESP32/files/6473630/diybmsC2-3_2-Fails_0511-load.log)

> dan

stuartpittaway, 2021-05-13:
Ok, you may also be getting spurious interrupts triggering due to the input ports on the controller.

I probably need to look at doing something different with the configuration of those.

If you have chance connect all the ports to ground with a bit of wire.

goferthat, 2021-05-14:
I/O ports .... the J4 external I/O ports?

AlbertWeterings, 2021-05-14:
I have to say I did not see the reboots for a while, also have to say did not test much. My test setup is on my desk running with me not looking at it. Well This morning I replaced the TCA6408AQPWRQ1 chips for the 5V version and updated to the latest released software. Will try to keep an eye on it the upcoming day's. Finally I hope to find time to hook the BMS up to my pack and start using it for real.

stuartpittaway, 2021-05-14:
> I/O ports .... the J4 external I/O ports?

Yes - marked A/B/C/D on the screw terminals - if you link these together and then connect to the GND on the RS485 or CAN bus screw terminals.

Its possible these inputs are picking up static/interference and causing the interrupts to trigger.

goferthat, 2021-05-14:
OK ... I'll give that a go (J4 inputs to GND). 

As a side note and perhaps supporting Stuart's thoughts here ... yesterday after my testing in the morning that resulted in one of the boards appearing to be 100% stable after the upgrade to the most recent 5/11 release, I took the setup in situ (1 controller, 4 modules and 1 2P4S lifepo battery with all connections), out of my tidy office to the workshop. When I plugged in the power for the controller (using the same p/s as in the office) it powered up nicely and I was able to access it via web no problem. Then.... it reloaded a second or so after I touched the screen!!!  Reboot after reboot it remained unstable.

What's different in the workshop is the proximity of a large switching power supply, fluorescent lights, and miscellaneous live wiring for various projects.  I don't have a scope of any quality that would really show me the noise level out there but it has to be significant vs my office.

Wiring inputs to ground should tell the tale here.

dan

stuartpittaway, 2021-05-14:
Ok, so I can also change the code to force those "inputs" to be "outputs" and hence not triggering on static etc.

Lets see how the grounding goes first.

goferthat, 2021-05-14:
Grounding the I/O pins worked 100% ... was able to re-create the problem at will.

I can see the attraction for the i/o pins (sensing external relay closures or switches) ... was there a plan / purpose for the 'input' aspect of these pins?

dan

stuartpittaway, 2021-05-14:
> purpose for the 'input' aspect of these pins?

They are general purpose I/O - by default they are inputs on the chip, so its an easy change to the code to make them outputs.

At some point in time, rules can be written to read the inputs and respond, but as they are floating at the moment, its causing random interrupts to occur (probably thousands of times per second) which I think is the root cause of the problem

goferthat, 2021-05-14:
Sounds good ... no compelling need to rush as it can be addressed by grounding for the time being. 

Question... do you think pulling the inputs to gnd via a reasonably low value resistor like maybe 250-300 ohms would be any advantage over just wiring to gnd?

dan

stuartpittaway, 2021-05-14:
A resistor won't hurt, even higher values would be fine like 10k

goferthat, 2021-06-01:
As a follow up on this .... I've had this running for more than 2 weeks on a 16S pack with a number of high current charging / discharging sessions and a disgraceful spaghetti mess for wiring.... and I've yet to see ANY odd behavior or screen glitches. 

I ended up tying all 4 inputs together and pulling them to GND with a 10K resistor. All done on the screw terminals eezy peezy.

I'm a happy camper ...

dan

stuartpittaway, 2021-06-01:
Fantastic. On the next code release, I've made those Io ports all outputs so you won't need the resistors.

goferthat, 2021-06-01:
Thanks Stuart ... I think you might be a Rockstar!!

stuartpittaway, 2021-07-06:
Is anyone still having problems with this or can I close this issue?

goferthat, 2021-07-06:
I'm good to close it....

Thanks Stuart.

closed 2021-07-07 by stuartpittaway