stuartpittaway/diyBMSv4ESP32 #12: Wifi Settings lost via "BOOT" button randomly

stuartpittaway, 2021-04-14:
Probably worth removing the wifi reset and leave at serial/usb console only as means to restore lost wifi settings.

vferrandobe, 2021-04-14:
For me that would be more convenient than the current situation.

stuartpittaway, 2021-04-14:
To be fair, the boot button only resets the wifi data if its pressed (and held down for 5 seconds) during the first few minutes of the controller being reset - after that, the button does nothing.

vferrandobe, 2021-04-14:
Then the reset of wifi data must be caused by something else, in my case. 
I don't press the "boot" button to reboot the controller. I just turn the power off and on.
I'll keep on watching it.

stuartpittaway, 2021-04-14:
The "BOOT" button doesn't reboot the controller - thats what the "EN" button does.  The boot button is used to enter some sort of programming mode on power up - I've never actually needed to do this though.

stuartpittaway, 2021-04-14:
Just reviewing the code the BOOT button if held down for 4 seconds will clear the WIFI details.  This only works after a reboot of the controller, until 30 seconds after the controller has gone into running mode (LED would have flashed green)

vferrandobe, 2021-04-14:
Then it is definitely something else what causes the wifi settings lost on my controllers.

stuartpittaway, 2021-04-14:
Its possible the wifi settings would be wiped out if you re-flash the code as well - assume you are not doing that?

If you can get the debugging data from the USB port, that should have some clues

vferrandobe, 2021-04-14:
No, no re-flash when that happens. 
I'll get the debug info next time it happens.

stuartpittaway, 2021-11-26:
Closing this issue as no update for a long period of time.

closed 2021-11-26 by stuartpittaway