stuartpittaway/diyBMSv4ESP32 #22: RFE: monitor2.json on Serial output (micro USB connector)

dakoal, 2021-06-04:
Is it possible to let the monitor2.json output also go through the serial connection?
Maybe with a checkbox enable/disable in the GUI.

Then it will be (for example) possible to read this from a device directly connected, not needing WLAN/CANBUS/RS485 ..... 

Now the output is only 
[I][Rules.cpp:133] SetError(): Set error 1
[I][Rules.cpp:133] SetError(): Set error 1

stuartpittaway, 2021-06-15:
Hi Dakoal, the monitor2.json output isn't really in a formal for transmitting over serial.

We could output the MQTT data packets, thats in a JSON format.

dakoal, 2021-06-15:
I would apreciate any form of celldata.

As soon as I can use it on a Victron/Cerbo GX system I will post it in the openenergymonitor community.

stuartpittaway, 2021-07-06:
I'll be looking at integration with Victro Cerbo using CANBUS as per #27

dakoal, 2021-07-07:
But on other systems it will be very nice

dakoal, 2021-08-11:
Is there a chance to get some "debug" data out of the serial port? If possible, I can try to integrate myself like in the OLD controller. But I don't know if "visual studio code / platform io" is still working like in the past.

stuartpittaway, 2021-08-11:
Hi @dakoal , yes platformIO and VSCode are still used to compile the code.

The USB port on the ESP32 is used for outputting debug information, and you are free to adapt/amend if you wish to add additional debug data.

I'm quite far down the road with the Victron integration already, if you take a look at issue #27 you can see some screenshots.

dakoal, 2021-08-11:
You are very good on the integration, I see.
But the new controller board is not order/build able, because some of the chips are out of stock everywhere.

I am lucky having 3 left for my friends installation, but afterwards I will have to use the old controllers, and they don‘t have CAN.

Maybe I find a way to bend the „debug out“ of the old controller also back to the microusb as here with the ESP32.

dakoal, 2023-03-10:
Hi.

Updated a ESP32 Controller to last version.
Where is monitor2.json gone?

stuartpittaway, 2023-03-13:
It's still there at /api/monitor2

This must be quite an update, the change was made a long time ago!!

closed 2026-11-14 by stuartpittaway