stuartpittaway/diyBMSv4ESP32 #27: Victron support

donnib, 2021-06-28:
Hi @stuartpittaway ,
We discussed Victron support on the Forum. There is a big customer base of Victron users out there and i think Victron support would make this project shine even more so i asked the the question on the Victron Forum how we could get started. You may want to participate especially if i am saying something wrong.

https://community.victronenergy.com/questions/93436/what-kind-of-protocol-do-you-recommend-to-implemen.html

/donnib aka Peter on Victron forum

stuartpittaway, 2021-06-28:
Great, thanks for asking - looks like CANBUS is the way forward

donnib, 2021-06-28:
Sure is, i am willing to help out with testing if we get to that point.

stuartpittaway, 2021-06-28:
I've been offered a Cerbo GX unit from somebody on the forum to play with.

I wonder if Victron do R&D kit for development purposes?

donnib, 2021-06-28:
@stuartpittaway yes and that's a great start, in practice we need to test it in real life and that i can do. I can start preparing to make the new controller, i only have the old one but if you start on doing the implementation for the CAN-BMS support for Victron i have an incentive ;) 
In regard to your question, no i haven't heard of that. 
The CerboGX is the brain in the whole system so i guess that should be enough to get started but i am unsure, we probably have to ask @mpvader aka Matthijs Vader from Victron.

donnib, 2021-06-28:
@stuartpittaway i have sent you a PB in the openenergymonitor.org with my email so we can talk.

donnib, 2021-06-29:
@stuartpittaway check my PM

stuartpittaway, 2021-07-06:
I've received a Cerbo GX unit from Victron, and hope to explore that soon.

donnib, 2021-07-06:
@stuartpittaway great Stuart, ask away if i can help in anyway

donnib, 2021-08-09:
Hi @stuartpittaway,
Have you had anytime to look at this ?

stuartpittaway, 2021-08-09:
Yes, I've got basic CANBUS Victron support working now.  The code is no where near stable enough for release yet!

[image]

[image]

donnib, 2021-08-09:
@stuartpittaway pretty cool ! Well done. Great to hear you have reached so far ;)

SergioRius, 2021-08-14:
@stuartpittaway Amazing! You're the boss. Cheer up.

pas940, 2021-08-21:
This is super exciting. Just looking at your bms to add to my victron system. It looks hard to find hardware at the moment. Most out of stock etc. Will the can be on the same hardware or will it be a new controller, and do you maintain compatibility between all cell models? Just trying to work out what I need.

stuartpittaway, 2021-08-21:
No hardware changes needed for the can to work, and still supports the same cells.

You will probably need the current monitor as well to make it fully functional.

jdevine82, 2021-10-04:
What about adding a dynamic charge limit. What I mean is using the battery current shunt and the charge limit setting you have but make it so that it follows the battery current plus whatever balance current is needed. This will allow the inverter to still use whatever current it wants. This will work better in a full house off grid as it will allow balancing while the load is dynamic.

stuartpittaway, 2021-10-05:
> This is super exciting. Just looking at your bms to add to my victron system. It looks hard to find hardware at the moment. Most out of stock etc. Will the can be on the same hardware or will it be [...]

It is difficult to find the correct parts at the moment - low stock levels for a long time now.  All the modules are still supported, although if you are building new, use the 4.4 version.

stuartpittaway, 2021-10-05:
> What about adding a dynamic charge limit.

I'd have to give that some thought on how it would work.

stuartpittaway, 2021-10-05:
I'm going to close this issue, as its got several different comments/threads running.  Please open separate issues for different topics. Thanks.

closed 2021-10-05 by stuartpittaway