stuartpittaway/diyBMSv4ESP32 #28: Juntek shunts

slundell, 2021-06-28:
Today I installed a diyBMS (esp32) setup in my boat. One controller and 8 battery modules. I already have two Juntek VAT-1050 shunts for measuring currents installed (one on all charge connections and one on all consumption connections). These shunts output their data on RS485 (using a USB connector!) and wireless using an nRF24* clone.

These units are rather cheap (~$30) and are available from all the standard Chinese resellers.

They have some "rules" functionality such as over-current and under-voltage which can trigger a relay. The protocol is largely known. 

I was thinking about integrating these shunts in a similar way to the way the diyBMS-shunt is handled. Given that it is a rs485-protocol I was thinking I should re-use much of the code you have written for the diyBMS-shunt. Do you have any pointers?

stuartpittaway, 2021-06-28:
My shunts use rs485 and Modbus protocol.

If you look at the code there are two tasks, one to request some data from the shunt and another which receives it.

After that, the code is just unpacking the reply and putting it into memory of the controller.

slundell, 2021-07-12:
I have decoded the data from the shunt using the existing code, but changing a few registers and conversion factors. I am a bit confused about filling in the shunt-struct. How are `currentMonitor.milliamphour_in` and `currentMonitor.milliamphour_out` differentiated? As I understand the shunt will only measure the sum of the charge and discharge and not separate the current paths and measure them separately? If that is the case, when are `currentMonitor.milliamphour_in` and `currentMonitor.milliamphour_out` updated, and how?

stuartpittaway, 2021-07-12:
I can measure the current direction and hence calculate a total amo hour counter.

This is obviously only measuring the portion of power in/out of the battery.

slundell, 2021-07-12:
OK, so if current is greater than 0, increment `milliamphour_in` and if less than 0, increment `milliamphour_out`? (or vice versa, depending on shunt orientation)

stuartpittaway, 2021-07-12:
Yes, although the ina228 uses charge in columbs, which is then converted into energy and then current

buddhafragt, 2021-08-06:
Hello slundell,
did you manage to integrate the juntec? 
I would also like to integrate the current sensor, maybe you can provide the code.

stuartpittaway, 2022-02-24:
I didn't purchase one in the end, but it does look very similar to the JUNKTEC shunt I've just finished integration with.