stuartpittaway/diyBMSv4ESP32 #30: Influx Integration only working for bank 0

dakoal, 2021-07-11:
Influx Integration only working for bank 0

All following banks show same values as first one.

dakoal, 2021-08-13:
Is there an update?

stuartpittaway, 2021-08-13:
I've fixed a related bug today which causes the controller to reboot due to low memory.

I'll take a look at this issue soon

stuartpittaway, 2021-08-16:
Hi @dakoal.

Think I've fixed the issue, and at the same time I've upgraded the code to support the InfluxDB v2 API

The code is in this branch - are you okay compiling the code from this to test?

https://github.com/stuartpittaway/diyBMSv4ESP32/tree/influxdb2

dakoal, 2021-08-17:
I will try. Give you an answer here

dakoal, 2021-08-17:
[image]

Which file to flash? is it simply the "IMG" or do I have to do some magic with the ELF?

stuartpittaway, 2021-08-17:
If you have used platformio to compile it, just press the upload button in there (connect ESP32 using usb cable)

dakoal, 2021-08-17:
I am in crete and Karl in Austria. My Laptop here is the one with platform io and karl has the system with 5 banks.

stuartpittaway, 2021-08-17:
Ah ok, the ".bin" file should be the one to use.

You should be able to use a tool called "esptool.py" to send that to the ESP32

closed 2021-09-02 by stuartpittaway