stuartpittaway/diyBMSv4ESP32 #35: [Question] Persistance

slundell, 2021-07-24:
I need to persist a few values over resets, and for later calculations. What functions or strategies are the best for these scenarios?

Should ideally be persisted every update (ca 5 sec). Needs to be restored on reset
* `currentMonitor.milliamphour_in`
* `currentMonitor.milliamphour_out`
* current SOC for each cell 

On change
* capacity of each cell

For visualisations (best to use SD card i assume):
* Current, power, shunt voltage etc.

stuartpittaway, 2021-07-27:
Hello.

Each cell module would need to persist its own data, do we really need SOC for each cell?

The existing diybms current shunt persist its own values, so assuming the juntek can't do this?

I would continue to use the eeprom/flash emulation code on the ESP32 to store values, rather than SD card as not everyone uses those.

Not sure if there are penalty on speed or flash write cycles for writing so regularly

slundell, 2021-07-29:
My initial idea was that the nominal capacity for each cell should be estimated. In an unbalanced pack, cells could be charged over 100% (based on e.g. a cut-off voltage). As you indicate this might be an unnecessary complication. SoC per bank makes more sense. The current passing through one cell will pass through the others, and SoC as a bank is of more informational value. If a cell is being over charged, the bank should be considered fully charged. 

The SoC estimation is just counting mAh in and out from a given reference point (like shunt cumulative mAh) From this min/max charge is calculated, and from that capacity and SoC. If the reference point is reset, some of the calculated values also needs to be reset. 

As of now i need to persist cumulative count/min/max/soc (int32_t) per bank over reset. In my first version i commit this to SD as a json file every 5 seconds. The frequency could be lowered if it has any negative impacts. 

The flash/eeprom emulation you mention, is there any section of the code where this is used that I can use as an example?

stuartpittaway, 2021-07-29:
Take a look at the settings code that writes the configuration structure to flash.

slundell, 2021-07-31:
Thanks for the pointer! Using that code made things much easier :)  I have added persistance for the Juntek shunt data as needed.

stuartpittaway, 2021-07-31:
Ok, I think the SOC needs to be more complex than you describe, with basic amp hour counting, it very quickly drifts out of alignment.

There is some complex maths you can do to calculate this, but I've not seen anything concrete for lithium cells.

Perhaps just asking the end user for a total capacity, and using that as the 100 percent full level, and just subtract the amphours from that for a percentage? 

I think this is what you are suggesting anyway.

Should this logic live in the controller or the shunt?

slundell, 2021-07-31:
Yes, I am aware that "dead reckoning" Ah counting is not ideal. It needs a reference (like 0% charged, or 100% charged) and it needs to hit that reference regularily. From this charging efficiency coefficient may be estimated and then be included in the SoC estimation. 

I know that there exists a lot of algorithms for calculating SoC that will get better accuracy. In my use case, I hit either 0% or 100% (based on voltage) usually within 24h. According to https://www.mdpi.com/1996-1073/14/11/3284/pdf this method should hit 4% accuracy, which works for me.

Other contributors may patch my code for more accurate estimations, should they be needed. If it is not accurate enough for my use case,  I might patch it.

stuartpittaway, 2021-11-26:
No comments for a while, closing issue.

closed 2021-11-26 by stuartpittaway