stuartpittaway/diyBMSv4ESP32 #36: MQTT: cell 8 not sending any messages

SKOJau, 2021-07-28:
Somehow i don't get any MQTT messages from my 8th cell. 
Version: 6c3d59e04910a67302dc542fed006e4628ca6fa7 
Compiled: 2021-06-16T12:09:39.012Z

[image]

SKOJau, 2021-07-28:
Same problem in the newest Version:

Version: 4a507db2938399f81685992ceac2f189eff723b6
Compiled: 2021-07-06T10:45:43.400Z

stuartpittaway, 2021-08-02:
I've uploaded a new version which should fix this bug.

SKOJau, 2021-08-08:
Hey Stuart, thanks for the quick answer.
With the latest Version 53e5cb8b81e287c3202933d820292db1995476f5 I don't get any cell data via MQTT anymore. Just the Bank voltage is transmitted.
The Influx implementation seems to work now.

[image]

stuartpittaway, 2021-08-09:
Reviewing the code, the MQTT message won't be sent if there is an error on the BMS (the other values are sent) - you should always get something transmitted if there are no errors.
```
      //If the BMS is in error, stop sending MQTT packets for the data
      if (!rules.rule_outcome[Rule::BMSError])
      {
```

So do you know if any errors were active on the BMS (check the rule page), if you leave it running for longer do the MQTT messages start to appear?

stuartpittaway, 2021-08-10:
Ok, I've found the root cause of this, I've updated the code again, please re-test @SKOJau

closed 2021-09-02 by stuartpittaway