stuartpittaway/diyBMSv4ESP32 #38: Controller reboots by touching the screen

SergioRius, 2021-08-13:
Screen touches makes the controller reboot. You only have to slide the finger around.
I'm on version 4.3, but i suppose it will happen on master.

SergioRius, 2021-08-14:
I've updated to the latest automatic build (21/08/10) and the issue persists. Just draw circles on the screen and the controller reboots.

stuartpittaway, 2021-08-14:
Ok, mine doesn't do this.

Can you capture the USD serial output to get the error message?

SergioRius, 2021-08-14:
I suppose. Via micro USB port I understand.

stuartpittaway, 2021-08-14:
Yes the microusb

stuartpittaway, 2021-09-03:
Is this still occuring.  I've got 2 controllers running, both with screens - 1 will occasionally reset if I press the screen firmly, the other never reboots.

I'm wondering if the screen isn't built properly and its shorting out the power supply to the ESP?

SergioRius, 2021-09-03:
I've to receive another screen. I'll check it as soon as it arrives.

stuartpittaway, 2021-10-05:
Any update on this issue?

stuartpittaway, 2021-10-05:
````
[I][AsynGuru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400959e4  PS      : 0x00060434  A0      : 0x8009480f  A1      : 0x3ffc0260
A2      : 0x3ffc1d94  A3      : 0x3ffc1be0  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060423  A7      : 0x00000000  A8      : 0x3ffc1be0  A9      : 0x3ffc1be0
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060421  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9  
Core 1 was running in ISR context:
EPC1    : 0x4000beb4  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400959e4

ELF file SHA256: 0000000000000000

Backtrace: 0x400959e4:0x3ffc0260 0x4009480c:0x3ffc0280 0x40092b27:0x3ffc02a0 0x40110bed:0x3ffc02e0 0x400811c8:0x3ffc0340 0x40081281:0x3ffc0360 0x4008f465:0x3ffc0380 0x40081288:0x3ffb1e00 0x40093e61:0x3ffb1e10 0x4009611b:0x3ffb1e30 0x40096882:0x3ffb1e50 0x40081f9d:0x3ffb1e70 0x40081fce:0x3ffb1e90 0x40088335:0x3ffb1eb0 0x4000beaf:0x3ffb1ed0 0x4019f113:0x3ffb1ef0 0x4019ee95:0x3ffb1f10 0x401a22f1:0x3ffb1f30 0x4010d25c:0x3ffb1f70 0x400e940f:0x3ffb1f90 0x40110ef1:0x3ffb1fb0 0x40092cee:0x3ffb1fd0
  #0  0x400959e4:0x3ffc0260 in vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188 (discriminator 1)
  #1  0x4009480c:0x3ffc0280 in vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #2  0x40092b27:0x3ffc02a0 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2038
  #3  0x40110bed:0x3ffc02e0 in log_printf at C:\Users\Stuart\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-uart.c:544
  #4  0x400811c8:0x3ffc0340 in TFTScreenTouchInterrupt() at src/tft.cpp:56
  #5  0x40081281:0x3ffc0360 in __onPinInterrupt at C:\Users\Stuart\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-gpio.c:274
  #6  0x4008f465:0x3ffc0380 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #7  0x40081288:0x3ffb1e00 in __onPinInterrupt at C:\Users\Stuart\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-gpio.c:274
  #8  0x40093e61:0x3ffb1e10 in vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #9  0x4009611b:0x3ffb1e30 in multi_heap_internal_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
  #10 0x40096882:0x3ffb1e50 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305
  #11 0x40081f9d:0x3ffb1e70 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #12 0x40081fce:0x3ffb1e90 in heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #13 0x40088335:0x3ffb1eb0 in _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
  #14 0x4000beaf:0x3ffb1ed0 in ?? ??:0
  #15 0x4019f113:0x3ffb1ef0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:50
  #16 0x4019ee95:0x3ffb1f10 in operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #17 0x401a22f1:0x3ffb1f30 in WiFiUDP::parsePacket() at C:\Users\Stuart\.platformio\packages\framework-arduinoespressif32\cores\esp32/Udp.h:46
  #18 0x4010d25c:0x3ffb1f70 in ArduinoOTAClass::handle() at C:\Users\Stuart\.platformio\packages\framework-arduinoespressif32\libraries\ArduinoOTA\src/ArduinoOTA.cpp:105
  #19 0x400e940f:0x3ffb1f90 in loop() at .pio/libdeps/esp32-devkitc-debug/ArduinoJson/src/ArduinoJson/Numbers/parseNumber.hpp:144
  #20 0x40110ef1:0x3ffb1fb0 in loopTask(void*) at C:\Users\Stuart\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #21 0x40092cee:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)    

Core 0 register dump:
PC      : 0x40093cf2  PS      : 0x00060034  A0      : 0x800944d5  A1      : 0x3ffbfd00
A2      : 0x3ffbf75c  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffca328
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3ffbde2c
A14     : 0x0000002a  A15     : 0x3ffd36b0  SAR     : 0x0000001b  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256: 0000000000000000

Backtrace: 0x40093cf2:0x3ffbfd00 0x400944d2:0x3ffbfd30 0x40092ddb:0x3ffbfd50 0x40095861:0x3ffbfd70 0x4008f46e:0x3ffbfd80 0x40009217:0x3ffd35e0 0x40007d13:0x3ffd3600 0x40007c69:0x3ffd3620 0x40008106:0x3ffd3640 0x40110bf7:0x3ffd36d0 0x400f2ecb:0x3ffd3730 0x400ebe21:0x3ffd3760 0x40092cee:0x3ffd3a80
[... cut ...]
  #12 0x40092cee:0x3ffd3a80 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)    

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
````

stuartpittaway, 2021-10-05:
Seems to be caused by excessive ESP_DEBUG logging during ISR

closed 2021-10-05 by stuartpittaway