stuartpittaway/diyBMSv4ESP32 #40: Add "Battery capacity"

magpern, 2021-08-23:
**State of charge**
For a State of Charge to work, Shunt needs to know the battery capacity.
Add a "Battery capacity" field in shunt settings.
Display SoC in main window (with one decimal).

**"Consumed Ah"** would be a nice feature.
When Battery is full, manually reset the SoC is 100% or otherwise set to "now battery is full", Consumed Ah is 0Ah
While discharing, Consumed Ah decresses, it counts down.
When charging, it counts up.
When it hits Consumed Ah 0Ah it stops counting. It will never raise above 0Ah.

**Syncronize SoC**
There should be a way to manually set SoC. If you know batteries are full, you manually syncronize SoC to 100%.
This resets "Consumed Ah" to 0Ah
Likewize, if you know SoC is 50%, "Consumed Ah" is set to half of "Battery capacity" times minus 1

**Automatic SoC syncronize**
Introduce a "Battery full voltage".
When shunt voltage reaches this value and stays above this value för "Charge detection time" (3 minutes) and current is less than "Tail current" (1%) (1% of "Battery capacity" eg. 560Ah battery when hits 56V for three minutes and it charges with less than 560*0.01 = 5.6A during this time = battery is full. "Consumed Ah" is reset to 0Ah

**Unnecessary values**
The other values "Ah In" and "Ah out", contributes with no usefull data (circled green).
[image]
Personally I don't think they belong in a overview window, more like a "extended data section"
Since they never resed, they are more a "Total Ah in/out" counter.

stuartpittaway, 2021-08-23:
Hi, thanks for raising the question/issue.

Bizarely, I'm already working on SOC calculation for the current monitor.  These new options will soon be available in the configuration to set the battery capacity, and the charge efficiency.

The voltage and tail current also allow automatic reset of the counters when 100% is reached.

[image]

magpern, 2021-08-25:
Somewhere in the code there is a bug. I have no other info. But restarted everything last night, so it started from zero.
[image]

closed 2021-09-02 by stuartpittaway