stuartpittaway/diyBMSv4ESP32 #46: Controller reboots when downloading (large) files in GUI

sparkiesweden, 2021-09-03:
Downloading any logged files larger than ~100k results in a hang and reboot of controller.
Small files (readme.txt) downloads correctly.

-----------------------------------------------------

Platform & Version
Host name:DIYBMS-00312664
Processor: ESP32
Version: 4a507db2938399f81685992ceac2f189eff723b6
Compiled: 2021-07-06T10:45:43.400Z
Language: language
SDK Version:v3.3.5-1-g85c43024c
Min free Heap:85328
Free heap:89320
Heap size:297220

----------------------------------------------------------------------------------------------------------------------------------------------------
[I][AsyncMqttClient.cpp:429] _handleQueue(): p #[I][AsyncMqttClient.cpp:429] _handleQueue(): p #3 rel
[I][DIYBMSServer.cpp:1504] downloadFile(): Download SDCard file /modbus90_20210903.csv
[I][main.cpp:1659] mqtt2(): Send MQTT Status
[I][AsyncMqttClient.cpp:740] publish(): PUBLISH
[I][AsyncMqttClient.cpp:387] _addBack(): new back #3
[D][main.cpp:2315] rs485_rx(): Rec 83 bytes, id=90
[D][main.cpp:2096] ProcessCurrentMonitorRegisterReply(): Read relay trigger settings 1 252
[I][AsyncMqttClient.cpp:740] publish(): PUBLISH
E (31847) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (31847) task_wdt:  - async_tcp (CPU 0/1)
E (31847) task_wdt: Tasks currently running:
E (31847) task_wdt: CPU 0: IDLE0
E (31847) task_wdt: CPU 1: loopTask
E (31847) task_wdt: Aborting.
abort() was called at PC 0x40168e5c on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40090d8c:0x3ffbfd10 0x40091009:0x3ffbfd30 0x40168e5c:0x3ffbfd50 0x4008f46d:0x3ffbfd70 0x401977ef:0x3ffbca00 0x4016a803:0x3ffbca20 0x400944a5:0x3ffbca40 0x40092cf6:0x3ffbca60

Rebooting...
ets Jun  8 2016 00:22:57

sparkiesweden, 2021-09-03:
Seems to work in latest (September) release. So fixed? Sorry for spam if so.

stuartpittaway, 2021-09-03:
Ok, I've not specifically seen that issue (or fixed it!) but glad its working now.

closed 2021-09-03 by stuartpittaway

delboy711, 2022-02-08:
I have seen the same issue when accessing the 'Storage' page.  If there are a lot of files on the SD card I get a watchdog timeout just like the OP.  My diyBMS code is heavily modified so my CPU is more heavily loaded than a standard config so I am not expecting Stuart to do anything about it.

There is an issue open for this type of error on the AsyncTCP project [https://github.com/me-no-dev/AsyncTCP/issues/121](url)
A fix has been proposed [https://github.com/me-no-dev/AsyncTCP/pull/122](url)

I have tried out the fix and it works for me.  When accessing the storage page it takes a long time before coming back with the file list, but it does not trigger the watchdog.