stuartpittaway/diyBMSv4ESP32 #49: MQTT compatibility with HomeAssistant

vajpeters, 2021-09-11:
Since HomeAssistant is getting a defacto standard, it would be nice to create a MQTT compatibility mode with HomeAssistant.

shampeon, 2021-09-11:
It would be a sensor device, where you set the class and unit, etc.

For example, this is the discovery topic for the sensors from an MPP Solar inverter.

```
homeassistant/sensor/Inverter_100177/Battery_voltage/config {
            "name": "Battery_voltage_Inverter",
            "uniq_id": "100177_Battery_voltage",
            "device": { "ids": "100177", "mf": "MPPSolar", "mdl": "PIP3048LV-MK", "name": "Inverter", "sw": "61.01"},
            "state_topic": "homeassistant/sensor/Inverter_100177/Battery_voltage",
            "state_class": "measurement",
            "unit_of_meas": "V",
            "icon": "mdi:battery-outline"
        }
```

natiz, 2021-12-28:
+1 for this, I would look into adding a [esphome](https://esphome.io/) component ([a custom one](https://esphome.io/custom/custom_component.html) for starters) instead of using mqtt directly.

Have a look at [Daly BMS component](https://esphome.io/components/sensor/daly_bms.html) the for inspiration: https://github.com/esphome/esphome/tree/dev/esphome/components/daly_bms

Linusten, 2023-05-30:
+1 on this, maybe someone is able to add this :)

stuartpittaway, 2023-08-18:
It's in the pipeline!