stuartpittaway/diyBMSv4ESP32 #58: Firmware updates erase all settings, except wifi

magpern, 2021-10-05:
When I did a firmware update, all settings, battery levels, relay settings, shunt settings where erased.
I selected "Wipe Eeprom = No"

The wifi setting was not erased. I did press save wifi and save configuration before the firmware update

[image]

stuartpittaway, 2021-10-05:
Unfortunately this is how the current firmware updates work.  The WIFI details are stored on the SD card and automaticallly restored when powered up after an upgrade.

I intend to do the same with the rest of the system settings  - which is what the save configuration button intends to do, but not fully implemented.

stuartpittaway, 2022-05-04:
The next version of the controller code supports saving and restoring configuration to the sd card.  (NOV2021 branch)

closed 2022-05-04 by stuartpittaway