stuartpittaway/diyBMSv4ESP32 #65: ESP32 errors when trying to connect to wifi via putty

gjonesy84, 2021-11-23:
Hi,

Flashed ESP32 with nodemcu-pyflasher and all was ok no issues. Then used putty and connected via serial to try and connect o wifi but getting this response.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

                _          __
    _|  o      |_)  |\/|  (_
   (_|  |  \/  |_)  |  |  __)
           /
[I][main.cpp:3286] setup(): CONTROLLER - ver:019382f764db386f6cab8d8338ad073d21b                                                                                        7bad0 compiled 2021-10-05T10:47:31.492Z
[I][main.cpp:3291] setup(): ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
[I][HAL_ESP32.cpp:232] ConfigureI2C(): Configure I2C
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x400                                                                                        91293
file: "src/HAL_ESP32.cpp" line 60
func: esp_err_t HAL_ESP32::writeByte(i2c_port_t, uint8_t, uint8_t, uint8_t)
expression: i2c_master_cmd_begin(i2c_num, cmd, pdMS_TO_TICKS(100))
[E][HAL_ESP32.cpp:277] ConfigureI2C(): TCA9534APWR Error
[E][HAL_ESP32.h:181] Halt(): SYSTEM HALTED
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40091                                                                                        293
file: "src/HAL_ESP32.cpp" line 60
func: esp_err_t HAL_ESP32::writeByte(i2c_port_t, uint8_t, uint8_t, uint8_t)
expression: i2c_master_cmd_begin(i2c_num, cmd, pdMS_TO_TICKS(100))
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40091                                                                                        293
file: "src/HAL_ESP32.cpp" line 113
func: void HAL_ESP32::Led(uint8_t)
expression: writeByte(I2C_NUM_0, TCA9534APWR_ADDRESS, TCA9534APWR_OUTPUT, TCA953                                                                                        4APWR_Value)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40091                                                                                        293
file: "src/HAL_ESP32.cpp" line 60
func: esp_err_t HAL_ESP32::writeByte(i2c_port_t, uint8_t, uint8_t, uint8_t)
expression: i2c_master_cmd_begin(i2c_num, cmd, pdMS_TO_TICKS(100))
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40091                                                                                        293

Any idea why I am getting these errors.

The esp32 i am using are the recommeneded ones.

https://www.amazon.co.uk/gp/product/B08DXSMZSB/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

cheers

stuartpittaway, 2021-11-23:
It's not a problem with the esp, see this line....

TCA9534APWR Error

There is a fault with this part or the soldering. The esp cannot communicate with it.

gjonesy84, 2021-11-23:
Ah nice. Cheers. I will replace. Maybe a duff chip.

closed 2021-11-26 by stuartpittaway