stuartpittaway/diyBMSv4ESP32 #69: Spontaneous reboots of the esp32 controller

voort4444, 2021-12-04:
Hi Stuart, 
I finally got the serial outut when my esp32 controller reboots:
I pasted a bit before and an bit after the reboot so you have an idea what it looks like "normally", hope this helps

[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40151736  PS      : 0x00060830  A0      : 0x80149f26  A1      : 0x3ffd8d40  
A2      : 0x3ffecb3c  A3      : 0x00450008  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3ffd81e4  A7      : 0x3ffd8280  A8      : 0x007ff834  A9      : 0x0080036c  
A10     : 0x00000000  A11     : 0x00000359  A12     : 0x00800013  A13     : 0x00000b38  
A14     : 0x00060820  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00450018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40151736:0x3ffd8d40 0x40149f23:0x3ffd8d70 0x400e5de2:0x3ffd8d90 0x40149064:0x3ffd8db0 0x40092ce2:0x3ffd8de0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

                _          __ 
    _|  o      |_)  |\/|  (_  
   (_|  |  \/  |_)  |  |  __) 
           /                  
[I][main.cpp:3286] setup(): CONTROLLER - ver:019382f764db386f6cab8d8338ad073d21b7bad0 compiled 2021-10-05T10:47:31.492Z
[I][main.cpp:3291] setup(): ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
[I][HAL_ESP32.cpp:232] ConfigureI2C(): Configure I2C
[I][HAL_ESP32.cpp:291] ConfigureI2C(): Found TCA9534A
[I][HAL_ESP32.cpp:321] ConfigureI2C(): Found TCA6408
[D][HAL_ESP32.h:288] TouchScreenUpdate(): Touch = touch=0 pressure=0 x=0 y=0
[I][main.cpp:3313] setup(): TFT screen is NOT installed
[I][main.cpp:934] SetControllerState(): ** Controller changed state from Unknown to PowerUp **
[I][main.cpp:3328] setup(): LITTLEFS mounted, totalBytes=589824, usedBytes=81920
[I][main.cpp:241] mountSDCard(): Mounting SD card
[W][sd_diskio.cpp:101] sdWait(): Wait Failed
[E][sd_diskio.cpp:123] sdSelectCard(): Select Failed
[W][sd_diskio.cpp:505] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[W][sd_diskio.cpp:101] sdWait(): Wait Failed
[E][sd_diskio.cpp:123] sdSelectCard(): Select Failed
[E][main.cpp:260] mountSDCard(): Card Mount Failed
[I][HAL_ESP32.cpp:140] ConfigureCAN(): CAN driver installed.  Filter=1621098496 Mask=6291455
[I][HAL_ESP32.cpp:150] ConfigureCAN(): CAN driver started
[D][settings.cpp:20] ReadConfig(): ReadConfig diybms
[D][settings.cpp:42] ReadConfig(): checksum verified
[I][main.cpp:3352] setup(): 
[D][main.cpp:200] SetupRS485(): Setup RS485
[D][main.cpp:187] ConfigureRS485(): Configure RS485
[D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 0=153
[D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 1=153
[D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 2=153
[D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 3=153
Press SPACE BAR to enter terminal based configuration................skipped
[D][settings.cpp:20] ReadConfig(): ReadConfig diybmswifi
[D][settings.cpp:42] ReadConfig(): checksum verified
[I][main.cpp:934] SetControllerState(): ** Controller changed state from PowerUp to Stabilizing **
[D][main.cpp:757] tca6408_isr_task(): tca6408_isr
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[I][main.cpp:1422] connectToWifi(): Hostname: DIYBMS-0043C924, current state 255
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.36, MASK: 255.255.255.0, GW: 192.168.1.1
[I][main.cpp:1511] onWifiConnect(): Wi-Fi status=3
[I][main.cpp:1513] onWifiConnect(): Request NTP from time.google.com
[I][DIYBMSServer.cpp:2684] StartServer(): Start Web Server complete
[I][ArduinoOTA.cpp:141] begin(): OTA server at: DIYBMS-0043C924.local:3232
[I][main.cpp:1547] onWifiConnect(): mDNS responder started
[I][main.cpp:1552] onWifiConnect(): You can access DIYBMS interface at http://DIYBMS-0043C924.local or http://192.168.1.36
[I][Rules.cpp:142] SetError(): Set error 4
[I][Rules.cpp:142] SetError(): Set error 4

OOS Error, expected=33A, got=1
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[I][main.cpp:934] SetControllerState(): ** Controller changed state from Stabilizing to Running **
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1

voort4444, 2021-12-04:
In addition, I also found this one, looks different to me:

I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

and this sometimes:
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[I][Rules.cpp:128] SetWarning(): Set warning 1
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[I][Rules.cpp:128] SetWarning(): Set warning 1

and one more of the guru meditation:
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40151736  PS      : 0x00060e30  A0      : 0x80149f26  A1      : 0x3ffd8d40  
A2      : 0x3ffebee0  A3      : 0x00450008  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3ffd8238  A7      : 0x00000000  A8      : 0x80092dd2  A9      : 0x3ffbfd30  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffcae44  A13     : 0x00000b38  
A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00450018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40151736:0x3ffd8d40 0x40149f23:0x3ffd8d70 0x400e5de2:0x3ffd8d90 0x40149064:0x3ffd8db0 0x40092ce2:0x3ffd8de0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
[... 64 lines cut ...]
OOS Error, expected=3A0, got=2
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][main.cpp:1459] influxdb_task(): Influx task
[D][influxdb.cpp:140] influx_task_action(): Create new client
[I][influxdb.cpp:177] influx_task_action(): Influx host 192.168.1.50 port 8086
[D][influxdb.cpp:73] influxdb_onConnect(): Influx connected
[D][influxdb.cpp:126] influxdb_onConnect(): Influx data sent
[D][influxdb.cpp:47] influxdb_onData(): Influx data received
[I][influxdb.cpp:65] influxdb_onDisconnect(): Influx disconnected
[I][Rules.cpp:128] SetWarning(): Set warning 1
[I][Rules.cpp:128] SetWarning(): Set warning 1

kendownie, 2021-12-14:
I'm also seeing these issues when using Influx. I have one example captured and my attempt to use the decode.py script on the stack dump.

[D][influxdb.cpp:47] influxdb_onData(): Influx data received
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x1563010c  PS      : 0x00060a30  A0      : 0x800e76c0  A1      : 0x3fff11f0
A2      : 0x3fff121c  A3      : 0xb60192e0  A4      : 0x00000002  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x80198e44  A9      : 0x3fff12d0
A10     : 0x3fff121c  A11     : 0x3fff121c  A12     : 0x00000003  A13     : 0x00000000
A14     : 0x3ffc4928  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x00000014
EXCVADDR: 0x1563010c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x1563010c:0x3fff11f0 0x400e76bd:0x3fff1210 0x400eea71:0x3fff1250 0x400f088a:0x3fff12c0 0x400e6fa6:0x3fff12e0 0x400e7129:0x3fff1310 0x400e7139:0x3fff1330 0x400e746b:0x3fff1350 0x40092ce2:0x3fff1380

Rebooting...
ets Jul 29 2019 12:21:46

**Output from decode.py on the above backtrace:**

stack:
0x400e76bd: std::function<void (void*, AsyncClient*, signed char)>::~function() at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:1036
  \-> inlined by: std::function<void (void*, AsyncClient*, signed char)>::operator=(std::function<void (void*, AsyncClient*, signed char)> const&) at c:\users\ken\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:2075
  \-> inlined by: AsyncClient::onError(std::function<void (void*, AsyncClient*, signed char)>, void*) at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:649
0x400eea71: AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\ESP Async WebServer\src/WebRequest.cpp:797
0x400f088a: std::_Function_handler<void (void*, AsyncClient*), AsyncWebServer::AsyncWebServer(unsigned short)::{lambda(void*, AsyncClient*)#3}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\ESP Async WebServer\src/StringArray.h:73
  \-> inlined by: _M_invoke at c:\users\ken\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
0x400e6fa6: std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:1036
0x400e7129: AsyncServer::_accepted(AsyncClient*) at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:1036
0x400e7139: AsyncServer::_s_accepted(void*, AsyncClient*) at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:1036
0x400e746b: _async_service_task(void*) at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:1036
  \-> inlined by: _async_service_task at C:\Users\ken\Desktop\diyBMSv4ESP32-Tag-2021-12-06-15-12\ESPController/.pio\libdeps\esp32-devkitc\AsyncTCP\src/AsyncTCP.cpp:197
0x40092ce2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

stuartpittaway, 2021-12-18:
Related to #47

stuartpittaway, 2022-12-05:
Closing this issue as reboot issues have been resolved over 6m+

closed 2022-12-05 by stuartpittaway