stuartpittaway/diyBMSv4ESP32 #71: influxDB timing - configure intervals in gui

virtuvas, 2021-12-18:
I noticed that influxDB sends records in a v.rapid succession.
Having VenusOs already sending data to a separate influxDB at 60sec intervals (configurable at the gui) which are then viewed in Graphana, I feel that this would be a worthy inclusion in diyBMS. 

Considering that cell Voltage/intTemp/extTemp/balancing status(?) are the only things exported via influxDB, I'm pretty sure that sending a record every 20-30 or even 60secs would be sufficient especaialy in a well setup and running system. OK, under debugging, I could see it sending every sec or less.

I guess there are two options:
A. internally record/average between postings and sent the average at set intervals
B. when time comes just sent current value
probably wouldn't mind either, up to you I guess.

cheers

V.

stuartpittaway, 2021-12-18:
Thanks for the issue.  Currently the influx data is transmitted every 8 seconds.

I'll add this to the list of future enhancements.

virtuvas, 2022-03-01:
thanks, works fine, closing

closed 2022-03-01 by virtuvas