stuartpittaway/diyBMSv4ESP32 #74: Packet Loss

allanlind, 2021-12-25:
I seem to be getting packet loss, and the data LEDs seem to illuminate up as far as module 16 when the data is going around the daisy chain but there is a few second delay before module 17 data LED illuminates, then the next 16 modules flash the data LED in sequence.
Version is Compiled_Firmware_2021-12-06-15-12.
[image]

[image]

stuartpittaway, 2021-12-25:
Another user has reported similar behaviour which I've fixed and they are just finishing testing.

What version of the modules are you using?

allanlind, 2021-12-25:
Modules are 4.4 and flashed from the controller with Dec 6 2021 compiled version.

allanlind, 2022-01-01:
Another problem I discovered is that if I connect any more than about 84 modules at a time I lose connectivity with the last modules int the group after an hour or so. I currently have a string of 112 modules. 
Stuart, is there any logs I can send you?
I sent Tim Yorba 10 of the integrated 14 cell boards so he will be able to run tests as well as soon as he receives the shipment.

allanlind, 2022-01-01:
[image]

stuartpittaway, 2022-01-03:
The high number of cells is currently being worked on and looks like we have a suitable fix for it, assuming they are 4.4 version

allanlind, 2022-01-03:
Fantastic thank you. I’ll keep an eye out for it.

stuartpittaway, 2022-01-05:
@allanlind I've just merged the code into the master branch, so a new release should be available - its just a controller update.

allanlind, 2022-01-05:
Thanks Stuart. That works better. Not losing connectivity to the last 20 or so modules now. 
Still having those big delays though. 
I was discussing with Tim Yorba and he suggested we might want to increase the watchdog timeout as the modules may be timing out waiting for the next round of packets.

stuartpittaway, 2022-01-06:
You can't change the watchdog, it's hardware based and currently at its maximum of 8 seconds.

What do you mean by delays? Are You using v4.4 modules at a higher Comms speed?

allanlind, 2022-01-06:
Stuart, as I said previously, the blue data LEDs illuminate up as far as module 16 when the data is going around the daisy chain but there is a few seconds delay before module 17 data LED illuminates, then the next 16 modules flash the data LEDs in sequence around the loop, always in blocks of 16 modules. 
I have flashed all the modules at 9.6k.

stuartpittaway, 2022-01-06:
That's perfectly normal behaviour and how the system works. Only 16 modules can be addressed at a time, so the 17 module has to wait for the request to skip through the previous 16 modules before the lights flash.

allanlind, 2022-01-06:
Whew! thanks Stuart.  Now I can sleep well.

stuartpittaway, 2022-01-06:
More importantly, since the update, all modules are reporting correctly and you don't see any "OOS" errors ?

What round trip time do you have?

allanlind, 2022-01-06:
[image]
[image]

allanlind, 2022-01-06:
just under 6 seconds. Only a few OOS errors and looks like most packets are arriving back.

closed 2022-01-19 by stuartpittaway