stuartpittaway/diyBMSv4ESP32 #75: TFT screen text is stuck to previous values? Restarting from webinterface wont update screen!

virtuvas, 2021-12-28:
an odd one happened 2-3 times already in last week testing.

latest s/w, canbus/influx on 4.4 modules, system charging, values on screen move as normal and in tune with webinterface. All fine.
At some point TFT screen gets stuck, add a load to the bank, values drop on webinterface, remain stuck to old values on the TFT.
Restart from webinterface, came back up with old data (12.29V whereas I'm currently at 10.4V) influx/vecan report 10.4V.
Note when restarting, it goes through the led flashing as per normal, relays clicking but screen doesn't come with powering up and then stabilising as normal. Screen is off for 2secs and then straight on to old data display. Touch works fine, waking up screen comes with same data.
Only solution is to unplug the usb from the PC, then it goes through the whole booting process (at screen display level)

known issue? any tests to do?

virtuvas, 2021-12-28:
I think it is somehow related to my PC going into hybernation mode and turning off (leaving power to the USB port that the ESP is connected to!) ESP/bms works fine throughout the period that PC is off hybernating.
Did it again, PC turned off an hour ago, turned it on, screen works showing 10.30 or something (matching the canbus one), then turned on a load, dropped to 9.46V and you can now see the 9.46 on top of the older 4digit value. 
Some sort of screen refresh issue?

[image]

V

stuartpittaway, 2021-12-28:
I've not seen that one, could it be an issue with the 5v line during pc hibernation?

virtuvas, 2021-12-29:
that is also my guess Stuart, just received a 20euro (☹️)  2.7A 5V powersupply and once I find a decent diode for D7 I'll try it.
btw, would a 1N4001 (or 1N4007) be OK? (also have a few 1N5363B lying around which I guess is not OK)

cheers

V.

virtuvas, 2021-12-30:
Closing:

with proper power supply on terminals providing 5V 2+A and usb cable connected PC waking from hybernation doesn't affect diyBMS screen. Interestingly Victron and all displays pickup battery bank updates a split second faster than the onboard screen. So from being idle, turning on a consumption rpi reports the .8-.9V dip in voltage before the tft onboard.

V.

closed 2021-12-30 by virtuvas