stuartpittaway/diyBMSv4ESP32 #76: On/Off delay for relays

YarLeo, 2021-12-30:
Hello Stuart,

I would like to suggest/request a new feature for rules / relays.
It would be nice to have an ON and OFF delay for each relay (below, where a selector for relay type is, place 2 more fields: ON-delay and OFF-delay). 
Personally I would like to have such feature for a selective disconnect of the battery: for safety reasons I prefer to have 2 devices, which in case of some failure (rules) would be able to disconnect a battery from load/chargers. 1st device - power relay, 2nd device - MCCB with shunt trip. With the same rule I wish to activate both of them with different relays with some delay in between, so that if 1st fail do disconnect the battery, the 2nd can save a day.

I believe this feature should not consume too much time, but can increase system safety. 
I hope you'll fine this suggestion useful.

Best regards,
Yaroslav

Jestergnet, 2022-02-14:
Great idea.

stuartpittaway, 2022-02-25:
Hello @YarLeo thanks for the suggestion.

To add a delay into the current ruleset would take quite a bit of work, but to clarify why do you need to delay between the relay firing?

If you have two safety devices, why not trigger both at the same time?  What benefit does having a delay give you in this scenario?

YarLeo, 2022-02-25:
Hi @stuartpittaway 

The main thing here is selectivity, so that there is a "normal" switching device and an "emergency" switching device. For example, in my particular situation, those are: a high current dc relay (remotely retrigable) as a "normal" switching device, and a MCCB with a shunt trip (can only be switched bac on manually on site) as an "emergency" device. Thus it would be nice first to switch off only relay, and only if it doesn't helped - the MCCB (the coil of shunt trip is powered after power relay, so if ralay works fine, the mccb would not be triggered).

The other thing about delays: it would be nice to have them, for example for overcurrent rules, or undervoltage rules, so that the rule is not triggered during sturtup of heavy load. etc.  

There are some more applications where this feature could be useful, like selective disconnect of loads/chargers, immediate/too fast restart protection, critical loads "last minute"power for data backup, etc. But I do not know if such thing are really necessary.

If it is not so easy to implement such on/off delays, for me it is also not that critical, I have at the moment a hardware timer on my MCCB. But thought it might be nice feature to have, which can add additional flexibility to the rules funtionality.

PS: Some more ideas about new rules:
- differential external temperature (to determine if some cell starts to heat up and prevent problems even faster then it reaches the external temperature high rule setpoint)
- differential voltage (for example to activate external active balancer, many people stile like to use them additionally, or gust to have additional message that something is wrong with some cell block)

Cycor, 2022-06-01:
I would also love a delay, especially for startup, now every time I reboot the controller, or it crashes, it triggers some rule, not sure with one yet, and the relay turns off all the power.

So in my case I would be happy with a rule based on controller uptime, like the midnight timer, so I can at least block the relay OFF signal

stuartpittaway, 2022-06-01:
@Cycor the controller shouldn't be crashing if you have the latest version of the software/code installed.

I can't prevent the relays from switching/triggering when the ESP is rebooted (or powered up) as the relay state - as the ESP isn't controlling the relays at that point (there is no code running).

Cycor, 2022-06-01:
@stuartpittaway The crash is probably a bad power supply or a bad connection, I still have to debug that, so you don't have to worry about that.

it's a pulse relay with the 2 states on output 3 and 4, so it doesn't do anything on it's own, the relay only triggers when I set rules so it is something the controller does, my fan on relay 1 also fires a second on startup and the only rule that has is module over temperature.

My controller is running the latest version
Compiled: 2022-05-13T08:58:50.073Z

Here is a video of the relay triggering on boot, the rule set is "Module over temperature (internal) ON"
https://www.youtube.com/shorts/yo6N15LfFko