stuartpittaway/diyBMSv4ESP32 #77: switching from English to Russian

Garkus98, 2022-01-01:
When switching from English to Russian in the panel on the newest firmware, it remains English although everything has been translated

Jestergnet, 2022-02-15:
I also set the language to Russian, but remained in English.

Jestergnet, 2022-02-16:
Причину нашел, нужно в файле ESPController/web_src/lang_ru.js найти:
$("#rt2").text("Правила обрабатываются от самого низкого приоритета до самого высокого(снизу вверх).Управляйте реле с помощью опций. Значение "X" означает, что вам все равно/оставьте значение, определенное правилами с более низким приоритетом.");
$("#rt3").text("Правила срабатывают, когда соответствующее значение соответствует или превышает значение "триггера". Правило будет отключено только тогда, когда значение затем передаст значение "сброс". Это может помочь предотвратить грохот реле и быстрое включение / выключение правил.");
$("#rt4").text("Правила "Таймер 1" и "Таймер 2" допускают работу по времени, это правило активно, когда достигнуто количество минут после полуночи, например, установка триггера "Таймер 1" на 495 и сброс до 555 включится в 8: 15 утра и выключится в 9: 15 утра. Это работает только при подключении к Интернету для регулярных обновлений времени.");

Заменить на:
$("#rt2").text("Правила обрабатываются от самого низкого приоритета до самого высокого(снизу вверх).Управляйте реле с помощью опций. Значение \"X\" означает, что вам все равно/оставьте значение, определенное правилами с более низким приоритетом.");
$("#rt3").text("Правила срабатывают, когда соответствующее значение соответствует или превышает значение \"триггера\". Правило будет отключено только тогда, когда значение затем передаст значение \"сброс\". Это может помочь предотвратить грохот реле и быстрое включение / выключение правил.");
$("#rt4").text("Правила \"Таймер 1\" и \"Таймер 2\" допускают работу по времени, это правило активно, когда достигнуто количество минут после полуночи, например, установка триггера \"Таймер 1\" на 495 и сброс до 555 включится в 8: 15 утра и выключится в 9: 15 утра. Это работает только при подключении к Интернету для регулярных обновлений времени.");

Garkus98, 2022-02-16:
так а ошибка то где? в каком именно слове/месте?

Garkus98, 2022-02-16:
исправление сейчас внесу спасибо

Jestergnet, 2022-02-16:
Не экранировали кавычки, НЕ правильный например:
$("#rt2").text("текст текст "X" текст");

Правильно:
$("#rt2").text("текст текст \"X\" текст");
или
$("#rt2").text("текст текст 'X' текст");

Garkus98, 2022-02-16:
ясно спасибо сейчас исправлю

closed 2022-02-16 by Garkus98