stuartpittaway/diyBMSv4ESP32 #79: Feature request: Flash ESP32 via Chrome Web Serial API

renne, 2022-01-18:
Hi,

currently OS-specific command-line tools or IDEs are necessary to flash diyBMS on the ESP32.
This makes the use of diyBMS impossible for non IT-savy users.

I suggest flashing the ESP32 via the [Chrome Web Serial API](https://codelabs.developers.google.com/codelabs/web-serial#0).
[Tasmota](https://tasmota.github.io/install/) and [PedalinoMini](https://alf45tar.github.io/PedalinoMini/installer) use the Chrome Web Serial API via the [ESP Web Tools](https://esphome.github.io/esp-web-tools/).

stuartpittaway, 2022-01-19:
Thanks for the idea, I've not seen that before - the https://www.improv-wifi.com/ also looks like a neat solution.

renne, 2022-01-20:
@stuartpittaway The ESP Web Tools use improv-wifi :smiley: