stuartpittaway/diyBMSv4ESP32 #81: SOC calibration at low battery stater

bauer1980, 2022-01-28:
Hello, the actual system (software) calibrates the SOC when the battery is full (100%). The problem is, that during the winter season the average battery is never full —> there is never a SOC calibration during winter season. So, the SOC is very wrong at this time of the year. Battery systems that are out of LIFEPO suffer a lot from this problem, because they have higher self discharging than NMC Lithium types of batteries. But nevertheless the SecondLife community has this problem also with NMC batteries. The higher the self discharging the bigger the SOC problem in winter.

Conclusion: I would recommend for this incredible DIYBMS to adapt the software for SOC calibration, so that the calibration is also executed at low battery state. E.g. REC BMS does it as well, top and bottom calibration and it works good.  

(The bigger the battery the less times it is full in winter times … !!!)

stuartpittaway, 2022-02-14:
Ok, so if you are not charging from the grid (overnight for example to topup) do the cells just get lower and lower?

How would you know what value of SOC to supply if the system at any point in time?  Is it just an educated guess?

stuartpittaway, 2022-02-14:
Ok, I think I'm following you.  So if we also needed a "bottom" reset, how would this be achieved?

Can we do this purely on voltage measurement, so if the pack voltage drops below  X volts we set the SOC to Zero (or 10% etc.)

bauer1980, 2022-02-16:
Yes. Ist purly by voltage measurement.

 

But, if you compare this feature by REC-BMS, there is one more very important functionality that is essential for safety, which has to do with this low voltage value:

 

1.	Value of CLOW: cell unter voltage discharge protection. 

If one cell reaches this level, the DCL (Discharge Current Level) that is sent by the CAN bus to the Victron System is set to 0.

 

2.	Value of CMIN: cell unter voltage protection switch off

But, lets say the CAN bus has a problem, the cable is defect, the value is wrong, the Victron system is dead, there is any high resistance short of the battery pack, so that the fuse is not interrupting. Than the battery is still discharging, although the DCL = 0. 

Than the first cell reaches the CMIN voltage. At this voltage, the BMS does switch off the high current switch to the battery to be safe of undervoltage.

 

This redundant safety feature is one of the most important and most other BMS’s do not offer this. This is a reason why I decided for the REC BMS at one installation.

 

Safety is the most important functionality. 

 

REC BMS SOC calibration, the calibration works really well – compared to only top calibration done by most standard systems – just from my experience.

 

 

 

Best regards,

Wolf 

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

bauer1980, 2022-02-16:
This is how REC-BMS is doing the calibration, this works quite well in my installation:

 

Low voltage calibration:

“When the lowest cell open-circuit-voltage is discharged bellow the set
threshold CLOW, maximum discharging current starts to decrease down to 0.02 C (5 % of cell’s
capacity in amperes). After decreasing, maximum allowed discharging current is set to 0 A. SOC is reset
to 3 % and discharging SOC hysteresis is set to 5 %. If the cell discharges below Minimum cell voltage
CMIN, BMS signals Error 2 and SOC is reset to 1 %.
When the lowest cell is discharged under 0.98 x CMIN for more than 30 seconds, BMS goes to sleep
to conserve energy. Manual restart of the BMS is required to wake it up.”

 

High voltage calibration:

“When the highest cell reaches the Balance start voltage setting, the charging current starts to ramp
down to 1.1 A until the last cell rises to the End of charge voltage. At that point the Maximum charging
voltage allowed is set to Number of cells x (End of charge voltage per cell – 0.5 x End of charge
hysteresis per cell). End of charge SOC hysteresis and End of charge cell voltage hysteresis is set to
prevent unwanted switching. SOC is calibrated to 100 % and power LED light is on. Charge optocoupler
is turned off.
Charging is stopped in case of system errors (see System Error Indication section). SOC is calibrated to
96 % when the maximum open circuit cell voltage rises above the 0.502 x (Balance start voltage +
Balance end voltage), minimum open circuit voltage above balance start voltage and system is in
charge regime.”

 

-- 
Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

stuartpittaway, 2023-02-15:
Going to close this one down, compared to the Victron SmartShunt the DIYBMS is aligned - I've also added the feature to the user interface to configure the SoC, which is useful to calibrate as needed.

closed 2023-02-15 by stuartpittaway