stuartpittaway/diyBMSv4ESP32 #82: enhancement: Victron CANBUS broadcasting all cell voltages/balance status

virtuvas, 2022-02-06:
hello,

over at https://github.com/Louisvdw/dbus-serialbattery there's a  pull request (just approved whilst I was typing!) for sending all cell voltage balance status as well as diff from highest to lowest cell (so Range in diyBMS speak) to VenusOS https://github.com/Louisvdw/dbus-serialbattery/pull/87

there's also some simple code to visualise them in Venus. I did the easy bit (vis in Venus) which for my 8S would look like this:

[image]

and then this opens another screen:
[image]
(obvs easy to support more...)

If something like this is implemented, the ones of us that have Victron kit wont need the touch screen to view all the (less) vital statistics of our lifepo4 banks.

Now, what they have done is sending to dbus some extra paths :
/Voltages/Cell#
/Balances/Cell#
/Voltages/Sum
/Voltages/Diff
on top of all the other "normal" paths like all the /System/* etc.

I went through diyBMS code trying to find where the real paths are declared so that I can add the necessary code and see how I can get that done, but failed miserably. I can see the defs of the messages sent, the info included etc, but cannot understand how for example the actual /System/MinCellVoltage  is actually sent and what's involved in doing so. 
If you can point me in the right direction I could have a go at it!

cheers

V.

stuartpittaway, 2022-02-06:
I used the official documentation provided by Victron for the API integration. Have they updated the specs?

Is having every cell voltage actually useful? Some people have 80+ cells connected!!

virtuvas, 2022-02-06:
thanks for the quick response!

I doubt they've updated the specs, I feel they are just pumping extra data which are received by the GUI and shown.

TBH Stuart, no boater will have 80+cells onboard, I think that's more a static powerwall sort of application with 18650 which I think wont find their way in boats which are typically 12-24V.
Most boaters have 4S (under 40-45ft) or 8S (larger than 40ft usually) and less space to integrate yet one more screen, hence my thinking.

back to my Q, for example, looking at victron_message_373() I see that you create this  ```struct data373``` which includes the 4 basic info of the message, furnish it with values and send it using ```send_canbus_message(0x373, (uint8_t *)&data, sizeof(data373));```
fine, however this in victronOS terms becomes:
/System/MinVoltageCellID
/System/MaxVoltageCellID
and the equiv for temps

where's this "conversion" taking place? that's mainly my question in case I've missed something in diyBMS codebase.
I maybe talking bull and you're just sending message 373 and VictronOS deals with the values itself creating the paths as above, that I don't know.

It seems that what the guys at https://github.com/Louisvdw/dbus-serialbattery  are doing is building and sending new paths like /Voltages/Diff or whatever.

cheers

V.

PS. also asking another (simpler) thing in the forum main thread 😀

stuartpittaway, 2022-02-06:
I don't send any victron paths, it's all mapped by the victron side of things

virtuvas, 2022-02-06:
I see, is that API doc available? got a pointer to it? searched a bit, downloaded a couple of docs, doesn't seem I got it

virtuvas, 2022-02-06:
looks like I'm getting somewhere it's 0x381: battery instance, cell instance, cell voltage. BTW, why are the values in 0.01V and not 0.001V?

stuartpittaway, 2022-02-06:
I was given a copy but think it's under a non disclosure agreement. The source code you are looking at has most of the info in it, particularly the comments. I think I use most of the APIs available

virtuvas, 2022-02-07:
OK, before I get any further, can you please check if the NDA doc you have includes any structure that similarly to 381 ids cells that are currently balanced?
if not whole exercise is rather pointless and I can stick with what you've got - in the screen I showed on first post cell background turns red when cell is balancing but unless I can get that info over it cannot work.
Point is the other data paths I was looking at, are sent via RS485 and not canbus. Obvs that gives different levels of flexibility but I'm not interested in getting into that.

OTOH, I see that 0x303 (number of charge cycles) and 0x305 (Cumulative number of Amp hours drawn from the battery) and 0x310 (The amount of discharged energy) are there and could be an interesting addition that will help in the long run to determine the health of the bank. What do you think?

cheers
V.

stuartpittaway, 2022-02-07:
I've used all the commands that I've had documentation for - there was only the min/max cell voltage as part of the can messages, so I've no documentation on how to pass all the cell voltages.

virtuvas, 2022-02-07:
0x0381 Cell Voltage [read only]
un8 battery The battery instance
un8 cell The cell instance
un16 voltage Cell voltage in 0.01V
0xFFFF: Not available

is that not in the docs you've got? It's included in the VE.Can registers Victron Registers in NMEA 2000 – version 20 I got yesterday off Victron site (with email registration so not a NDA I guess)
Tested it a bit, but struggling on getting this datastructure right.

Similarly,  section 1.16 (p35) shows all history values (0x3**) some of which seem useful

apologies, I feel I'm being a nuisance...

stuartpittaway, 2022-02-07:
Not a nuisance!

I don't have that in my documentation, it would be easy to implement, however for people with large battery installations, its going to put a lot of CPU strain on the ESP32, so that may cause other issues, if I have to send 80+ messages every few seconds, compared to the 4 or 5 that are currently transmitted.

stuartpittaway, 2022-02-07:
Indeed, just checking and CAN message 0x381 relates to the "Serial number" of the battery bank !

I'm using the CAN-BMS protocols, the document I have is dated 17/04/2021 - perhaps the data you have is more for generic CAN messages?

virtuvas, 2022-02-07:
indeed and how can comms to N2K by the looks of it.
OK, so this idea is scrapped, and yes, if it was to work it would be for 4-7-8S systems only...

final Q (for now 😀 ) is there anywhere on the system a log of the cumulative Ah used by the bank?
And if so, are such data written on the SD card? As I posted a few messages above yesterday, would be useful to have some age/wear related metrics for our banks.
I have an SD on but haven't yet shutdown the system to remove it to check what's written there.

stuartpittaway, 2022-02-07:
You can download the log files from the web interface in diybms - no need to remove the SD card.

There should be a log file which starts with the filename "modbus" and has this information in it.

DateTime,valid,voltage,current,mAhIn,mAhOut,power,temperature,shuntmV,relayState

The amp hour counters are automatically reset when the shunt determines the state of charge to be at 100% - which is configured using the "tail current" settings.

closed 2022-02-12 by virtuvas