stuartpittaway/diyBMSv4ESP32 #91: Update Feb 2022 #90 leads to controller continuous reboots

virtuvas, 2022-02-27:
hello,

tried the latest update, uploaded fine on the ESP32, only it starts a never ending loop of reboots. The following is the capture of the terminal:

```

                _          __
    _|  o      |_)  |\/|  (_
   (_|  |  \/  |_)  |  |  __)
           /
[I][main.cpp:3091] setup(): CONTROLLER - ver:LocalCompile compiled 2022-02-27T12:44:22.547Z
[I][main.cpp:3096] setup(): ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
[I][HAL_ESP32.cpp:232] ConfigureI2C(): Configure I2C
[I][HAL_ESP32.cpp:291] ConfigureI2C(): Found TCA9534A
[I][HAL_ESP32.cpp:321] ConfigureI2C(): Found TCA6408
[D][HAL_ESP32.h:288] TouchScreenUpdate(): Touch = touch=0 pressure=128 x=0 y=3171
[I][main.cpp:3111] setup(): TFT screen is INSTALLED
[I][main.cpp:946] SetControllerState(): ** Controller changed state from Unknown to PowerUp **
[I][main.cpp:3133] setup(): LITTLEFS mounted, totalBytes=589824, usedBytes=98304
[I][main.cpp:254] mountSDCard(): Mounting SD card
[I][main.cpp:267] mountSDCard(): SD card available
[I][HAL_ESP32.cpp:140] ConfigureCAN(): CAN driver installed.  Filter=1621098496 Mask=6291455
[I][HAL_ESP32.cpp:150] ConfigureCAN(): CAN driver started
[D][settings.cpp:20] ReadConfig(): ReadConfig diybms
[D][settings.cpp:42] ReadConfig(): checksum verified
[I][main.cpp:3157] setup(): 
[D][main.cpp:213] SetupRS485(): Setup RS485
E (515) uart: uart_set_stop_bits(142): stop bit error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400912a8
file: "src/main.cpp" line 232
func: void SetupRS485()
expression: uart_param_config(rs485_uart_num, &uart_config)

ELF file SHA256: 0000000000000000

Backtrace: 0x40090d78:0x3ffb1eb0 0x400912ab:0x3ffb1ed0 0x400e0bc2:0x3ffb1ef0 0x400e4786:0x3ffb1f30 0x40100a96:0x3ffb1fb0 0x40092ce2:0x3ffb1fd0
  #0  0x40090d78:0x3ffb1eb0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:695
  #1  0x400912ab:0x3ffb1ed0 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:721
  #2  0x400e0bc2:0x3ffb1ef0 in SetupRS485() at .pio/libdeps/esp32-devkitc/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:268
  #3  0x400e4786:0x3ffb1f30 in setup() at .pio/libdeps/esp32-devkitc/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:268
  #4  0x40100a96:0x3ffb1fb0 in loopTask(void*) at C:\Users\Vassilis\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:18
  #5  0x40092ce2:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
```

not got a shunt connected (or anything on RS485 for that mater. Still waiting for the final chip to arrive from China...

cheers

V.

stuartpittaway, 2022-02-27:
Ok, I think the configuration settings are messed up.

Can you clear the flash on the esp32 completely and then try programming again?

virtuvas, 2022-02-27:
of course if you give me some hints on how that can be done!
Mind I'm rather confused with VSC and Platformio (after 5+ years working with Arduino IDE) so instructions must be clear

V.

virtuvas, 2022-02-27:
genuine Q
can settings be messed up and latest release gets into this rebooting loop whereas reloading on VSC the previous code version, compiling and reloading it to the ESP32, system works fine?

stuartpittaway, 2022-02-28:
Yes and no!

I would have expected the configuration to have been reset/cleared automatically as soon as you installed the latest version of the controller code.  Thats what normally happens when I introduce new configuration data (and has happened on many previous releases).

The code is halting on setting up the RS485, "uart_set_stop_bits(142): stop bit error".

The stop bit configuration is stored in the configuration data, hence why I think the configuration data my be incorrect.

To completely clear out the flash, try this...

```
esptool.py --chip esp32 p com7 erase_flash
```

This was taken from here.
https://rntlab.com/question/how-perform-reset-factory-esp32/

Then follow the instructions from here [https://github.com/stuartpittaway/diyBMSv4ESP32/blob/master/ProgrammingHardware.md] on how to program the chip.

virtuvas, 2022-03-01:
thanks Stuart,

as if it was so simple...
anyway, good thing I have a spare ESP32 :-)
tried the python line above after downloading the esptool, tried to run it off an admin command line in win10, not much luck, so since I'm struggling with time atm, I just left it aside.

Anyway, spare ESP had some code from before xmas. Uploaded new code, no issues - other than having to reconfigure everything (except wifi!). So, couple of questions:
Q1: I have a .json backup config file on my sdcard. Is there some way to upload it when you swap ESPs? For what's worth, it may have tried but the data there were pre-final install with an old 4S vs the now 8S setup so wouldn't help anyway, asking for next time and or/if ESP crashes mid summer with me onboard!

Q2: interpacket gap on my 8S lifepo4 works fine at 5k baud throughout the range from 2k to 9k (ok, at 9k gaps between flashing sessions are much longer) What I noticed is that at large values, in random (say once every 4-6 flashing sessions) I get some odd flashing: module#0 then module#1 and then module#6 then a gap and starts again 0-7. Means something? Wont matter to me as it looks like I'll settle for something close to 2.5-3k although even then there's no consistency in the flashing patterns (will do two consecutive loops, then a bit of a gap then maybe another 2 then 6loops, then relax and start again)

Influx intervals work absolutely fine, so will close issue #71 . Wondering if random (rare) reboots are addressed with this. I'm waiting for the last chip for my shunt, so will leave the bank idle (esp and a rpi only running out of them) and will keep on logging and checking.

diyBMS gets better in each update, congrats!

V.

stuartpittaway, 2022-03-01:
Glad you got it working on a fresh ESP32, not sure what happened on the old one, this is the first report I've been given of such behaviour.

Although you can backup the configuration, at the moment there isn't a way to restore it (yet - work in progress)

On Q2, if you find a "inter packet gap" which works well in your situation without CRC or OOS errors, then thats great, just leave it at that.  Don't worry about the flashing LED light sequence, different requests are made to the modules at different times (like the error count values) so sounds perfectly normal.

virtuvas, 2022-03-01:
thanks,
will try wiping the old ESP alone (as in not plugged into the controller) when I get some time later on.

re Q2, do I remember right that the original setup (pre #90 ) was leading to a circa 6k gap on a 8S system (4.4modules)?
never had issues with OOS and CRC, v.rarely and after days running I might get a couple (but usually I was tinkering with something in the meantime so never too sure of the cause). I'll leave it at 3k for a couple of days and see!

Ah, another issue I notice (wont open a new for it) is that screen is for some odd reason "taller" and on a full HD monitor I struggle to get the last line in unless I get the chrome window almost full height - don't want it! So something's changed in the layout? any reason? info is still the same.

cheers
V.

stuartpittaway, 2022-03-01:
I don't think I've changed anything on the screen layout - has the voltage scale on the graph changed after the settings were cleared?

virtuvas, 2022-03-01:
well it did, but reset it to 3-3.5 as before. 
It seems to me that the top gap between Voltage/Range and cell bars below is too large.
[image]

I may be talking bull and the tab I had on my browser before on the previous ESP32 was at 90% vs 100% 
switched to 90% and looks fine now but the point that there's plenty of unused space on the top stands.

cheers
V.

stuartpittaway, 2022-03-01:
Not sure whats changed to be honest, don't think I've done anything on that one.

virtuvas, 2022-03-01:
no worries, probably me!
If you don't mind i'll leave the issue open for a few days until I figure out how to sort out the other ESP32 (and to see if anyone gets similar behaviour)

cheers
V.

stuartpittaway, 2022-03-01:
Just noticed there is an erase flash option on VSCODE/PlatformIO if that helps?

[image]

virtuvas, 2022-03-01:
good find, now cannot run any code on the ESP :-)
Upgraded pip as recommended:
```
WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
```
then rebooted/powered off/on and erased flash (a couple of times for good measure).
plugged it on the controller board 3led was red flashing blue.
Erased flash again plugged in, 3led was permanently on blue.

Then uploaded the code, now 3led is again red and flashes blue, screen is dead and no comms.
ESP32 red led is constantly on, terminal message is that it halted on littleFS failure:

```
                _          __
    _|  o      |_)  |\/|  (_
   (_|  |  \/  |_)  |  |  __)
           /
[I][main.cpp:3091] setup(): CONTROLLER - ver:LocalCompile compiled 2022-03-01T13:08:38.601Z 
[I][main.cpp:3096] setup(): ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
[I][HAL_ESP32.cpp:232] ConfigureI2C(): Configure I2C
[I][HAL_ESP32.cpp:291] ConfigureI2C(): Found TCA9534A
[I][HAL_ESP32.cpp:321] ConfigureI2C(): Found TCA6408
[D][HAL_ESP32.h:288] TouchScreenUpdate(): Touch = touch=0 pressure=128 x=0 y=3176
[I][main.cpp:3111] setup(): TFT screen is INSTALLED
[I][main.cpp:946] SetControllerState(): ** Controller changed state from Unknown to PowerUp 
**
.pio\libdeps\esp32-devkitc\LittleFS_esp32\src\lfs.c:1076:error: Corrupted dir pair at {0x0, 
0x1}
E (69) esp_littlefs: mount failed,  (-84)
E (73) esp_littlefs: Failed to initialize LittleFS
[E][LITTLEFS.cpp:90] begin(): Mounting LITTLEFS failed! Error: -1
[E][main.cpp:3127] setup(): LITTLEFS mount failed, did you upload file system image?        
[E][HAL_ESP32.h:181] Halt(): SYSTEM HALTED
```
looks like it's well and truly wiped, now how do we load it again?

V.

atanisoft, 2022-03-01:
> now how do we load it again?

The latest output looks like you need to upload the filesystem. In PlatformIO you can use the "Upload Filesystem Image" option (two above erase flash, not the OTA one). After that you should be able to reboot and have a working spare again.

stuartpittaway, 2022-03-01:
As @atanisoft said!

Or just follow the instructions on GITHUB https://github.com/stuartpittaway/diyBMSv4ESP32/blob/master/ProgrammingHardware.md

virtuvas, 2022-03-01:
👍 
that was easy and quick!
cheers Mike

So to recap if ESP32 gets into a rebooting loop, you can use platformIO to Erase Flash and then Upload Filesystem Image

V.

virtuvas, 2022-03-01:
> 
> On this error, the LED should be flashing white and red

OK then, that's my eyes, I see that as a heavily bluish tint bright thing that I'd rather call blue.

so closing this issue!

thanks

closed 2022-03-01 by virtuvas