stuartpittaway/diyBMSv4ESP32 #92: occasionally bank is reported as 0V on influx. Individual cell values are correct.

virtuvas, 2022-03-02:
hi,

have my 8S bank (304Ah) charged to circa 70% and slowly discharging (running the controller board as well as a Rpi4 and the can controller) at 0.2A (according to my amp clamp - current board still missing chips...)
Noticed (2-3 times already over the last couple of weeks at different states of charge) that my influx data drops for a min or less to 0V from the 26.whatever. Cell values are fine throughout. I'm attaching a screen cap from grafana:

[image]

Influx is sending every 30sec. All other data look fine and consistent, it's only overall bank voltage messed up.

I wonder if that's a glitch in the influxdb code or if it's reported generally as such throughout the system. There's no pattern to it whatsoever. Can dig the last month log in influx and see other occasions but don't think that will help much.

Haven't got sd card logging on atm, wonder if that would pick something up!

ideas?

V.

stuartpittaway, 2022-03-03:
I think this may be a timing issue.  As the ESP32 is dual core CPU, its possible that the influx request is sent at the same time as the data is being updated from the modules, resulting in zero or a wrong number. It would also match the "random" pattern.

Would be useful to compare with the SD card logs just to check nothing else is going on.

virtuvas, 2022-03-03:
thanks, started logging at 15sec intervals, that's going to be more than enough to capture that. 
Bearing in mind that the 30sec before and 30sec after value is unaffected, I'd consider that as an influxDB timing issue. If it was indeed something else, I'd expect before and after values to be slightly affected, am I right?
anyway, I'll leave it open and report when it happens again.

In case it is indeed timing, would it make sense to sent previous known good value? Would that be possible I wonder!

cheers
V.

virtuvas, 2022-03-16:
two weeks, not happened again, so closing thread for now. 
If and when it happens, I'll update with logs and more info!

closed 2022-03-16 by virtuvas