stuartpittaway/diyBMSv4ESP32 #98: influx output additions

virtuvas, 2022-03-16:
stuart not an issue, an idea for an enhancement,

to be able to better comprehend the behaviour of batteries/modules as they get into balancing mode I've added/altered one line in the influx.cpp file. Wonder if you'd be interested to get it into the "normal" code. 
No big deal to edit and add it myself, but seems to me that it having a % of balancing is much more useful compared to the boolean balance/no balance that we currently get, and makes visualisation in grafana possible:
[circa line 103]
```
poststring = poststring + "cells," + "cell=" + String(bank) + "_" + String(module) + " v=" + String((float)cmi[i].voltagemV / 1000.0, 3) + ",i=" + String(cmi[i].internalTemp) + "i" + ",e=" + String(cmi[i].externalTemp) + "i" + /* ",b=" + (cmi[i].inBypass ? String("true") : String("false")) + */ ",p=" + String(cmi[i].PWMValue / 255.0 * 100) + "\n";
```

fwiw, I also added a couple of lines after the while [circa line 110] for debuting, reporting overall uptime, canbus sent/receive traffic and the only one that may be useful is the range in mV:

```
poststring = poststring + "bank" + " vr=" + String(rules.VoltageRangeInBank(0)) + ",v=" +  String((float)rules.highestPackVoltage / 1000.0, 3) + "\n";
```
[bit of a hack as I only have one bank, and needs to be modded for more banks]. Needs:
```
#include "victron_canbus.h"
``` 
to work.
Feel free to ignore both :-)

cheers

V.

stuartpittaway, 2022-03-16:
Far from ignore it, I'm happy to take suggestions or even better pull requests with the working code :-)

The influx code has been re-written in the NOV2021 branch - if you care to try/take a look I suspect the code you have supplied will need modification to make it work going forward.

The NOV2021 branch is work in progress.

virtuvas, 2022-03-16:
I'll have a go tomorrow, so you mean I just download NOV2021, compile and test it?
sounds too easy :-)

now pull requests is pushing it as I've only tried it once in Signal K and I made a mess (I think)
OK, more tomorrow.

V.

closed 2022-03-17 by virtuvas