stuartpittaway/diyBMSv4ESP32 #100: NOV2021 branch influxDB 1.8 not working and controller reboots

virtuvas, 2022-03-17:
stuart,

your code, as is today, AP with passwd (changed my home wifi net and added WPA2 passwd) and with the influxDB settings that work with the master branch exhibit the following two (must be related) issues:

1. wont send anything to an influx v1.8 DB that the master branch does fine
2. reboots the controller every 5-10mins

extract below:

```
expression: esp_http_client_perform(http_client)
I (344875) diybms-webreq: API call: monitor2
D (345939) diybms-tft: Update TFT display
D (348410) diybms-tft: Update TFT display
I (349585) diybms-mqtt: MQTT Payload for cell data
I (351042) diybms-webreq: API call: monitor2
D (351413) diybms-tft: Update TFT display
D (354413) diybms-tft: Update TFT display
I (354590) diybms-mqtt: MQTT Payload for cell data
I (354594) diybms: Influx task

assert failed: http_utils_assign_string http_utils.c:49 (old_str)

Backtrace:0x40084491:0x3ffda8500x4008d935:0x3ffda870 0x40093e59:0x3ffda890 0x40118c83:0x3ffda9c0 0x40118a8f:0x3ffda9e0 0x401172b2:0x3ffdaa10 0x400d50c9:0x3ffdaa30 0x400d5500:0x3ffdabc0 
  #0  0x40084491:0x3ffda8500 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[��mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (24) diybms: 

               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:LocalCompile compiled 2022-03-17T17:58:13.703Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (36) diybms-hal: Configure I2C
I (38) diybms-hal: Scanning i2c bus
I (44) diybms-hal: Found i2c device at address 0x20
I (47) diybms-hal: Found i2c device at address 0x38
I (56) diybms-hal: TCA6416A not fitted, assume v4.2 board
I (58) diybms-hal: Found TCA9534A
I (59) diybms-hal: Found TCA6408
D (61) diybms: Configure VSPI
I (64) diybms: ** Controller changed state from Unknown to PowerUp **
D (70) diybms: Configure VSPI
D (73) diybms-tft: Configure TFT display
[    80][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (82) gpio: gpio_set_level(226): GPIO output gpio_num error
[    92][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (93) gpio: gpio_set_level(226): GPIO output gpio_num error
[   103][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected

Press SPACE BAR to enter console WiFi configuration....

....................

No key press detected
I (5461) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (5462) diybms-hal: CAN driver started
I (5464) diybms: LittleFS mounted, total=589824, used=12288
I (5467) diybms: Mounting SD card
I (5480) diybms: SD card available
[  5484][D][settings.cpp:20] ReadConfig(): ReadConfig diybmswifi
[  5488][D][settings.cpp:42] ReadConfig(): checksum verified
I (5484) diybms: Checking for /diybms/wifi.json
D (5496) diybms: Found file /diybms/wifi.json
D (5508) diybms: Deserialized /diybms/wifi.json
I (5509) diybms: Wifi JSON config is identical, ignoring
I (5509) diybms: Fetch config
[  5514][D][settings.cpp:20] ReadConfig(): ReadConfig diybms
[  5525][D][settings.cpp:42] ReadConfig(): checksum verified
D (5522) diybms: Setup RS485
D (5524) diybms: Configure RS485
I (5528) diybms: ** Controller changed state from PowerUp to Stabilizing **
D (5533) diybms: starting wifi_init_sta
D (5592) diybms: tca6408_isr
I (5650) diybms: Hostname: DIYBMS-004CE8AC
D (5651) diybms: wifi_init_sta finished
I (5651) diybms: TFT screen is INSTALLED
D (5652) diybms: total_free_byte=159052 total_allocated_byte=140232 largest_free_blk=110580 min_free_byte=158524 alloc_blk=341 free_blk=4 total_blk=345
I (13280) diybms: got ip:192.168.178.101
I (13281) diybms: Request time from time.google.com
I (13306) diybms: You can access DIYBMS interface at http://DIYBMS-004CE8AC.local or http://192.168.178.101
W (13405) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (15378) diybms-rules: Set error 4
I (15379) diybms: Active errors=1
I (15381) diybms-tft: Wake up screen
D (15414) diybms-tft: Update TFT display
W (15455) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (16240) diybms-tft: Update TFT display
I (17380) diybms-mqtt: MQTT Payload for cell data
W (17492) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (18379) diybms: ** Controller changed state from Stabilizing to Running **
I (18872) diybms: Task 1
D (19240) diybms-tft: Update TFT display
W (19524) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
W (21556) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (22243) diybms-tft: Update TFT display
I (22389) diybms-mqtt: MQTT Payload for cell data
I (22393) diybms: Influx task
I (22564) diybms-influxdb: HTTP Client Connected!
I (22705) diybms-influxdb: HTTP Client finished
W (23602) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (25243) diybms-tft: Update TFT display
I (25372) diybms: Task 2
W (25652) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (27497) diybms: total_free_byte=137376 total_allocated_byte=160108 largest_free_blk=92148 min_free_byte=131464 alloc_blk=491 free_blk=9 total_blk=500
W (27697) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (27708) diybms-mqtt: MQTT Payload for cell data
W (29736) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
W (31772) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (31872) diybms: Task 3, s=0 e=7
D (32253) diybms-tft: Update TFT display
I (32718) diybms-mqtt: MQTT Payload for cell data
I (32722) diybms: Influx task
I (32777) diybms-influxdb: HTTP Client Connected!
I (32925) diybms-influxdb: HTTP Client finished
D (33654) diybms-tft: Update TFT display
W (33816) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
W (35862) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (37162) diybms-tft: Update TFT display
W (37899) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (37926) diybms-mqtt: MQTT Payload for cell data
I (37930) diybms-mqtt: General status payload
I (37931) diybms-mqtt: Bank(0) status payload
I (37931) diybms-mqtt: Rule status payload
I (37932) diybms-mqtt: Outputs status payload
D (38562) diybms-tft: Update TFT display
W (39941) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (40243) diybms-tft: Update TFT display
W (41982) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (42483) diybms: Open log /data_20220317.csv
D (42498) diybms: total_free_byte=132400 total_allocated_byte=164424 largest_free_blk=88052 min_free_byte=127088 alloc_blk=546 free_blk=10 total_blk=556
D (42502) diybms: Wrote to SD log
I (42940) diybms-mqtt: MQTT Payload for cell data
I (42944) diybms: Influx task
I (43001) diybms-influxdb: HTTP Client Connected!
I (43152) diybms-influxdb: HTTP Client finished
D (43243) diybms-tft: Update TFT display
W (44017) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (44872) diybms: Task 1
W (46056) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (46432) diybms-tft: Update TFT display
W (48101) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (48152) diybms-mqtt: MQTT Payload for cell data
D (49247) diybms-tft: Update TFT display
W (50148) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
I (51373) diybms: Task 2
W (52188) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (52248) diybms-tft: Update TFT display
I (53167) diybms-mqtt: MQTT Payload for cell data
I (53171) diybms: Influx task
I (53225) diybms-influxdb: HTTP Client Connected!
I (53391) diybms-influxdb: HTTP Client finished
W (54246) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (55247) diybms-tft: Update TFT display
W (56286) diybms-webfuncs: Incorrect cookie received GmoI#TYCApQage6*Zlfg
D (57499) diybms: total_free_byte=127196 total_allocated_byte=168656 largest_free_blk=81908 min_free_byte=121532 alloc_blk=627 free_blk=15 total_blk=642
I (57873) diybms: Task 3, s=0 e=7
D (58250) diybms-tft: Update TFT display
```

if I disable influxdb, system seems more stable (got half an hour out if it but will have to leave it overnight)

V.

stuartpittaway, 2022-03-17:
Ok, thanks for testing.  This change wasn't done my me, and I've not tested it yet - all pre-release stuff as I mentioned.

The controller code should be stable (except for this bit!) I've been running a ESP32 for 24hrs now without a reboot, and this is with MQTT and logging enabled.

virtuvas, 2022-03-17:
np stuart,
wonder how I could help testing where it fails, seems like nothing reaches the work server.
maybe start from the basics and see if anyone has tried this code with v2.0 influxDB.

ah, was also meant to ask why it prints I lines with MQTT when I've got it disabled (never have enabled it)

V.

atanisoft, 2022-03-17:
@virtuvas Can you decode the backtrace? It looks likely that the http client received some data that it didn't expect.

virtuvas, 2022-03-17:
Mike, that's above my paygrade I'm afraid unless you can help explain how!

just reuploaded code, enabled influxdb and got a reboot within two mins:

```
I (65080) diybms-mqtt: MQTT Payload for cell data
I (65084) diybms-mqtt: General status payload
I (65085) diybms-mqtt: Bank(0) status payload
I (65085) diybms-mqtt: Rule status payload
I (65086) diybms-mqtt: Outputs status payload
I (65090) diybms: Influx task
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (period)
Core  0 register dump:
PC      : 0x4008f5e9  PS      : 0x00060036  A0      : 0x4008da4b  A1      : 0x3ffda280  
A2      : 0x3ffc706c  A3      : 0x3ffc724c  A4      : 0x3ffc724c  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc7074  A9      : 0x3ffda260  
A10     : 0x3ffdac5c  A11     : 0x3ffdac5c  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x3ffdac54  A15     : 0x80000001  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008b231  LEND    : 0x4008b241  LCOUNT  : 0xfffffffd

Backtrace:0x4008f5e6:0x3ffda2800x4008da48:0x3ffda2b0 0x4008d9f8:0xa5a5a5a5  |<-CORRUPTED
  #0  0x4008f5e6:0x3ffda2800 in taskSelectHighestPriorityTaskSMP at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3436
      (inlined by) vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3519
```

atanisoft, 2022-03-17:
Looks like it is simply running out of stack space, increase the `2500` in this line: https://github.com/stuartpittaway/diyBMSv4ESP32/blob/0d8592ee414f4064d65b61f969a3f983265efbbb/ESPController/src/main.cpp#L3258 to `3000` and see if that helps.

virtuvas, 2022-03-17:
not much of a difference Mike:

```
E (136007) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (136008) TRANSPORT_BASE: Failed to open a new connection: 32770
E (136009) HTTP_CLIENT: Connection failed, sock < 0
E (136014) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (136291) diybms-tft: Update TFT display
I (139041) diybms-tft: TFT switched off
I (139399) diybms-webreq: API call: monitor2
D (139800) diybms-tft: Update TFT display
I (141043) diybms-mqtt: MQTT Payload for cell data
I (141051) diybms-mqtt: General status payload
I (141052) diybms-mqtt: Bank(0) status payload
I (141053) diybms-mqtt: Rule status payload
I (141054) diybms-mqtt: Outputs status payload
I (141057) diybms: Influx task
E (141073) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (141074) TRANSPORT_BASE: Failed to open a new connection: 32770
E (141075) HTTP_CLIENT: Connection failed, sock < 0
E (141080) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (142296) diybms-tft: Update TFT display
I (143491) diybms-webreq: API call: monitor2
D (145296) diybms-tft: Update TFT display
I (146108) diybms-mqtt: MQTT Payload for cell data
I (146116) diybms: Influx task
E (146130) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (146132) TRANSPORT_BASE: Failed to open a new connection: 32770
E (146132) HTTP_CLIENT: Connection failed, sock < 0
E (146137) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (147535) diybms: total_free_byte=71348 total_allocated_byte=214844 largest_free_blk=40948 min_free_byte=67720 alloc_blk=1432 free_blk=39 total_blk=1471
I (147593) diybms-webreq: API call: monitor2
D (148298) diybms-tft: Update TFT display
I (148902) diybms: Task 1
I (151161) diybms-mqtt: MQTT Payload for cell data
I (151169) diybms: Influx task
E (151183) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (151184) TRANSPORT_BASE: Failed to open a new connection: 32770
E (151185) HTTP_CLIENT: Connection failed, sock < 0
E (151190) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (151299) diybms-tft: Update TFT display
I (151688) diybms-webreq: API call: monitor2
D (154301) diybms-tft: Update TFT display
I (155402) diybms: Task 2
I (155785) diybms-webreq: API call: monitor2
I (156218) diybms-mqtt: MQTT Payload for cell data
I (156225) diybms: Influx task
E (156240) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (156241) TRANSPORT_BASE: Failed to open a new connection: 32770
E (156242) HTTP_CLIENT: Connection failed, sock < 0
E (156247) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
[... 185 lines cut ...]
I (15389) diybms-rules: Set error 4
I (15390) diybms: Active errors=1
I (15392) diybms-tft: Wake up screen
D (15425) diybms-tft: Update TFT display
D (16251) diybms-tft: Update TFT display
W (16295) diybms-webfuncs: Incorrect cookie received fq9l8hieKfo2883*wTgJ
I (17391) diybms-mqtt: MQTT Payload for cell data
W (18333) diybms-webfuncs: Incorrect cookie received fq9l8hieKfo2883*wTgJ
I (18390) diybms: ** Controller changed state from Stabilizing to Running **
I (18883) diybms: Task 1
D (19253) diybms-tft: Update TFT display
W (20375) diybms-webfuncs: Incorrect cookie received fq9l8hieKfo2883*wTgJ
D (22254) diybms-tft: Update TFT display
I (22397) diybms-mqtt: MQTT Payload for cell data
I (22410) diybms: Influx task
W (22423) diybms-webfuncs: Incorrect cookie received fq9l8hieKfo2883*wTgJ
I (22740) diybms-influxdb: HTTP Client Connected!
I (22941) diybms-influxdb: HTTP Client finished
W (24473) diybms-webfuncs: Incorrect cookie received fq9l8hieKfo2883*wTgJ
```

http client connection failed,  client error, etc.
managed 4mins before reboot (doubt it is related seems to be a randomness in these reboots

V.

virtuvas, 2022-03-17:
and another go with 5000 instead of 2500. Just  shows when it starts turning sour:

```
D (57559) diybms: total_free_byte=114460 total_allocated_byte=179616 largest_free_blk=67572 min_free_byte=108024 alloc_blk=775 free_blk=17 total_blk=792
I (57932) diybms: Task 3, s=0 e=7
D (58312) diybms-tft: Update TFT display
I (59637) diybms-webreq: API call: monitor2
I (59734) diybms-mqtt: MQTT Payload for cell data
I (59738) diybms: Influx task
I (59831) diybms-influxdb: HTTP Client Connected!
I (60064) diybms-influxdb: HTTP Client finished
D (61812) diybms-tft: Update TFT display
I (63736) diybms-webreq: API call: monitor2
D (64310) diybms-tft: Update TFT display
I (65065) diybms-mqtt: MQTT Payload for cell data
I (65069) diybms-mqtt: General status payload
I (65070) diybms-mqtt: Bank(0) status payload
I (65070) diybms-mqtt: Rule status payload
I (65071) diybms-mqtt: Outputs status payload
I (65075) diybms: Influx task
E (65084) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (65085) TRANSPORT_BASE: Failed to open a new connection: 32770
E (65090) HTTP_CLIENT: Connection failed, sock < 0
E (65094) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (67310) diybms-tft: Update TFT display
I (67872) diybms-webreq: API call: monitor2
I (70118) diybms-mqtt: MQTT Payload for cell data
I (70122) diybms: Influx task
E (70128) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (70129) TRANSPORT_BASE: Failed to open a new connection: 32770
E (70130) HTTP_CLIENT: Connection failed, sock < 0
E (70135) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (70312) diybms-tft: Update TFT display
I (70932) diybms: Task 1
I (71976) diybms-webreq: API call: monitor2
D (72560) diybms: total_free_byte=106708 total_allocated_byte=185964 largest_free_blk=60404 min_free_byte=103292 alloc_blk=892 free_blk=18 total_blk=910
D (72655) diybms: Open log /data_20220317.csv
D (72669) diybms: Wrote to SD log
D (73313) diybms-tft: Update TFT display
I (75158) diybms-mqtt: MQTT Payload for cell data
I (75162) diybms: Influx task
E (75168) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (75169) TRANSPORT_BASE: Failed to open a new connection: 32770
E (75170) HTTP_CLIENT: Connection failed, sock < 0
E (75175) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 202
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
I (76083) diybms-webreq: API call: monitor2
```

atanisoft, 2022-03-17:
@virtuvas is your InfluxDB connection using http or https? It looks like it is using https and that it has pushed data on the 5000 stack size once but failed on the next push.

After https://github.com/stuartpittaway/diyBMSv4ESP32/blob/0d8592ee414f4064d65b61f969a3f983265efbbb/ESPController/src/influxdb.cpp#L202 can you add:
```
ESP_ERROR_CHECK_WITHOUT_ABORT(esp_http_client_cleanup(http_client));
```

It looks like I missed adding that line to cleanup the client after it is done being used.

atanisoft, 2022-03-17:
@stuartpittaway I've raised a PR with the change above.

virtuvas, 2022-03-17:
Mike,

added the line, hasn't crashed (in 5-10mins) so looking good (that is with 10sec influxdb interval)
btw, I'm only using http for my connection to the office influxdb. Not https

fwiw, no data gets over to the db, so no crash but no result either.

hope that helps somehow.

atanisoft, 2022-03-17:
Well, that seems like progress at least!

Can you change https://github.com/stuartpittaway/diyBMSv4ESP32/blob/2c1fdf3401d1c77d95d92613049db39be7d3a8eb/ESPController/src/main.cpp#L3076 from `ESP_LOG_INFO` to `ESP_LOG_VERBOSE`? That should dump a bit more info as it is being collected.

virtuvas, 2022-03-18:
sorry, picking it up again this morning,

system stayed on without a crash for 7h overnight - not connected to a PC, standalone with a 3A powersupply on the controller. No influxdb crashes or system reboots. Which is excellent as typically on the master branch I'd get a reboot within the first 4-6h of firing up and then it would stay without a reboot for 4-5days max at a time. No data on influx at my work server though.

Mike, changed the line to ESP_LOG_VERBOSE and then even ESP_LOG_DEBUG, no decipherable difference on the log:

```
I (4308029) diybms: Task 1
D (4309593) diybms-tft: Update TFT display
I (4310281) diybms-webreq: API call: monitor2
I (4311788) diybms-mqtt: MQTT Payload for cell data
D (4312169) diybms-tft: Update TFT display
I (4314358) diybms-webreq: API call: monitor2
I (4314529) diybms: Task 2
D (4315173) diybms-tft: Update TFT display
I (4316792) diybms-mqtt: MQTT Payload for cell data
I (4316796) diybms: Influx task
I (4316850) diybms-influxdb: HTTP Client Connected!
D (4316895) diybms: total_free_byte=130016 total_allocated_byte=167168 largest_free_blk=79860 min_free_byte=122376 alloc_blk=516 free_blk=11 total_blk=527
I (4316996) diybms-influxdb: HTTP Client finished
I (4316997) diybms-influxdb: HTTP Client Disconnected!
D (4318169) diybms-tft: Update TFT display
I (4318446) diybms-webreq: API call: monitor2
D (4318684) diybms: Open log /data_20220318.csv
D (4318693) diybms: Wrote to SD log
I (4321029) diybms: Task 3, s=0 e=7
D (4321171) diybms-tft: Update TFT display
I (4322001) diybms-mqtt: MQTT Payload for cell data
I (4322534) diybms-webreq: API call: monitor2
D (4324676) diybms-tft: Update TFT display
I (4326604) diybms-webreq: API call: monitor2
I (4327005) diybms-mqtt: MQTT Payload for cell data
I (4327009) diybms: Influx task
I (4327064) diybms-influxdb: HTTP Client Connected!
D (4327173) diybms-tft: Update TFT display
I (4327214) diybms-influxdb: HTTP Client finished
I (4327215) diybms-influxdb: HTTP Client Disconnected!
D (4330171) diybms-tft: Update TFT display
I (4330684) diybms-webreq: API call: monitor2
D (4331896) diybms: total_free_byte=137452 total_allocated_byte=160344 largest_free_blk=79860 min_free_byte=122376 alloc_blk=465 free_blk=27 total_blk=492
I (4332220) diybms-mqtt: MQTT Payload for cell data
I (4332224) diybms-mqtt: General status payload
I (4332225) diybms-mqtt: Bank(0) status payload
I (4332225) diybms-mqtt: Rule status payload
I (4332227) diybms-mqtt: Outputs status payload
D (4333174) diybms-tft: Update TFT display
I (4334029) diybms: Task 1
I (4334762) diybms-webreq: API call: monitor2
D (4336174) diybms-tft: Update TFT display
I (4337231) diybms-mqtt: MQTT Payload for cell data
I (4337235) diybms: Influx task
I (4337288) diybms-influxdb: HTTP Client Connected!
I (4337425) diybms-influxdb: HTTP Client finished
I (4337425) diybms-influxdb: HTTP Client Disconnected!
I (4338835) diybms-webreq: API call: monitor2
D (4339173) diybms-tft: Update TFT display
I (4340529) diybms: Task 2
D (4342177) diybms-tft: Update TFT display
I (4342430) diybms-mqtt: MQTT Payload for cell data
I (4342907) diybms-webreq: API call: monitor2
D (4345177) diybms-tft: Update TFT display
D (4346897) diybms: total_free_byte=137672 total_allocated_byte=160136 largest_free_blk=79860 min_free_byte=122376 alloc_blk=464 free_blk=28 total_blk=492
I (4346998) diybms-webreq: API call: monitor2
I (4347029) diybms: Task 3, s=0 e=7
I (4347436) diybms-mqtt: MQTT Payload for cell data
I (4347440) diybms: Influx task
I (4347496) diybms-influxdb: HTTP Client Connected!
I (4347635) diybms-influxdb: HTTP Client finished
I (4347636) diybms-influxdb: HTTP Client Disconnected!
D (4348801) diybms: Open log /data_20220318.csv
D (4348812) diybms: Wrote to SD log
D (4349990) diybms-tft: Update TFT display
I (4351073) diybms-webreq: API call: monitor2
D (4351395) diybms-tft: Update TFT display
I (4352641) diybms-mqtt: MQTT Payload for cell data
D (4354176) diybms-tft: Update TFT display
I (4355149) diybms-webreq: API call: monitor2
D (4357177) diybms-tft: Update TFT display
I (4357645) diybms-mqtt: MQTT Payload for cell data
I (4357649) diybms-mqtt: General status payload
I (4357649) diybms-mqtt: Bank(0) status payload
I (4357650) diybms-mqtt: Rule status payload
I (4357651) diybms-mqtt: Outputs status payload
I (4357655) diybms: Influx task
I (4357712) diybms-influxdb: HTTP Client Connected!
I (4357853) diybms-influxdb: HTTP Client finished
I (4357854) diybms-influxdb: HTTP Client Disconnected!
I (4359220) diybms-webreq: API call: monitor2
I (4360030) diybms: Task 1
```
just an extract of two full loops of Task 1-3 as reported.
tried upping the platformio.ini  
```
-DCORE_DEBUG_LEVEL=5
```
to 7 no difference. I do remember at my early tests with influxdb back in Dec that I could see all comms between controller and my server (even the full data posted with the IP and token!) anyone knows how I can do that now? Thought it was a matter of debug level, obvs not.

Weather's going to be crap this w/e so will be mainly at home working, good opportunity to do some more testing!

cheers

V.

stuartpittaway, 2022-03-18:
There was a simple typo on the word "bucket" - try it again please @virtuvas

virtuvas, 2022-03-18:
mqqt stopped reporting when disabled 👍 
bucket typo didn't help though :-(
@stuartpittaway anyway to get a FULL log of influxdb.cpp actions? was able to do it in Dec-Jan, not now - or forgot how to!

V.

virtuvas, 2022-03-18:
also stuart,

I think that line 145 on  influxdb.cpp is wrong:
is:
```
            ESP_LOGV(TAG, "Index:%d, bank:%d, module:%d, id:%s, voltage:%s, int-temp:%d, ext-temp:%s, bypass:%s",
```
should be (imho):
```
            ESP_LOGV(TAG, "Index:%d, bank:%d, module:%d, id:%s, voltage:%s, int-temp:%s, ext-temp:%s, bypass:%s",
```
(int and ext temp are same units, no? should be %s for both)

still nothing arrives at the server though!

V.

stuartpittaway, 2022-03-18:
@atanisoft I've also noticed that I can't get VERBOSE logging to ever spit out the log info (after changing the correct tag/log level in main.cpp).

@virtuvas The logging is correct, the voltage has already been converted to a string, but the temperature should remain as a decimal.

virtuvas, 2022-03-18:
so int-temp:%d
but ext-temp:%s
I thought they are both treated/exported in influx as integers and anyway expected the same behaviour for int and ext.

stuartpittaway, 2022-03-18:
I didn't spot that one, you are correct.

virtuvas, 2022-03-18:
np,

btw, shouldn't:
```
    SERIAL_DEBUG.println(module_data.c_str());
```
produce something on the VSCode terminal?

stuartpittaway, 2022-03-18:
No that no longer works, its a legacy "arduino" thing which is being slowly removed!

The ESP_LOG stuff is similar.

So I've spun up a docker container with influxdb 2, and its not working for me either.

This is what I get.  Still no VERBOSE output, even though its enable.

```
I (58044) diybms: Influx task
E (58071) esp-tls: [sock=56] delayed connect error: Connection reset by peer
E (58073) TRANSPORT_BASE: Failed to open a new connection: 32772
E (58073) HTTP_CLIENT: Connection failed, sock < 0
E (58077) diybms-influxdb: HTTP Client Error encountered
E (58083) diybms-influxdb: Error perform http request ESP_ERR_HTTP_CONNECT
I (58089) diybms-influxdb: HTTP Client Disconnected!
```

virtuvas, 2022-03-18:
care to explain how I could get the current string created earlier in, printed using this ESP_LOGI or whatever?

fe VSCode doesn't like:
```ESP_LOGI(TAG, module_data.c_str()); 
```

stuartpittaway, 2022-03-18:
Okay, progress - but I still don't see anything in influxdb !!

```
I (301079) diybms: Influx task
I (301099) diybms-influxdb: HTTP Client Connected!
D (301174) diybms-influxdb: Header: key=Accept-Ranges, value=bytes
D (301175) diybms-influxdb: Header: key=Cache-Control, value=public, max-age=3600
D (301177) diybms-influxdb: Header: key=Content-Length, value=511
D (301183) diybms-influxdb: Header: key=Content-Type, value=text/html; charset=utf-8
D (301190) diybms-influxdb: Header: key=Etag, value="51121171319"
D (301196) diybms-influxdb: Header: key=Last-Modified, value=Tue, 21 Sep 2021 17:13:19 GMT
D (301204) diybms-influxdb: Header: key=X-Influxdb-Build, value=OSS
D (301210) diybms-influxdb: Header: key=X-Influxdb-Version, value=2.0.9
D (301216) diybms-influxdb: Header: key=Date, value=Fri, 18 Mar 2022 10:20:30 GMT
D (301224) diybms-influxdb: HTTP Client data recevied: len=219
D (301231) diybms-influxdb: HTTP Client data recevied: len=292
I (301235) diybms-influxdb: HTTP Client finished
I (301239) diybms-influxdb: Status = 200, content_length = 511
I (301244) diybms-influxdb: HTTP Client Disconnected!
D (301317) diybms-tft: Update TFT display

```

stuartpittaway, 2022-03-18:
FINALLY GOT IT WORKING!

Bit of a doh! moment, forgot to use the correct path on the URL ```/api/v2/write```

So my URL in diybms looks like this...

```
http://192.168.0.49:8086/api/v2/write
```

virtuvas, 2022-03-18:
hm, you think you did :-)

it was working all along according to my latest testing...
somehow, doesn't like the multidata format that comes with your standard setup.
As soon as I added some extra data I wanted it to post, it started working (for them) v.confusing. 
I now get:
```
I (75280) diybms-influxdb: influxdata=
cells, cell=0_0 v=3.2630,i=21i,e=21i,b=false,p=0
cells, cell=0_1 v=3.2650,i=21i,e=21i,b=false,p=0
cells, cell=0_2 v=3.2660,i=21i,e=21i,b=false,p=0
cells, cell=0_3 v=3.2650,i=22i,e=21i,b=false,p=0
cells, cell=0_4 v=3.2680,i=21i,e=21i,b=false,p=0
cells, cell=0_5 v=3.2650,i=22i,e=21i,b=false,p=0
cells, cell=0_6 v=3.2640,i=21i,e=21i,b=false,p=0
cells, cell=0_7 v=3.2650,i=22i,e=21i,b=false,p=0
bank vr=5
systemVAR uptime=68.4440
```
but only bank vr and systemVAR uptime show up...

doh! doesn't need a comma after cells!!! ffs!

stuartpittaway, 2022-03-18:
```cells, cell```  Get rid of the space between !

virtuvas, 2022-03-18:
and needs a comma between cell and v?

let me check it carefully to make sure it's compatible with the previous setup!

stuartpittaway, 2022-03-18:
> and needs a comma between cell and v?

Yes.  The "cells" are a group and then each "cell" is a tag.  The space then seperates the tag from the values.

virtuvas, 2022-03-18:
👍 

in a while I'll sent my additions for you to check.

atanisoft, 2022-03-18:
> I've also noticed that I can't get VERBOSE logging to ever spit out the log info (after changing the correct tag/log level in main.cpp).

GAH, I forgot that IDF introduced a new config parameter that limits the log level. It looks like DEBUG is the highest level possible. We should adjust all usages of `ESP_LOGV` to `ESP_LOGD`. Once you move over to fully native IDF we can add an `sdkconfig.defaults` to override the default level.

Thanks for fixing up the typos I created and adding the logging updates. I had tried to preserve what was there previously but looks like I missed a couple pieces.

stuartpittaway, 2022-03-18:
Thanks for your work on this @atanisoft it seems to work well.

atanisoft, 2022-03-18:
Glad that it is working after your / @virtuvas updates.

closed 2022-03-18 by virtuvas