stuartpittaway/diyBMSv4ESP32 #99: NOV2021 cannot connect to open/no passwd WIFI ?

virtuvas, 2022-03-17:
subject line really,

downloaded, edited the bits I want for influxDB logging, upped the code, but where I am wifi is open/no passwd.
can spacebar, select wifi, but cannot hit enter and pass the passwd command. Any tricks I need?
don't remember having to do anything fancy on the previous versions.
Mind I did all that from within PlatformIO

cheers

V

stuartpittaway, 2022-03-17:
The access point has been removed going forward. The space bar or a JSON text file on the SD card are the only options.

What software/terminal are you using to connect to the esp?

virtuvas, 2022-03-17:
I'm on win10.
running latest VSC and Platformio, using a plain usb cable to connect win10 to esp32. Running the terminal from within PlatformIO.
could use putty or something else if it helps, however, a json txt file sounds nice and means I can swap ESP32chips and reload code using the site specific SDcard, so any pointers welcomed

V.

stuartpittaway, 2022-03-17:
If you boot up the old ESP32 (with older code on it) go to the storage option in the web interface and select save WiFi.

Then swap the esp to the new one and boot up, you should see the code load up the configuration file automatically.

virtuvas, 2022-03-17:
stuart,

reopened Feb22 code on platformio, recompiled/installed it, booted fine, found the wifi, interface ok, ip the old one.
went to storage/save wifi and for good measure did a storage/save configuration
I now can see on the sdcard (well the list of files as reported by the storage web interface - haven't pulled out the sd card yet) the following two new files:

/output_status_20220317.csv
/backup_config_20220317_113756.json

the first one just has one line:
DateTime,Bits,Output_0,Output_1,Output_2,Output_3

the second one I'm not going to post as it has the influxdb passwds but doesn't seem to have the wifi data anywhere. 
Am I missing something?

asking as I'd rather not go back and forth lots of times this "Rebuilding Intellisense Index" is not swift on my h/w!

cheers

saw a .csv file created

virtuvas, 2022-03-17:
well, 

pulled the sd card, there is indeed a new file wifi.json in diybms folder off the root of the SD card which is the right date/time/data.
However, recompiled and uploaded the NOV2021 code, still wont connect:

```
D (38127) diybms: starting wifi_init_sta
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008d928
file: "src/main.cpp" line 1644
func: void wifi_init_sta()
expression: esp_event_loop_create_default()

abort() was called at PC 0x4008d92b on core 1

Backtrace:0x40084491:0x3ffb26c00x4008d935:0x3ffb26e0 0x40093d55:0x3ffb2700 0x4008d92b:0x3ffb2780 0x400d6a07:0x3ffb27a0 0x400d7af5:0x3ffb27c0 0x400eba9d:0x3ffb2820
  #0  0x40084491:0x3ffb26c00 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[��mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (36) diybms: 

               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:LocalCompile compiled 2022-03-17T12:57:37.360Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (48) diybms-hal: Configure I2C
I (51) diybms-hal: Scanning i2c bus
I (56) diybms-hal: Found i2c device at address 0x20
I (60) diybms-hal: Found i2c device at address 0x38
I (68) diybms-hal: TCA6416A not fitted, assume v4.2 board
I (70) diybms-hal: Found TCA9534A
I (71) diybms-hal: Found TCA6408
D (73) diybms: Configure VSPI
I (77) diybms: ** Controller changed state from Unknown to PowerUp **
D (82) diybms: Configure VSPI
D (85) diybms-tft: Configure TFT display
[    93][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (94) gpio: gpio_set_level(226): GPIO output gpio_num error
[   104][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (106) gpio: gpio_set_level(226): GPIO output gpio_num error
[   116][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected

Press SPACE BAR to enter console WiFi configuration....

....................

No key press detected
I (5472) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (5473) diybms-hal: CAN driver started
I (5475) diybms: LittleFS mounted, total=589824, used=12288
I (5478) diybms: Mounting SD card
I (5491) diybms: SD card available
[  5495][D][settings.cpp:20] ReadConfig(): ReadConfig diybmswifi
[  5498][D][settings.cpp:42] ReadConfig(): checksum verified
I (5495) diybms: Checking for /diybms/wifi.json
D (5507) diybms: Found file /diybms/wifi.json
D (5519) diybms: Deserialized /diybms/wifi.json
I (5520) diybms: Wifi JSON config is identical, ignoring
I (5520) diybms: Fetch config
[  5525][D][settings.cpp:20] ReadConfig(): ReadConfig diybms
I (5526) diybms: Apply default config
D (5531) diybms: Setup RS485
D (5533) diybms: Configure RS485
I (5537) diybms: ** Controller changed state from PowerUp to Stabilizing **
D (5542) diybms: starting wifi_init_sta
D (5606) diybms: tca6408_isr
I (5667) diybms: Hostname: DIYBMS-004CE8AC
D (5667) diybms: wifi_init_sta finished
I (5667) diybms: TFT screen is INSTALLED
D (5668) diybms: total_free_byte=158908 total_allocated_byte=140364 largest_free_blk=110580 min_free_byte=158392 alloc_blk=342 free_blk=4 total_blk=346
I (15850) diybms: retry to connect to the AP
I (15851) diybms: connect to the AP fail
I (16672) diybms-rules: Set error 4
I (17904) diybms: retry to connect to the AP
I (17905) diybms: connect to the AP fail
I (18672) diybms-mqtt: MQTT Payload for cell data
I (19958) diybms: retry to connect to the AP
I (19959) diybms: connect to the AP fail
I (20166) diybms: Task 1
I (21682) diybms: Active errors=1
I (21684) diybms-tft: Wake up screen
D (21716) diybms-tft: Update TFT display
I (22013) diybms: retry to connect to the AP
I (22014) diybms: connect to the AP fail
I (23672) diybms-mqtt: MQTT Payload for cell data
I (24067) diybms: retry to connect to the AP
I (24067) diybms: connect to the AP fail
I (24682) diybms-rules: Set error 4
I (26120) diybms: retry to connect to the AP
I (26121) diybms: connect to the AP fail
```

curious why it finds the wifi.json and then comes up with this:
```I (5520) diybms: Wifi JSON config is identical, ignoring```
wonder if I should wipe clean the ESP32 first, rebuilt f/s and then reprogram it?

V.

stuartpittaway, 2022-03-17:
Is this the absolutely latest code from GitHub? I did a lot of check-in yesterday, so make sure it's up to date.

It's found the WiFi config, loaded it and stored it in flash, so everything looks good there.

stuartpittaway, 2022-03-17:
Just to add - the keyboard input on the "PRESS SPACE BAR" should be working fine - try using the ENTER key on the numeric keypad if your keyboard has one, just as a test.

virtuvas, 2022-03-17:
tried the space bar, works, even tried typing ENTER on the numeric keyboard. Didn't work.
fwiw, I have 3 of these:
[image]

ESP-WROOM-32 (on the front on top of the chip)

and on the back:
NODE MCU
ESP-32S
v1.1 101010

hope it's compatible.
As a temp measure, about to setup another pass protected wifi at home and connect (cannot mess with the other as there are a few important devices connected to it)

V

virtuvas, 2022-03-17:
doesn't want to connect even with passwd!

```
.....I (1078) diybms: ** Controller changed state from PowerUp to NoWIFI **

DIYBMS CONTROLLER - Scanning Wifi

Number  SSID Name                             RSSI      Channel
0:      NOEMON_Guest                          -62       4
1:      NOEMON                                -62       4
2:      CYTA6F9D                              -88       9
3:      NOEMON_EXT                            -94       4
Enter the NUMBER of the Wifi network to connect to:0
Enter the password to use when connecting to 'NOEMON_Guest':*********
[ 19232][D][settings.cpp:5] WriteConfig(): WriteConfig diybmswifi

Rebooting In 10 seconds
Rebooting In 9 seconds
Rebooting In 8 seconds
Rebooting In 7 seconds
Rebooting In 6 seconds
Rebooting In 5 seconds
Rebooting In 4 seconds
Rebooting In 3 seconds
Rebooting In 2 seconds
Rebooting In 1 seconds
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[��mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (22) diybms: 

               _          __
  _|  o       |_)  |\/|  (_
 (_|  |  \/   |_)  |  |  __)
         /

CONTROLLER - ver:LocalCompile compiled 2022-03-17T13:24:13.292Z
ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
I (34) diybms-hal: Configure I2C
I (37) diybms-hal: Scanning i2c bus
I (42) diybms-hal: Found i2c device at address 0x20
I (46) diybms-hal: Found i2c device at address 0x38
I (55) diybms-hal: TCA6416A not fitted, assume v4.2 board
I (56) diybms-hal: Found TCA9534A
I (57) diybms-hal: Found TCA6408
D (59) diybms: Configure VSPI
I (63) diybms: ** Controller changed state from Unknown to PowerUp **
D (68) diybms: Configure VSPI

Press SPACE BAR to enter console WiFi configuration....

....................

No key press detected
I (5078) diybms-hal: CAN driver installed.  Filter=1621098496 Mask=6291455
I (5078) diybms-hal: CAN driver started
I (5081) diybms: LittleFS mounted, total=589824, used=12288
I (5084) diybms: Mounting SD card
I (5096) diybms: SD card available
[  5101][D][settings.cpp:20] ReadConfig(): ReadConfig diybmswifi
[  5104][D][settings.cpp:42] ReadConfig(): checksum verified
I (5101) diybms: Checking for /diybms/wifi.json
D (5113) diybms: Found file /diybms/wifi.json
D (5125) diybms: Deserialized /diybms/wifi.json
I (5126) diybms: Wifi config is different, saving
[  5131][D][settings.cpp:5] WriteConfig(): WriteConfig diybmswifi
[  5141][D][settings.cpp:20] ReadConfig(): ReadConfig diybmswifi
[  5143][D][settings.cpp:42] ReadConfig(): checksum verified
I (5140) diybms: Fetch config
[  5147][D][settings.cpp:20] ReadConfig(): ReadConfig diybms
[  5153][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: bytes NOT_FOUND
I (5156) diybms: Apply default config
D (5161) diybms: Setup RS485
D (5163) diybms: Configure RS485
I (5167) diybms: ** Controller changed state from PowerUp to Stabilizing **
D (5172) diybms: starting wifi_init_sta
D (5231) diybms: tca6408_isr
I (5293) diybms: Hostname: DIYBMS-004CE8AC
D (5293) diybms: wifi_init_sta finished
I (5293) diybms: TFT screen is NOT installed
D (5294) diybms: total_free_byte=161188 total_allocated_byte=138312 largest_free_blk=110580 min_free_byte=160856 alloc_blk=323 free_blk=6 total_blk=329
I (15102) diybms: retry to connect to the AP
I (15103) diybms: connect to the AP fail
I (15928) diybms-rules: Set error 4
I (17156) diybms: retry to connect to the AP
I (17157) diybms: connect to the AP fail
I (17929) diybms-mqtt: MQTT Payload for cell data
I (19210) diybms: retry to connect to the AP
I (19211) diybms: connect to the AP fail
I (19422) diybms: Task 1
I (20939) diybms: Active errors=1
I (21264) diybms: retry to connect to the AP
I (21264) diybms: connect to the AP fail
I (22929) diybms-mqtt: MQTT Payload for cell data
I (23318) diybms: retry to connect to the AP
I (23318) diybms: connect to the AP fail
I (23939) diybms-rules: Set error 4
I (25372) diybms: retry to connect to the AP
I (25373) diybms: connect to the AP fail
I (25922) diybms: Task 2
```

V.

virtuvas, 2022-03-17:
fwiw, router is a tp-link Archer600, never had issues with denied connections, typically 5-10devices wirelessly connected at any given time.

stuartpittaway, 2022-03-17:
How strange, just to confirm, the "SPACE BAR" wifi setup worked okay and the keyboard responded as you would expect?

Its obviously trying to connect to the AP but failing for some reason, looks like there are no catastrophic errors though, which is good.  The signal strength (RSSI) looks okay as well.

Does this looping just continue with "connect to the AP fail" 

Its trying to do WPA2_PSK encryption - is this also whats configured on the AP ?

stuartpittaway, 2022-03-17:
Ah, just seen this...

Its resetting the wifi back to the one in the configuration file because the data in that is different to what you have entered in the SPACE BAR setup.

Try it again, but take out the SD CARD (or delete the wifi file).

```
I (5101) diybms: Checking for /diybms/wifi.json
D (5113) diybms: Found file /diybms/wifi.json
D (5125) diybms: Deserialized /diybms/wifi.json
I (5126) diybms: Wifi config is different, saving
```

virtuvas, 2022-03-17:
doh!

removed SD, worked! I had seen this message but thought it meant that it's saving as in updating the .json file and using it :-)

edited wifi.json on the SD, refitted it, rebooted, fine again, however all settings gone :-(
at some point restoring from the saved backup_config_*.json would be helpful.

BTW, all hell broke loose as I now have the ESP connected to a "guest" network and looks like it's failing in connecting to the influxDB which is at my work office (outside the house network)...

```
I (204581) diybms-mqtt: MQTT Payload for cell data
I (204585) diybms: Influx task
E (204591) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (204592) TRANSPORT_BASE: Failed to open a new connection: 32770
E (204593) HTTP_CLIENT: Connection failed, sock < 0
E (204598) diybms-influxdb: HTTP Client Error encountered
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x7002 (ESP_ERR_HTTP_CONNECT) at 0x4008d913
file: "src/influxdb.cpp" line 219
func: void influx_task_action()
expression: esp_http_client_perform(http_client)
D (204914) diybms-tft: Update TFT display
I (205875) diybms-webreq: API call: monitor2
I (207016) diybms: Task 2
D (207150) diybms: total_free_byte=87328 total_allocated_byte=201492 largest_free_blk=40948 min_free_byte=83800 alloc_blk=1213 free_blk=28 total_blk=1241
D (207917) diybms-tft: Update TFT display
I (209621) diybms-mqtt: MQTT Payload for cell data
I (209949) diybms-webreq: API call: monitor2
D (210919) diybms-tft: Update TFT display
I (213516) diybms: Task 3, s=0 e=7
D (213932) diybms-tft: Update TFT display
I (214039) diybms-webreq: API call: monitor2
I (214625) diybms-mqtt: MQTT Payload for cell data
I (214629) diybms-mqtt: General status payload
I (214630) diybms-mqtt: Bank(0) status payload
I (214630) diybms-mqtt: Rule status payload
I (214631) diybms-mqtt: Outputs status payload
I (214635) diybms: Influx task
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[��mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I (23) diybms: 
```
annoyingly I have a "normal" passwd protected network at home but it's at 5GHz and not 2.4 and ESP doesn't see it...
will sort something out and report back.
What would be the way to accept non-passwd wifis?

V.

stuartpittaway, 2022-03-17:
Glad its working.

It won't work with non-password wifis it expects WPA2 encryption.