stuartpittaway/diyBMSv4ESP32 #102: Screen not displaying after booting up

Torsdan, 2022-03-18:
Having an issue where the screen will show the Powering Up but then go dark, flicker briefly as it tries to display something but remains dark. 
Have tried 2 different screens. Same happens with either USB or Ext 5v.
Web interface still works fine with or without LCD, although with LCD attached web shows Emergency stop.

[image]

Output from COM while on USB.
```
[I][main.cpp:3091] setup(): CONTROLLER - ver:897622a0f7255d860bccef62272602a6d2984b7b compiled 2022-02-26T13:27:00.013Z
[I][main.cpp:3096] setup(): ESP32 Chip model = 1, Rev 1, Cores=2, Features=50
[I][HAL_ESP32.cpp:232] ConfigureI2C(): Configure I2C
[I][HAL_ESP32.cpp:291] ConfigureI2C(): Found TCA9534A
[I][HAL_ESP32.cpp:321] ConfigureI2C(): Found TCA6408
[D][HAL_ESP32.h:288] TouchScreenUpdate(): Touch = touch=0 pressure=128 x=0 y=3055
[I][main.cpp:3111] setup(): TFT screen is INSTALLED
[I][main.cpp:946] SetControllerState(): ** Controller changed state from Unknown to PowerUp **
[I][main.cpp:3133] setup(): LITTLEFS mounted, totalBytes=589824, usedBytes=98304
[I][main.cpp:254] mountSDCard(): Mounting SD card
[I][main.cpp:267] mountSDCard(): SD card available
[I][HAL_ESP32.cpp:140] ConfigureCAN(): CAN driver installed.  Filter=1621098496 Mask=6291455
[I][HAL_ESP32.cpp:150] ConfigureCAN(): CAN driver started
[D][settings.cpp:20] ReadConfig(): ReadConfig diybms
[D][settings.cpp:42] ReadConfig(): checksum verified
[I][main.cpp:3157] setup():
[D][main.cpp:213] SetupRS485(): Setup RS485
[D][main.cpp:197] ConfigureRS485(): Configure RS485
[D][HAL_ESP32.cpp:86] Setasks(): Sleep
etOutputState 0=153
[I][main.cpp:2713] lazy_tOutputState(): [D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 1=153
[D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 2=153
[D][HAL_ESP32.cpp:86] SetOutputState(): SetOutputState 3=153
Press SPACE BAR to enter terminal based configuration................skipped
[D][settings.cpp:20] ReadConfig(): ReadConfig diybmswifi
[D][settings.cpp:42] ReadConfig(): checksum verified
[I][main.cpp:3002] LoadWiFiConfigFromSDCard(): Checking for /diybms/wifi.json
[D][main.cpp:3008] LoadWiFiConfigFromSDCard(): Found file /diybms/wifi.json
[D][main.cpp:3022] LoadWiFiConfigFromSDCard(): Deserialized /diybms/wifi.json
[I][main.cpp:3060] LoadWiFiConfigFromSDCard(): Wifi JSON config is identical, ignoring
[I][main.cpp:946] SetControllerState(): ** Controller changed state from PowerUp to Stabilizing **
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[I][tft.cpp:388] tftwakeup_task(): Wake up screen
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:770] tca6408_isr_task(): tca6408_isr
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[I][main.cpp:1444] connectToWifi(): Hostname: DIYBMS-00E1287C, current state 255
[D][main.cpp:3362] loop(): total_free_byte=117524 total_allocated_byte=186200 largest_free_blk=84048 min_free_byte=112836 alloc_blk=308 free_blk=13 total_blk=321
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[I][Rules.cpp:142] SetError(): Set error 4
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.79, MASK: 255.255.255.0, GW: 192.168.1.1
[I][main.cpp:1498] onWifiConnect(): Wi-Fi status=3
[I][main.cpp:1500] onWifiConnect(): Request NTP from time.google.com
[I][DIYBMSServer.cpp:2745] StartServer(): Start Web Server complete
[I][main.cpp:1532] onWifiConnect(): mDNS responder started
[I][main.cpp:1537] onWifiConnect(): You can access DIYBMS interface at http://DIYBMS-00E1287C.local or http://192.168.1.79
[I][tft.cpp:388] tftwakeup_task(): Wake up screen
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[I][main.cpp:2718] lazy_tasks(): Task 1
[I][Rules.cpp:142] SetError(): Set error 7
[I][Rules.cpp:142] SetError(): Set error 4
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][Rules.cpp:142] SetError(): Set error 7
[I][main.cpp:946] SetControllerState(): ** Controller changed state from Stabilizing to Running **
[I][main.cpp:2729] lazy_tasks(): Task 2
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[I][Rules.cpp:142] SetError(): Set error 7
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[D][main.cpp:3362] loop(): total_free_byte=80700 total_allocated_byte=219236 largest_free_blk=49096 min_free_byte=71728 alloc_blk=536 free_blk=46 total_blk=582
[I][Rules.cpp:142] SetError(): Set error 7
```

stuartpittaway, 2022-03-19:
Hi, let's see if we can get to the bottom of this.

First, you shouldn't be seeing emergency stop unless the two pins next to the led are shorter together. Can you check the soldering around that area, otherwise the estop will keep triggering and the BMS won't be any use!

Your ESP32 looks like it should have an aerial connected, do you have this? If so, connect it as the esp can use a lot of power while it scans for WiFi signal, especially without the aerial.

Torsdan, 2022-03-19:
The emergency stop doesn't appear if the LCD isn't connected. But I checked the pins and they aren't shorted.
No change with an aerial connected.

Connecting the LED pin of the LCD to 3.3v and restarting the ESP allows the display to function correctly and progress to the next pages showing WiFi connected and waiting for modules.
And, webpage no longer shows emergency stop but can't see the 1 module connected (without LCD it see's it correctly).
During this, the module no longer flashes it's communication LED, once power is removed from the controller the module correctly falls back to standalone.

Possibly some helpful info, I had to replace the TCA9534A with a PCA9674APW, which is listed as a direct replacement.
Also the TCA6408A > PCA6408APW, which appears to be a drop in replacement. I was unable to source the TCA parts.

stuartpittaway, 2022-03-22:
I've never tested the controller with the alternative PCA chips, so Im not sure if that combination will work or not. 

It sounds like it's nearly there but the PCA chip is not supplying enough voltage/current to the TFT screen.

Is the TFT the same one as I linked to on my YouTube video?

Torsdan, 2022-03-23:
The pinout on the PCA's are the same as the TCA's. And as far as I can tell, the datasheet shows near identical functions.

It's odd, because the screen does turn on for a while. Short video of it doing it's startup.
https://youtu.be/8dpx6BXXzZA

The TFT looks pretty much identical to yours.
[image]

virtuvas, 2022-03-23:
> Possibly some helpful info, I had to replace the TCA9534A with a PCA9674APW, which is listed as a direct replacement. Also the TCA6408A > PCA6408APW, which appears to be a drop in replacement. I was unable to source the TCA parts.

fwiw, I'm also using PCA chips but with the same #, so PCA6408APW and PCA9534. 
Everything works fine (for almost 3months now) only issue with the screen I do have is that it wont wake up from the push buttons which is no big deal tbh since screen is touch sensitive. Wouldn't mind a solution but that's for a different thread.

maybe this PCA9674APW is a bit off?

Torsdan, 2022-03-23:
I'll see if I can source a PCA9534 to test. PCA9534APW seems correct?

virtuvas, 2022-03-23:
I bought in Dec2021 from Mouser these:

[595-PCA9534APWR](https://gr.mouser.com/ProductDetail/595-PCA9534APWR)

[771-PCA6408APW,118](https://gr.mouser.com/ProductDetail/771-PCA6408APW%2c118)

no stock atm though :-(

V.

stuartpittaway, 2022-03-23:
@torsdan this is interesting.  The controller obviously can detect both chips and the TFT screen, and it displays correctly during power up, so that must be the correct pin out.

I see these lines in the log file - these should only occur when an "action" or pin changes - so you pressed the TFT screen or the ESTOP is pressed.

```
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:770] tca6408_isr_task(): tca6408_isr
```

These ISR's are directly driven from the two chips you have replaced with alternative parts, so this leads me to believe that there must be some internal difference in the way they work (even though they have the same footprint).  

The other thing to check is R24 resistor - this should be a pullup.  If you use a multimeter on the two estop pins (voltage measurement) you should get 3.3 or 5v - I forget which is connect to that!

virtuvas, 2022-03-23:
> Having said all that, the screen should still work if you touch it to wake it up!

well, for sure that doesn't work on mine Stuart! Any tests/measurement that could help identify issues related to these PCA chips?

stuartpittaway, 2022-03-23:
Ok, although the pin out of the PCA9674APW is the same as the TCA9534A, I believe they fundamentally work differently internally.

For example the PCA9674 has extra commands over i2c like "Software Reset" "Device ID".

Its pins are also described as ``A quasi-bidirectional I/O is an input or output port without using a direction control register```

Whilst the TCA9534A is described as ```The TCA9534A consists of one 8-bit Configuration (input or output selection), Input Port, Output Port, and Polarity Inversion (active high or active low) registers.```

Although the PCA9674APW is detected on the i2c bus (it has the same address) I don't believe that the device is responding correctly to the commands the controller is sending it.  This chip will then raise random interrupts like ESTOP as its input pins are incorrectly configured.

@Torsdan does the LED on the controller board flash GREEN about every 5 seconds?

@virtuvas If you have the PCA9534APWR does your ESTOP work (connect the two pins together)

virtuvas, 2022-03-23:
> @virtuvas If you have the PCA9534APWR does your ESTOP work (connect the two pins together)

don't know what my ESTOP is Stuart 😀 got to be more specific I'm afraid!
Mind, I have the NOV2021 code on now and about to break my personal best of 5days without rebooting and that is with influxDB posting every 10secs, Victron working, the lot, well impressed! 
So not keen on messing with the system right now. 
However, would like to test that sometime tomorrow any pointers welcomed.
I do have a second test bank with a second complete controller on the old code, but no screen. If I unplug the screen as is with the system running, will it be OK or reboot?

stuartpittaway, 2022-03-23:
Understand that @virtuvas no rush.  The ESTOP is the two pins next to the ISP header.  Just join the two pins together - it was designed for you to have a "big red button" to provide an emergency stop.

virtuvas, 2022-03-23:
oh, had forgotten about these pins.
OK, tested it on the second controller (without screen) on jumpering these two, led turns red and I get lots of:
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
lines.
System doesn't reboot though. Just on previous wifi settings with no pswd and need to wipe it clean to get it back to connect.

```
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:1444] connectToWifi(): Hostname: DIYBMS-00BADE2D, current state 6
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[E][main.cpp:1548] onWifiDisconnect(): Disconnected from Wi-Fi.
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2718] lazy_tasks(): Task 1
[D][main.cpp:3362] loop(): total_free_byte=146928 total_allocated_byte=157132 largest_free_blk=113792 min_free_byte=145112 
alloc_blk=288 free_blk=10 total_blk=298
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2729] lazy_tasks(): Task 2
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2764] lazy_tasks(): Task 3, s=0 e=3
[D][main.cpp:3362] loop(): total_free_byte=146928 total_allocated_byte=157132 largest_free_blk=113792 min_free_byte=145112 
alloc_blk=288 free_blk=10 total_blk=298
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2713] lazy_tasks(): Sleep
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2718] lazy_tasks(): Task 1
[I][main.cpp:1444] connectToWifi(): Hostname: DIYBMS-00BADE2D, current state 6
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[E][main.cpp:1548] onWifiDisconnect(): Disconnected from Wi-Fi.
[D][main.cpp:3362] loop(): total_free_byte=146928 total_allocated_byte=157132 largest_free_blk=113792 min_free_byte=145112 
alloc_blk=288 free_blk=10 total_blk=298
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2729] lazy_tasks(): Task 2
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2764] lazy_tasks(): Task 3, s=0 e=3
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[D][main.cpp:791] tca9534_isr_task(): tca9534_isr
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2713] lazy_tasks(): Sleep
[I][Rules.cpp:142] SetError(): Set error 7
[D][main.cpp:3362] loop(): total_free_byte=146928 total_allocated_byte=157132 largest_free_blk=113792 min_free_byte=145112 
alloc_blk=288 free_blk=10 total_blk=298
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
[I][main.cpp:2718] lazy_tasks(): Task 1
[I][Rules.cpp:142] SetError(): Set error 7
[D][PacketReceiveProcessor.cpp:79] ProcessReply(): Finished all reads
```

stuartpittaway, 2022-03-23:
> OK, tested it on the second controller (without screen) on jumpering these two, led turns red

Thats whats meant to happen - does this 2nd controller PCB have the PCA9534APWR  chip on it, thats what I'm trying to test!!

virtuvas, 2022-03-23:
yep, only have PCAs on both controllers Stuart, couldn't find TCAs last December. So all is fine then.

Torsdan, 2022-03-23:
Testing done _without_ TFT connected.

LED does flash green at ~5s intervals.

V difference across ESTOP is 0v.
Other side of R24 measures 3.3v.
Resistance from the output of the 3.3v regulator to the right pin of ESTOP (closest to ISP header) measures 10k.
I assume the PCA9674APW is pulling it down internally?

Shorted ESTOP pins, nothing happened. Controller continues to run fine.

By the looks, I'll source a PCA9534APWR since the 9674 is not a drop-in replacement.
Shall I leave this open until it arrives and I can test that it functions correctly?

virtuvas, 2022-03-23:
Torsdan, since it's working for me fine (9534 I mean) I'd probably close the issue and if there is a problem once you find and fit a new chip you can reopen. 
That's what I've done with a couple of other issues I had...

Torsdan, 2022-03-23:
Ok, I shall do that then.
Thank you @virtuvas for helping us test and confirm things!
Thank you @stuartpittaway for this project and willingness to help!

closed 2022-03-23 by Torsdan

stuartpittaway, 2022-03-23:
Thanks for testing!

Torsdan, 2022-04-22:
Just confirming that replacing the PCA9674APW with a PCA9534APWR does indeed fix the issue.

stuartpittaway, 2022-04-22:
Thank you for feeding back @Torsdan.  I'll close this ticket.