stuartpittaway/diyBMSv4ESP32 #103: Power calculation incorrect

jdevine82, 2022-03-24:
Hi
I am not sure if it has always been wrong but just noticed that my power calculation on homepage seems incorrect. I am running a current shunt and the current value seems correct. 

Version ending in 8939d from 26/11/21
[image]
Not a big deal but yes can see why it is like this.
Thanks
Jason

jdevine82, 2022-03-24:
Not sure why image didn't upload but here is a link to it in. My drive 
https://drive.google.com/file/d/1lX7dWfxIGPwz9I61ipd2mbd99TeAdnIg/view?usp=drivesdk

stuartpittaway, 2022-03-24:
How bizarre!

Has it always been that way?  Do you know if the power reading is correct when you have a negative current ?

I'm wondering if its a "sign" issue.

atanisoft, 2022-03-24:
> I am running a 75mv shunt but I image that is not any issue.

The code is designed for a 50mV shunt, changing to a 75mV shunt very likely will require some code adjustments to ensure the accuracy of the readings by the INA228.

jdevine82, 2022-03-25:
So today it is afternoon again and the power seems incorrect again. It must be some sort of difference between the averageing for the current measurement of the ina228 and the Power sampling averaging. With my elcheapo clamp meter I have about 5A flowing through the shunt.
[image]
A reboot of the controller has not fixed it so it is a system issue rather than a code issue. I am wondering if changing the sample average from 256 up to 1024 would help?

stuartpittaway, 2022-03-25:
All the voltage/current/power calculations happen inside the shunt - actually inside the INA228 chip.

To get the power reading out of the INA chip, I'm using this code inside the current shunt microcontroller.

Given that the shunt voltage and shunt current appear to be correct (and confirmed with clamp meter), I'm not quite sure what to suggest.  

The INA228 uses a scaling value (CURRENT_LSB) to scale its readings depending on the shunt mV and maximum current rating - although if this were incorrect, the voltage and current would also be wrong.

```
//Calculated power output.  Output value in watts. Unsigned representation. Positive value.
double Power()
{
  //POWER Power [W] = 3.2 x CURRENT_LSB x POWER
  return (double)i2c_readUint24(INA_REGISTER::POWER) * (double)3.2 * registers.CURRENT_LSB;
}
```

stuartpittaway, 2022-03-25:
Just another thought, does the "millivolt" reading on the current monitor screen look correct?

It will be a tiny number.

jdevine82, 2022-03-25:
Yes it seems correct as per my 200a 75mv shunt. At 12a I am getting 4.6mv pretty close to calculated 4.5.  

Don't spend too much effort on this as it is not a show stopper just more of a slight nuisance.

stuartpittaway, 2022-03-25:
Thanks.  So with a 200A/75mV shunt, the maximum the DIYBMS current monitor will ever report is

200 / 75 * 40.96 = 109.2Amps  (the INA228 has a 40.96mV max. scale) but you are well below that with the examples you have provided.

On my test bench, I can only go up to 6Amp (bench power supply) but my tests so far have been spot on.

I've asked a question on the open energy monitor forum to see if anyone has the same problem.

I'm still thinking its a formatting issue of the data bytes being returned to the controller, rather anything fundamentally wrong with the shut itself (because the voltage and current are correct).

stuartpittaway, 2022-04-28:
Hi @jdevine82 is this still a problem for you?

Chris87-ux, 2022-07-18:
Hi @stuartpittaway i just did some reserch, as i found the same issue (transmit the wrong power value to mqtt).
I use a 500A/50mV shunt ang get arround the same values as @jdevine82.
As i have seen, if the Batterie is arround 100% SoC the value is calculated wrong: 
[image]

If the SoC is lower than 100% all seems right.
[image]

Please excuse the strange calculation, as I log Ampare and Volt i nedded to calculate with them.

PS: Would be great to include the Shunt valued directly to grafana ;-)

stuartpittaway, 2022-07-18:
That's interesting, the SOC has no relationship to the power calculation! I will have to investigate further.

Chris87-ux, 2022-07-18:
> That's interesting, the SOC has no relationship to the power calculation! I will have to investigate further.

May its not reated directly to the SoC, but its interesting as the SoC gets down the calculation gets right ;-)

stuartpittaway, 2023-02-15:
I'll close this issue based on my tests comparing Victron SmartShunt vs DIYBMS shunt - didn't find any issues in calculations or SoC

closed 2023-02-15 by stuartpittaway