stuartpittaway/diyBMSv4ESP32 #107: Queuing bug in ESPController

smurfix, 2022-04-06:
Hi,

`ESPController/src/main.cpp` has this:

```
void onPacketReceived()
{
  PacketStruct ps;
  memcpy(&ps, SerialPacketReceiveBuffer, sizeof(PacketStruct));

  if ((ps.command & 0x0F) == COMMAND::Timing)
  {
    // Timestamp at the earliest possible moment
    uint32_t t = millis();
    ps.moduledata[2] = (t & 0xFFFF0000) >> 16;
    ps.moduledata[3] = t & 0x0000FFFF;
  }

  if (!replyQueue.push(&ps))
  {
    ESP_LOGE(TAG, "Reply Q full");
  }
  // ESP_LOGI(TAG,"Reply Q length %i",replyQueue.getCount());
}
```

Um, you're pushing the address of a variable that's on the stack onto a queue. This is a serious bug and you're probably *very* lucky that nothing overwrites your data before it's processed. Most of the time.

stuartpittaway, 2022-04-06:
Hi, thanks for taking the time to review the code. Amazing how these things can go unnoticed for so long, especially without any negative issues being reported previously.

The latest NOV21 branch has replaced the cppqueue classes with the native FreeRTOS ones, which should resolve this.

stuartpittaway, 2022-04-06:
Actually, just reviewing the cppqueue code, and the first thing it does is to memcpy the data to its own global buffer, so there shouldn't be an issue to resolve.

smurfix, 2022-04-06:
Ah. I totally missed that. Oops.

closed 2022-04-06 by smurfix