stuartpittaway/diyBMSv4ESP32 #109: NOV2021: main.cpp minor typos

virtuvas, 2022-04-23:
hi,

trying to figure out why my controller wont report correctly tail current, max V and efficiency, so ended up checking and enabling some debug lines in main.cpp:

lines 2179-80:
  ESP_LOGD(TAG, "WDog = %u", currentMonitor.modbus.watchdogcounter);
  ESP_LOGD(TAG, "SOC = %i", currentMonitor.stateofcharge);

should be both float (%f) or at least with float they show values as is are blank :-)
sorry not at a level to do pull requests (yet)

V.

stuartpittaway, 2022-04-23:
Watchdog should be an unsigned integer, but no real purpose publishing that.

Think soc is a float (from memory)

virtuvas, 2022-04-23:
%u gives 0
%f gives something like WDog = 1.833847 on mine.
Sorry confused that with last communication, true, pointless but it was there in the bulk of the debug messages.
SOC is definitely a float.

closed 2022-04-23 by virtuvas