stuartpittaway/diyBMSv4ESP32 #110: cannot save Current Monitor Basic settings for fully charged voltage, tail current and charge efficiency factor?

virtuvas, 2022-04-24:
hi,

current monitor working fine, showing the right values for Current, Shunt voltage and Power.
Add a charger, value goes to 58-59A (60A Victron Centaur charger) stop charging goes down to -0.2A (consumption of controller itself and a raspberry pi I have for VictronOS testing)
So all is fine as far as reporting values received from current monitor.

However I cannot save Basic Settings and see the values I put in after I hit the save button: 
Fully charged voltage
Tail current (Amps)
Charge efficiency factor %
first is around 89-99, the other two zero.

Now I think SOC reporting is related, was initially around 24%.
After couple of hours running the charger that should go up to circa 63%. However it only reduced as per the 0.2A consumption of the controller and rpi.
Playing with shunt current and shunt output voltage as well as battery capacity, I got at one point a SoC of 105%. No matter what, this value hasn't moved for over 24h, not even the nominal -0.2A is calculated.

Had a one hour look at the code (currently running latest NOV2021, but same thing happened with the main branch code before that) trying to find how I could hardcode the values for testing purposes, failed.

so any ideas welcomed!
FWIW, I'm now running the proper RS485 chip on the controller and the proper ADwhatever on the shunt. Same thing happens on both controller boards I have for testing with different ESP32s. Haven't tried the second current monitor board (fair amount of cables to shift over...) Further since system is reporting current correctly I cannot see where a fault might be. My understanding is that all calcs happen on the controller, so guess that's where I have to figure out the problem.

cheers

V.

stuartpittaway, 2022-04-24:
Is this the nov2021 code?

virtuvas, 2022-04-24:
yep, but same happened with the other ver. I had initially.
(or at least I think I had the main branch initially)

stuartpittaway, 2022-04-24:
Does the web interface show the green success bar when you click save?

If you have the usb logging open on the controller, does it report success on the rs485 command at the same time?

virtuvas, 2022-04-24:
yes, top right corner says settings saved (or something similar) also get this on the serial monitor:

```
I (96555696) diybms: Write multiple regs, success
I (96557537) diybms-webreq: API call: monitor2
D (96558072) diybms: Send addr=90, func=3, len=8
I (96558368) diybms: Task 1
D (96558473) diybms: Recv 97 bytes, id=90, cmd=3
D (96558473) diybms: WDog = 0
D (96558473) diybms: SOC = 105.000000
D (96558474) diybms: ChEffic = 0.000000
D (96558476) diybms: Volt = 26.631836
D (96558479) diybms: Curr = -0.817500
D (96558483) diybms: Temp = 21
D (96559542) diybms: total_free_byte=138560 total_allocated_byte=159332 largest_free_blk=86004 min_free_byte=113216 alloc_blk=451 free_blk=14 total_blk=465
I (96561400) diybms-webreq: API call: rs485settings
I (96561424) diybms-webreq: API call: currentmonitor
I (96561732) diybms-webreq: API call: monitor2
D (96563072) diybms: Send addr=90, func=3, len=8
D (96563473) diybms: Recv 97 bytes, id=90, cmd=3
D (96563473) diybms: WDog = 0
D (96563473) diybms: SOC = 105.000000
D (96563474) diybms: ChEffic = 0.000000
D (96563477) diybms: Volt = 26.631639
D (96563479) diybms: Curr = -0.842500
D (96563483) diybms: Temp = 21
I (96564868) diybms: Task 2
I (96565934) diybms-webreq: API call: monitor2
D (96568072) diybms: Send addr=90, func=3, len=8
D (96568472) diybms: Recv 97 bytes, id=90, cmd=3
D (96568472) diybms: WDog = 0
D (96568472) diybms: SOC = 105.000000
D (96568473) diybms: ChEffic = 0.000000
D (96568475) diybms: Volt = 26.631836
D (96568478) diybms: Curr = -0.826250
D (96568482) diybms: Temp = 21
I (96570129) diybms-webreq: API call: monitor2
I (96571368) diybms: Task 3, s=0 e=7
D (96573072) diybms: Send addr=90, func=3, len=8
D (96573473) diybms: Recv 97 bytes, id=90, cmd=3
D (96573473) diybms: WDog = 0
D (96573474) diybms: SOC = 105.000000
D (96573474) diybms: ChEffic = 0.000000
D (96573476) diybms: Volt = 26.631836
D (96573480) diybms: Curr = -0.821250
D (96573483) diybms: Temp = 21
I (96574328) diybms-webreq: API call: monitor2
D (96574543) diybms: total_free_byte=138120 total_allocated_byte=159724 largest_free_blk=86004 min_free_byte=113216 alloc_blk=455 free_blk=15 total_blk=470
D (96578072) diybms: Send addr=90, func=3, len=8
I (96578420) diybms-webreq: API call: monitor2
D (96578473) diybms: Recv 97 bytes, id=90, cmd=3
D (96578473) diybms: WDog = 0
D (96578474) diybms: SOC = 105.000000
D (96578474) diybms: ChEffic = 0.000000
D (96578476) diybms: Volt = 26.631836
D (96578480) diybms: Curr = -0.820000
D (96578483) diybms: Temp = 21
I (96582622) diybms-webreq: API call: monitor2
D (96583072) diybms: Send addr=90, func=3, len=8
D (96583473) diybms: Recv 97 bytes, id=90, cmd=3
D (96583473) diybms: WDog = 0
D (96583473) diybms: SOC = 105.000000
D (96583474) diybms: ChEffic = 0.000000
D (96583476) diybms: Volt = 26.631639
D (96583479) diybms: Curr = -0.850000
D (96583483) diybms: Temp = 21       
I (96584369) diybms: Task 1
```

V.

stuartpittaway, 2022-04-24:
Ok, definitely something strange there, the charge efficiency is set to zero, so you will never get a proper result/calc.

stuartpittaway, 2022-04-24:
Is the voltage and current correct?

virtuvas, 2022-04-24:
yep, that's the strange thing, voltage current are spot on, charge efficiency and tail current 0. Also wrong and massive is the fully charge voltage above!
I guess you have tried that on your tests, could it be something on my setup?
Even erased flash on the ESP32, re installed it, same thing.

Am I right in thinking that these calcs are taking place in the controller board so current monitor and comms are fine?

stuartpittaway, 2022-04-24:
The current monitor board does all the calculation, the controller simply display the values.

I suspect it's a bug in sending the values from the web interface into the current monitor

virtuvas, 2022-04-24:
started an incognito window on chrome, opened it up (just in case it's a cookie issue), same behaviour.

virtuvas, 2022-04-25:
hm, most confusing!
Reinstalled FEB22 Master on same controller and shunt. Got the interface up on an incognito window (cookies), wonder if there's an easy way to delete only diyBMS cookies so I can go up and down code vers without issues!
Enabled current monitor
ticked temp coeffic enabled on the Relay settings
then edited fully charged voltage, tail current charge efficiency at the basic settings. Accepted them.

Well, really don't know what's going on (nor if editing relay settings before basic settings triggered something. 
If I've got that right, these settings are set in the webinterface and following the controller passes them over to the current monitor board to be stored. Could it be that since I'd not setup the current monitor with FEB22 release code (it's a possibility!), there is indeed something in NOV21 that breaks it?
To test that I first edited tail current from 2 to 3, accepted it fine.
Next reuploaded NOV21 code, same process, basic settings are right where I left them with FEB22.
HOWEVER, if I touch one setting I'm back to 98 fully charged voltage, and 0 for the tail amps and charge efficiency.
So looks like there's indeed something to check on NOV21 code.

BTW#1: on compiling NOV21 code I get the following, I guess nothing to worry about right?
```
LittleFSBuilder.py using x86_64-w64-mklittlefs.exe
Compiling .pio\build\esp32-devkitc\src\HAL_ESP32.cpp.o
Compiling .pio\build\esp32-devkitc\src\PacketReceiveProcessor.cpp.o
Compiling .pio\build\esp32-devkitc\src\PacketRequestGenerator.cpp.o
Compiling .pio\build\esp32-devkitc\src\Rules.cpp.o
Compiling .pio\build\esp32-devkitc\src\influxdb.cpp.o
Compiling .pio\build\esp32-devkitc\src\main.cpp.o
Compiling .pio\build\esp32-devkitc\src\mqtt.cpp.o
Compiling .pio\build\esp32-devkitc\src\tft.cpp.o
Compiling .pio\build\esp32-devkitc\src\victron_canbus.cpp.o
In file included from src/tft.cpp:33:
.pio/libdeps/esp32-devkitc/TFT_eSPI/TFT_eSPI.h:875:6: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will nCompiling .pio\build\esp32-devkitc\src\webserver.cpp.o
ot be available! [-Wcpp]
     #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      ^~~~~~~
Compiling .pio\build\esp32-devkitc\src\webserver_helper_funcs.cpp.o
Compiling .pio\build\esp32-devkitc\src\webserver_json_post.cpp.o
Compiling .pio\build\esp32-devkitc\src\webserver_json_requests.cpp.o
Generating partitions .pio\build\esp32-devkitc\partitions.bin
Compiling .pio\build\esp32-devkitc\lib31b\FS\FS.cpp.o
Compiling .pio\build\esp32-devkitc\lib31b\FS\vfs_api.cpp.o
```

BTW#2 any easy way to know what code base one is running? if you compile yourself as I do compilation date is not enough!
Now that web interfaces are identical, I've no other clues it seems.

cheers

V.

stuartpittaway, 2022-04-25:
Thanks for that feedback, very useful.

Not sure if anyone has actually tried the temperature compensation on the shunt, it may cause more problems than it solves. Try leaving it disabled for now.

virtuvas, 2022-04-25:
will do, the reason for ticking the box was that I thought I'd try to save some other states than what was failing previously, hoping to reset or change something and get the system working...
Looking forward to the patch :-)

stuartpittaway, 2022-04-25:
Ok. On the ```TOUCH_CS pin not defined``` issues - can you close down VSCODE and then delete the ".pio" folder from the DIYBMS controller folder.

This will force VSCODE/Platformio to download all the libraries again - you may have an old one cached.

stuartpittaway, 2022-04-25:
Actually, ignore me, I've got an older version - looking at the code, that message can be ignored.

stuartpittaway, 2022-04-25:
Simple typo in the code, so please try this NOV21 version.

virtuvas, 2022-04-25:
thanks stuart, obvs works now.
Can you point to where the actual reset and SoC behaviour is explained to some detail please?
Now all values are set and can be edited/saved, fine.
However, SoC is still 105% so I lowered the fully charge to 27.5V to accelerate the resetting to 100% SoC and started the 60A charger. 
Two observations: 
SOC is still at 105% even when shunt voltage went up to 27.9V.
shunt voltage has something like 0.4V diff to bank voltage when charging and 0.05V when idle.
Am I right in thinking that all conditions must be met to reset SoC?
I mean shunt voltage must be over the fully charge voltage value, AND shunt current LOWER than the tail current, right?

V.

stuartpittaway, 2022-04-25:
Correct.

The charge voltage must be higher than the tail voltage, and the current lower, for a about 3 minutes.

This should then trigger the SOC to reset to 100%

Note if the current or voltage are bouncing up and down, then the 3 minute timer will be reset.

virtuvas, 2022-04-25:
OK, and any idea how I got SoC 105%? (don't ask don't remember what values I put on bank size, shunt amp size, and bank voltage!)
I'm asking as it's also stuck there and wont even drop with my normal -0.2A 247 I have from the powersupply and rpi. It's stuck there for 2days now, when it was originally at 24% it dropped to circa 23.5% or so within a day.
I guess you have some error code checking for valid SoC value, maybe it should check and "limit" SoC to something a bit more reasonable. It looks like it managed to get off bounds, and now neither charging or discharging alter it...

V.

virtuvas, 2022-04-25:
so, upped tail current to 59A :-) and with voltage 27.5V now system resets every 3mins and goes back down to 100%
but as battery is really at circa 75%, victron still pumps 58A, so within the 3mins SoC goes to 101% and then down again.
Understand the logic, worried that will be odd for a user to see a SoC value higher than 100. 
Maybe think about it!
I'd be tempted to just not let this value go over 100 fullstop and no matter what.
And bring up an error message saying careful, SoC mismatch due to whatever...

closed 2022-04-26 by virtuvas