stuartpittaway/diyBMSv4ESP32 #111: Current drain due to ESP serial TX not being disabled

smurfix, 2022-04-25:
Hi,

the battery modules disable their serial transmitters when they're "done". The problem is that the controller does not. This causes an additional constant 1mA drain to the first module.

NB, re-designing the serial comms with TTL-serial active-low semantics in mind, so that "idle serial line" == "no current", would be a splendid idea – but that ship has sailed I'm afraid …

stuartpittaway, 2022-04-25:
Agreed, originally it was expected that the esp could disable the tx line after transmission, this has not been explored fully at this time.

virtuvas, 2022-05-05:
+1 please,

my 304Ah@24V lifepo4 bank stored at circa 30% for a couple of months (while testing various controller comms and settings and waiting for the shunt chips to arrive) had a circa 2-3Ah mismatch between the first cell and the other 7 when I top balanced it again last week. Not a massive deal, but if storing banks in boats/rvs offseason and want to be able to monitor them, it would be nice to have.

virtuvas, 2022-06-06:
[image]

after storing the bank for a week or so at 48% SoC, recharging it slowly for the last couple of days to reach 100% SoC this afternoon (via solar) it reached the balancing point (3.45V) and cell one is around 45mV down on the others. Throughout charging discharging cycles and idling, it's within 4-5mV.
After a few hours of slowly pushing the cells towards balancing point, it will eventually "catch up" maybe tomorrow afternoon as it stands now.
I know it's mentioned above just wondering where it's priority stands within your to do list 😀 

cheers
V.

stuartpittaway, 2023-02-21:
> I know it's mentioned above just wondering where it's priority stands within your to do list 😀

This has recently been talked about - unfortunately without changing the electrical design of the controller and modules, its not possible to sort out this 1mA drain on cell zero (first one).

If storing for long periods of time, I'd recommend disconnecting the power to the modules - but know this isn't ideal.

smurfix, 2023-02-21:
Well, you could always turn off the controller's serial port, use the Tx output as a GPIO, and set it to low.

However, why don't you switch to active-low in your new designs? You could even make them compatible with the "old" modules, that'd require adding five jumper pads, but if you put them on the back they don't take up any space at all.
[image]