stuartpittaway/diyBMSv4ESP32 #112: NOV21: rules blue circle is on, with no active rules.

virtuvas, 2022-04-27:
two problems with rules stuart (each one on a separate issue),

over the last almost 2 months of testing NOV21, I've had something like 4-5 occasions where the blue circle with a white 2 in it turns on (and one of the mechanical relays latches- that's how I notice it!) for say 5-10mins then turns off.
during that time, no relay on/off policy is setup and as seen from the screen capture, nothing is out of bounds anyway!

[image]

V.

virtuvas, 2022-04-27:
hm, 1:00am right now in Greece (apparently 11:00pm in the UK which is GMT+1, thing did it again for around a minute only. So guess it's just a matter of sorting out saving rule settings as per issue #113 
At some point I'd like to understand the whole timezone mess, too busy right now...
so 11:00PM GMT+1 is 12 as far as diyBMS is concerned? Shouldn't that relate to the timezone I setup? totally confused!

[image]

cheers

V.

stuartpittaway, 2022-04-28:
Perhaps the timezone is also broken in this release?

virtuvas, 2022-04-28:
morning, well, tbh I can change values on the timezone and no matter what I get what *I think* is GMT.
like just now with timezone hour at 2
I know time is 9:51am in EET I see:
Time now is 2022-04-28T06:51:10.000Z

stuartpittaway, 2022-04-28:
Ok, another one to add to the list!

If you take a look at the rules page, there is a ```Minutes since midnight now is: 414``` value.

Is this accurate for you ?

virtuvas, 2022-04-28:
good thing is data passed to influxdb and shown on grafana are timestamped correctly for now. not shure how that works..
535 for me right now

stuartpittaway, 2022-04-28:
Ok, interesting, as thats also worked out internally from the controller.  The ```Time now is 2022-04-28T06:55:46.000Z``` on my system is currently also an hour behind (British Summer Time = GMT+1).  So potentially Time now is only showing the GMT value, I'll need to dig into the code, and perhaps change the label from "time now".

virtuvas, 2022-04-28:
yep, still wonder why it did it's trick at 1:00AM EET instead of local 12.
I think it would make sense to present hours in the Rules/NTP page to local timezone/daylight saving settings. TBH, I find it so confusing that I usually just play with settings until what I see on the device is what my watch shows, then I'm happy that it's working 😀

stuartpittaway, 2022-04-28:
There is a bug where the reply from the controller is not updating the Timer1/2, the reply is truncated so not returning data correctly.

virtuvas, 2022-04-28:
not quite sure that I'm missing now, but applying these two fixes, I now can change the relay states and it stays so, which is good.
However, on NPT settings I get a line saying:
Controller time is

doesn't pick  the timenow variable?

However, #113 is not sorted, so no matter what you put in a Rules, reset value cell, after a save and refresh of the page you'll be back in the trigger value. Also if you change the trigger value, after a save/refresh, the reset value will be the same as the new trigger.

virtuvas, 2022-04-28:
doh, missed one of the patches, will report back!

virtuvas, 2022-04-28:
OK, both issues sorted, so ATM I'm at EET and summer time and shows 16:49
thanks again Stuart!

V.

closed 2022-04-28 by virtuvas