stuartpittaway/diyBMSv4ESP32 #115: NOV2021: cannot change wifi settings (SD card installed, wifi.json file with old settings)

virtuvas, 2022-05-01:
I have a funny feeling that only me is testing NOV2021 branch changing all type of setups😀 

moved bank to the boat, different SSID name there (same passwd as home)
turned system on, hooked up laptop with VSC terminal on, spacebar, blah blah, selected wifi, added passwd, rebooted, no connection! controller reporting cannot connect to AP or something along these lines.
Did it a few times nope. 
Shut down, remove SD card, fitted on laptop, checked /diybms/wifi.json, still had the old home wifi settings.
Replaced with new settings, back in controller board, booted up just fine, straight into boat wifi.

Now, two issues:
A. cannot update settings from serial connection, guess easy to fix?
B. I find it hard to have a system that unless is connected to a wifi, it will constantly be rebooting (effectively not being a BMS at all!) Since some of us are installing such setups on boats/rvs etc, I'd expect the system to simply function doing what it's programmed to without internet (I guess it does!) but also with a failed/broken/disconnected router. Is there something that needs to be configured for that Stuart?

cheers

V.

stuartpittaway, 2022-05-01:
Ok, with the SD card installed it will attempt to reconfigure using the WiFi details stored on that, probably need to look at the sequence of boot up actions to prevent this.

stuartpittaway, 2022-05-01:
The controller is not designed (currently) to operate in a disconnected fashion. It expects to be able to connect to WiFi hotspot/access point.

virtuvas, 2022-05-01:
> The controller is not designed (currently) to operate in a disconnected fashion. It expects to be able to connect to WiFi hotspot/access point.

pretty sure some users wont be v.happy if router fails and system doesn't balance or whatnot messing the cells.
AFAIK, on board, I'd expect the BMS to communicate with Victron kit, and "organise" the charging/discharging regime that Victrons have to follow. No need for wifi doing that. Other than that, I'd expect if temps rise, v rise etc rules taking over the getting the bank offline if needed via the contactor (which is wired straight to the controller)

Genuine Q, once settings are configured properly with system online, what are wifi comms really needed for? Not talking about viewing condition (could just check on the touch screen!) is there anything else (bar influx/mqtt) needing comms?

cheers

V.

stuartpittaway, 2022-05-01:
Absolutely agree that the controller should operate standalone, just the code not written to fully support that.

It should work whilst not connected to the WiFi/internet, the balancing is controlled by the individual modules, so technically you don't even need a controller!

However for the victron Comms and canbus etc. The controller is required

virtuvas, 2022-05-01:
> Absolutely agree that the controller should operate standalone, just the code not written to fully support that.
> 
> It should work whilst not connected to the WiFi/internet, the balancing is controlled by the individual modules, so technically you don't even need a controller!

lol, you're right, never crossed my mind!

is this de-wifization of the controller code part of the NOV2021 code updates or will be another project?

V.

PS. going to be using a 2pair bus cable to connect the controller to the Victron (just over 1m apart) Cable has 4 cores, black and the two needed for the comms, fourth not used. Is it a problem if I use that fourth (and the GND going to canbus) to also power the controller from the central power panel? Guess not an issue, as canbus GND isn't isolated, right?

stuartpittaway, 2022-05-01:
The cable should be okay, the canbus is not isolated as you said.

virtuvas, 2022-05-04:
Currently onboard, all working fine, few secs after I unplug the power from the gsm/wifi router onboard, controller starts rebooting, goes through the normal procedure, comes up with the normal screen after all modules comms are fine, and five ten secs later reboots and goes through the whole lot again...
So short answer is yes 😁

virtuvas, 2022-05-04:
Mind, I do have influxdb and victron can running.

virtuvas, 2022-05-04:
Maybe an option in settings like: 
Reboot when wifi connection lost 
Would be enough and then us boaters can disable it...

atanisoft, 2022-05-04:
If you have InfluxDB enabled and WiFi drops it may reboot due to the loss of WiFi signal, it is probably worthwhile to have a "is online" check wrapping the InfluxDB/MQTT/etc calls and prevent them from going out at all when the connection is down.

stuartpittaway, 2022-05-05:
Ok, the WiFi issue is a code bug in the controller logic.

When the WiFi drops, on re-connect it attempts to call ```ESP_ERROR_CHECK(esp_event_loop_create_default());```, which has previously already been called and triggers ```abort() was called at PC 0x4008e59f on core 1```

stuartpittaway, 2022-05-05:
The controller reboot on lost WiFI was simple enough to fix, however the MQTT calls seem to (randomly) fail after the wifi is lost and then reconnects.

More debugging to do....

stuartpittaway, 2022-05-05:
Ok, think its all working now, over to @virtuvas to test :-)

closed 2022-05-05 by virtuvas