stuartpittaway/diyBMSv4ESP32 #116: SoC rising (slowly) with battery only powering the shunt?

virtuvas, 2022-05-09:
small one,
battery onboard, BlueSea disconnecting it form loads/charging.
controller powered from the main el.panel onboard which takes power now from the existing battery (not even sharing GND with canbus cabling as my bus cable was finally a proper Cat5one so 4pairs in it!).
current monitor OTOH is powered by the battery itself:
current monitor GND goes to lifepo4 -
current monitor V+sense & ++++ are bridged (as I've seen @stuartpittaway did in one of his videos) and goes to shunt next to it 
Was initially on the battery side of the shunt, had a constant +0.017A, so battery according to current monitor was "charging" being disconnected...
Moved the power to the consumption side of the shunt, similar albeit slightly lower +0.013A
SoC was slowly rising over the couple of days that I had battery disconnected: went up from 75.09% to 75.22% in 24h.

Guess I've got to separate ++++ and V+Sense? 
where to? 
Easy enough to power the shunt also from the main el.panel (GND and ++++ routed to the +5 GND on the controller?)
then V+ Sense to where?

stuartpittaway, 2022-05-09:
Do you have the battery connected to the shunt terminals correctly?

You should have a negative current when discharging the battery - the 0.017A (17mA) is the power the current monitor is using!

virtuvas, 2022-05-09:
well, that's why I've posted this.
They way I have it, I'd expect to show the consumption of the current monitor sucking a bit of power from the disconnected bank.
[image]
was initially like that when SoC was increasing by 17mA. Then moved the grey wire to the other side.
Don't freak out, wires are temporary waiting for the controller to be mounted next to it.

stuartpittaway, 2022-05-09:
Try moving the grey wire to the actual battery terminal (on the battery) rather than tapping off the shunt.

Also be careful with not cracking the PCB if you over tighten those screws!

virtuvas, 2022-05-09:
so both V+ SENSE and V++++ to the battery terminal V+? actually battery is just below the shunt and the only thing bypassing is a class-T fuse 40cm to the left of the shunt. My plan was not to have anything on the V+ other than the class-T fuse. Can do it for a test.
would it make sense to have V+ and GND to the controller power and just the V+ Sense to the battery V+ or will that mess things further?

True, v.careful with that, actually plan to remove the shunt off the PCB and bolt it alone on the wooden beam where it stands now and lift the PCB above. It's 50mm2 cables there, no fun!

cheers
V

virtuvas, 2022-05-11:
OK, rewired the lot with V+++ going to the consumption side of the shunt and V+Sense going to the V+
no difference, 0.012-0.013A CHARGE on the bank... so bank is idle, running the current monitor board AND charging whilst doing that, not quite right.
ideas welcomed!

cheers
V.

stuartpittaway, 2023-02-21:
Hi @virtuvas, have you tried the latest shunt code?  I re-wrote a lot of the SoC calculation so this might help.

virtuvas, 2023-04-24:
@stuartpittaway  sorry, been slow. Managed only yesterday to upload new code on both the ESP (was running last March code for a year with no crashes btw!) and the shunt.
Exactly the same thing happens +0.012-0.013A shows on it.
Will have again a look at the wiring and come back to you on that (although tbh it's not a big issue)

OTOH, sucking current from first cell when left idle for months and the imbalance it creates bothers me more off-season but I understand we can do nothing about it atm

cheers
V.