stuartpittaway/diyBMSv4ESP32 #120: Ability to run in AP Mode?

jdevine82, 2022-05-12:
Hi 
I am trying run this without a wifi network available. Can I set this to run as AP Mode and run the normal pages? The code seems to support it but I cannot seem to find a setting anywhere to allow it. It asks to select a wifi network and I cannot skip it?
Thanks for any help.
Jason

pkathey, 2022-06-17:
Recently closed issue #128 has a comment stating "AP mode is not supported in the latest firmware" and perusal of the ESPController code confirms very few references to "SoftAP".

However, for those hoping to use diyBMSv4ESP32 in standalone applications (where there is no available wifi network) does the comment above ( "Not at the moment" ) imply that AP mode has _not_ been ruled out altogether?

Perhaps the vast majority of diyBMS users will likely never need AP mode (except for initial setup in lieu of serial) though suspect there are some who would prefer it - e.g. in an RV (Recreational Vehicle) application. In that case an AP is required but not all the time so ability to enable/disable the AP as/when required while having the controller continue on in the background would be a useful feature.   Understand this would likely be a 'bridge too far' though if the basic AP functionality was there am sure one could figure out some sort of switch/GPIO hack to manage the AP enable/disable.

laserranger3000, 2022-08-15:
I'd also appreciate this functionality. Especially in off-grid applications it would be useful, since there's not necessarily any other wifi

pkathey, 2023-10-20:
Hi @stuartpittaway any chance of a genuine A.P mode in the not too distant future?

stuartpittaway, 2023-10-21:
Hi @pkathey I do plan to include this, just hasn't been a priority so far.