stuartpittaway/diyBMSv4ESP32 #123: relay pulse goes on for ever when activated?

virtuvas, 2022-05-14:
found this bit of code in main.cpp

```
void pulse_relay_off(TimerHandle_t xTimer)
{
  for (int8_t y = 0; y < RELAY_TOTAL; y++)
  {
    if (previousRelayPulse[y])
    {
      // We now need to rapidly turn off the relay after a fixed period of time (pulse mode)
      // However we leave the relay and previousRelayState looking like the relay has triggered (it has!)
      // to prevent multiple pulses being sent on each rule refresh
      hal.SetOutputState(y, RelayState::RELAY_OFF);

      previousRelayPulse[y] = false;
    }
  }

  // Fire task to record state of outputs to SD Card
  xTaskNotify(sdcardlog_outputs_task_handle, 0x00, eNotifyAction::eNoAction);
}
```
comment implies that when relay is triggered it only pulses once, stays in triggered mode but doesn't pulse all the time.
well it looks like it does pulse all the time in practice.

shouldn't it be
```
hal.SetOutputState(y, RelayState::RELAY_ON);
```

cheers

V.

stuartpittaway, 2022-05-14:
Ok. The actual relay state in the variable should be on. However the relay should only pulse.

virtuvas, 2022-05-17:
discard, me being silly, had only edited the trigger value to de-activate the relay and check that cabling works, and had forgotten to also edit the second restore value...
Checked again today, all is working fine!

V.

closed 2022-05-17 by virtuvas