stuartpittaway/diyBMSv4ESP32 #124: platformio.ini [env] platform = epressif32   compilation fails?

virtuvas, 2022-05-14:
cannot compile new code!
platformio.ini:
```
[env]
framework = arduino
; 4MB FLASH DEVKITC
platform = espressif32
;platform  = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```
gives:
```
Archiving .pio\build\esp32-devkitc\libFrameworkArduino.a
Linking .pio\build\esp32-devkitc\diybms_controller_firmware_espressif32_esp32-devkitc.elf
c:/users/vassilis/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-devkitc\diybms_controller_firmware_espressif32_esp32-devkitc.elf] Error 1
===================================================== [FAILED] Took 84.63 seconds =====================================================
Environment    Status    Duration
-------------  --------  ------------
esp32-devkitc  FAILED    00:01:24.627
================================================ 1 failed, 0 succeeded in 00:01:24.627 ================================================The terminal process "C:\Users\Vassilis\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

old style platformio.ini:
```
[env]
framework = arduino
; 4MB FLASH DEVKITC
;platform = espressif32
platform  = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```
compiles fine.
what am I missing?
laptop onboard yesterday compiled new code without issues, this one here at home wont no matter what!

cheers

V.

stuartpittaway, 2022-05-14:
Delete the .platformio folder in your windows user home directory and try again.

It's probably got a cached old version of the esp library

virtuvas, 2022-05-14:
ffs! that was 70k files and close to 5Gig! 
I'd previously deleted toolchain-xtensa32 dir as it's mentioned in the error log above to no avail...
compiled fine (eventually!) after re downloading almost 2Gig.

cheers
V

omixxer, 2022-05-23:
I have the same compilation fail and after deleting the folder it continues appearing. 
[fasfs.txt](https://github.com/stuartpittaway/diyBMSv4ESP32/files/8753941/fasfs.txt)

desegura, 2022-05-23:
same problem here @stuartpittaway
what will be the workaround to fix it? 🤔

stuartpittaway, 2022-05-23:
First question then, why do you need to compile it manually?  Are you making changes?

If not, please use the pre-compiled one from GITHUB

desegura, 2022-05-23:
Yes I'm just trying to make some changes to bring new functionalities to the project in the future
I was wondering if more users here have had too this compilation final link failed bad value?

stuartpittaway, 2022-05-23:
> Yes I'm just trying to make some changes to bring new functionalities to the project in the future I was wondering if more users here have had too this compilation final link failed bad value?

Only the people in this email thread have mentioned it.  The ESP IDF/Arduino libraries jumped up a significant number of revisions recently, and I don't think platformio handled it very well.

As @atanisoft mentioned, if you open up a terminal you can do a ```pio upgrade```, my PC reports

```
You're up-to-date!
PlatformIO 6.0.1 is currently the newest version available.
```

This alongside deleting ```packages``` and ```platforms``` from the ```.platformio``` folder usually helps.

stuartpittaway, 2022-06-24:
@desegura did you get this working in the end?

desegura, 2022-06-26:
Not yet, I haven't had time to continue working on it, but I'll try again soon

closed 2022-09-06 by stuartpittaway