stuartpittaway/diyBMSv4ESP32 #125: touch screen reboots controller

virtuvas, 2022-05-14:
OK, test system 4S2P 18650 cells.
just controller no shunt.
ESP32 powered from usb.
VSC running, just compiled, checking the serial monitor. few mins later, screen off, if I touch the screen system reboots and I get this on the VSC monitor:

```
E (150914) diybms: Short packet 0 bytes
I (150971) diybms-rules: Set warning 1
I (151745) diybms-webreq: API call: monitor2
I (152165) diybms: Task 1
I (153971) diybms-rules: Set warning 1
D (155665) diybms: Send addr=90, func=3, len=8
E (155915) diybms: Short packet 0 bytes
I (156746) diybms-webreq: API call: monitor2
I (156971) diybms-rules: Set warning 1
I (158665) diybms: Task 2
Exception in thread rx:
Traceback (most recent call last):
  File "C:\Users\Vassilis\.platformio\python3\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Vassilis\.platformio\python3\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Vassilis\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "C:\Users\Vassilis\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
```

V.

virtuvas, 2022-05-14:
each touch no matter screen on or off. push button works fine waking up screen and toggling through subscreens.
First time I get this, never had it before in any code I've run.
Didn't have it yesterday on same code 8S lifepo4 bank on board with proper 3A power supply.

stuartpittaway, 2022-05-14:
I need to see the esp error message after the line...

Exception in thread rx:

Does the reboot also occur using the precompiled version from GitHub?

stuartpittaway, 2022-05-14:
Do you have the latest PCB with the buttons?

virtuvas, 2022-05-14:
```
Exception in thread rx:
Traceback (most recent call last):
  File "C:\Users\Vassilis\.platformio\python3\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Vassilis\.platformio\python3\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Vassilis\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "C:\Users\Vassilis\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
```
sorry for being slightly thick, how do I upload the precompiled version to the controller on a win10 pc ?
I'll get the bench power supply to power the thing and see if it still does it.

pcb with two buttons on the bottom right hand side. and one set of connectors built in Nov/Dec iirc

stuartpittaway, 2022-05-14:
Use the nodemcu tool on windows, just like I did in the video

virtuvas, 2022-05-14:
well,

controller alone, runs fine draws 0.2A@5V on my powersupply.
with touch screen on, goes up to 5A@5V and that's the max this small powersupply can provide, system doesn't start.
that's a 2.8 TFT SPI 240X320 v1.2 screen (as recommended by you)
What's the expected draw of this thing? over 5A or is mine dodgy?

V.

virtuvas, 2022-05-14:
apologies!

checked the screen, there's an SD card underneath, on this board the diode next to the fuse was soldered (ok by me, stupid!) too high and wire off the diode was fouling on the sdcard metal surrounding which I guess is GND shorting the thing.
So, what was happening before was as soon as I touched the screen it was pushed down a bit, shorting diode to SDcard, voltage drop, reboot Controller. Bit of insulation tape at the back of the SDcard, all's fine board consuming 0.25A

now it would be hard to even try to achieve that!
sorry Stuart 😞 

V.

closed 2022-05-14 by virtuvas