stuartpittaway/diyBMSv4ESP32 #130: TCA9534APWR_INPUTMASK should be B10110000  on PCB with two pushbuttons SW1 & SW2 ???

virtuvas, 2022-05-26:
trying to understand the code and see how I can get TCA6408 to read the Input A B C D (and then pulse RE2 as per #121 )

anyway, considering that there are two push buttons on my 4.3 controller PCB to wake up the TFT screen, BOTH their pins be inputs.
SW1OUTPUT is P4
SW2OUTPUT is P6
E.Stop is P7
currently mask is 10010000 (p7 and p4 are input) 
SW2 wasn't waking up the TFT, changed mask to 11010000, now works fine.

Not sure how/if you use P6 in single SW PCBs though!

cheers
V.

stuartpittaway, 2022-05-26:
Well this is a bit of a grey area - V4.3 of the controller was never officially released - what version of the PCB do you have?  Should say "beta-XXXXXX" somewhere on the silkscreen.

I think the A-D inputs/outputs are wired into the other chip, as they run off 5v.

The bit mask was changed to make these outputs to prevent a lot of spurious interrupts triggering.

virtuvas, 2022-05-26:
I picked up a v4.3 from github, then modded it to enable the use of DIP chips for CAN bus (which I'm using) and TCA equiv (which I never found!), so version is completely unknown :-) Does work though!
No worries, changed the mask and second button works, no big deal having only one button working tbh.

I know that IO A to D are on TCA6408 and not TCA9534 where are these buttons, was just looking for the right bits of code to copy over and edit 😁 

turns out you have the basic code there , but I've not figured out why 
interrupt_task() is not triggered when pulling down A B C or D (having changed their mask TCA6408_INPUTMASK B11110000 )
I'm missing something somewhere, expect another Q later tonight unless you can point me in the right direction 😀 

looks like:
```
    if ((ulInterruptStatus & ISRTYPE::TCA6408A) != 0x00)
    {
      ESP_LOGD(TAG, "tca6408_isr");
      // Read ports A/B/C/D inputs (on TCA6408)
      ProcessTCA6408Input_States(hal.ReadTCA6408InputRegisters());
    }
```
is not triggered, whereas 
```
    if ((ulInterruptStatus & ISRTYPE::TCA9534) != 0x00)
    {
      ESP_LOGD(TAG, "tca9534_isr");
      // Read ports
      // The 9534 deals with internal LED outputs and spare IO on J10
      ProcessTCA9534Input_States(hal.ReadTCA9534InputRegisters());
    }
```
is working fine.

cheers
V.

virtuvas, 2022-05-27:
not sure why it wasn't working last night, with clear mind this morning it works. It works as in when pulling one of the pins down to GND it triggers.
Just unmasked B and C for now (will only need one tbh). Plan is to get the Victron battery monitor inbuilt and programmable relay to get the battery online (via the pulsing signal to the contactor) when V is below set high level.

diyBMS rules deal nicely with disconnecting the battery when it's over/under charged.

just a bit of code from the rules to steal and job done.
Considering that there's a danger of someone hooking up a 12/24 or even 48 battery "signal" in one of the inputs and burn the whole thing and since it can only easily function in receiving pull down (or pullup tricky!) signals I guess there's no point in incorporating any of that in the Rules?
planning to pull up with 100kOhm resistors the input I'll be using

cheers

V.

stuartpittaway, 2022-05-27:
The inputs are designed as a pull to ground / active low signal.

If there are any unused inputs, I would connect them to ground with a resistor as you mention.

virtuvas, 2022-05-27:
yes, will just splice the GND which goes to the controller board to the Victron battery monitor relay, and take that to say B. leave all others as outputs. Nothing more.
As I'm going to be using Node-RED, I may later on get two GPIO pins off the rpi running VenusOS and have a full remote control on the lifepo4 bank. Like this week wasn't doing anything onboard, so disconnected the bank at 47% SoC FLA batteries doing all the work. If I wanted to do that remotely, I'd have to enter diyBMS web interface, fake some Rule to turn off the contactor.
V.

closed 2022-05-29 by virtuvas