stuartpittaway/diyBMSv4ESP32 #136: Suggestion: Additional Rules

Torsdan, 2022-07-28:
- Rule for Cell Range.
- And in our use case, it would also be beneficial to have an additional 'Individual Cell Over Voltage' rule.

The reason for dual CellOV would be so we can set one tell the charge controller to limit input current, giving the balancers time to do their work. And a second one at a higher V, telling the charge controller to shutdown.

- Totally not important part but OCD related, 'Individual' prefix added to 'Cell under voltage' or removed from 'Cell over voltage'.

Thank you for your time.

Torsdan, 2022-07-28:
While I think of it, would it also be possible to have the Bank Voltage and Bank Range sent to Influxdb as well?

stuartpittaway, 2023-02-14:
Rule for cell range has now been added.