stuartpittaway/diyBMSv4ESP32 #139: Relay type "pulse" not respected after reset

laserranger3000, 2022-08-15:
I set the relay type to "pulse" and the relay default to "on". This works during normal operation.
But when the controller is reset, the relay signal is permanently on. This causes my bi-stable relay to get hot.
I assume that during the boot sequence, the setting "pulse" is not respected.

stuartpittaway, 2023-01-01:
Hi @laserranger3000, is this still a problem with the latest code?

Having the relay default to "on" and "pulse" is not a supported setup, so the code should prevent this from happening.

Pulse is used to trigger an "on" + "off" event, rather than an "off" + "on".

laserranger3000, 2023-01-23:
Hi, I can only test the new code in a few months time unfortunately.
However, as far as I understood the code change, it's still not possible to set a pulse relay to on after boot.
My workaround at the moment is a physical button that has to be pressed manually after startup that switches the relay to on once.

stuartpittaway, 2023-01-23:
> However, as far as I understood the code change, it's still not possible to set a pulse relay to on after boot.

Correct - that's not supported, unless one of the rules triggers it to "on".