stuartpittaway/diyBMSv4ESP32 #140: Suggestion: reduce ESP32 clock frequency

laserranger3000, 2022-08-15:
The CPU frequency could be reduced to save power.
Especially for (my) offgrid application where there is sometimes no charge current for months, the self-discharge due to the controller consumption becomes significant.
I've added this line to the platformio.ini and saw a current decrease of the controller on 5v from about 200mA to 170mA:
`board_build.f_cpu = 80000000L`
Even lower frequencies didn't work in the platformio environment, but should be technically possible.
Another idea would be a function to disable the wifi radio. Maybe controlled by a switch connected to the external IO ports.

stuartpittaway, 2022-08-15:
Good idea, would have to check the CPU still performs okay with the reduced speed.

This probably links into #120 perhaps we could use one of the buttons on the controller to "wake" up WIFI?

laserranger3000, 2022-08-15:
At 80 Mhz I didn't notice any negative impacts on the functionality or web-interface performance.

Yes, a button to wake up wifi would be awesome. It could go back to "sleep" after 30 minutes or so (should be enough time to check the status  and change settings)

pkathey, 2022-08-15:
+1 for either functionality/button to disable/sleep wifi OR (preferably) introducing an A P mode that is only enabled when it's needed.  On the latter, there are some spare GPIO and (in my case) am not too fussed about doing a wee bit of soldering to cobble in a switch (particularly since my controller has none being an early model) OTOH, for those with early controllers (and no existing switches), maybe some hardware hacking (to add switch or two) is preferable to upgrading the board (or migrating components to newer board) if wifi sleep option is chosen.

laserranger3000, 2022-08-22:
I was playing around in the code a bit, but I don't fully understand how it works yet...
To disable the wifi after a few minutes I tried to call this function with a timer in void loop() (including the <WiFi.h> library):
`WiFi.mode(WIFI_OFF);`
But it doesn't do anything.
The only way I could disable Wifi at all was to disable this function in void setup();
`wifi_init_sta();`

Can you give me a hint where I should edit the code to dynamically enable/disable wifi? is it even straightforward?

By the way, I'm down to 50mA with 80 Mhz clock speed and disabled wifi, it's promising :)