stuartpittaway/diyBMSv4ESP32 #143: No German translation

s1738berger, 2022-08-24:
If I choose German it shows up as **Language: de** on the page **About**, but no texts are shown in German.
Other languages seem to work as expected. I use the prebuilt version from 2022-08-16, `53fc96cd8b1e992f76d34a80da93ae04e0edccb0`.
Was the same bhaviour in earlier versions as well.

moscito1010, 2022-09-05:
Stuart, how can we help you? That shouldn't be a problem for us to translate all the texts into German for you. Is there a language file that we can edit?

stuartpittaway, 2022-09-06:
This is the file where the translation texts reside..

https://github.com/stuartpittaway/diyBMSv4ESP32/blob/master/ESPController/web_src/lang_de.js

It would be a case of editing that file, which can then be "pushed" into GITHUB for review/approval.

stuartpittaway, 2022-09-06:
The translation should now work, if you upgrade the firmware!

s1738berger, 2022-09-10:
It works now. The grman translation seems to be complete.
Thank you very much for the quick fix.
regards Stefan

closed 2022-09-10 by s1738berger