stuartpittaway/diyBMSv4ESP32 #146: Victron functionality enhancements proposal

jnsaff, 2022-11-05:
This comes off the back of investigations for https://github.com/stuartpittaway/diyBMSv4Code/issues/126

Currently the logic of the Victron part of the code defines different actions for Balancing, Default and Error operations.

Ideally the logic that I could think of would something like this:

if `Module over temperature (internal) °C` or `Cell over temperature (external) °C`:
CCL: 0
DCL: 0
else if  `Individual cell over voltage (mV)` or `Current monitor over voltage (mV)` or `Pack over voltage (mV)` or `Module under temperature (internal) °C` or `Cell under temperature (external) °C`
then:
CCL: 0
DCL: Default
else if `Cell under voltage (mV)` or `Current monitor under voltage (mV)` or `Pack under voltage (mV)`
CCL: Default
DCL: 0
else:
CCL: Default
DCL: Default

There are other possibilities as well. Like adding DCL/CCL as a virtual relay state in the rules page with either binary or numeric values when triggered.

I have no C++ experience (tho I know Python so not completely hopeless) but I could take a stab at trying to code this up if this solution makes sense and would be likely accepted.

Thanks for the great work!

jnsaff, 2022-11-05:
Actually the logic above probably requires a bit more nuance, something similar to what is currently in there with `number_of_active_errors` which adds up the individual constraints that would prevent charging or discharging and if in the end the value for those constraints is not 0 then respective current limit would set to 0.

stuartpittaway, 2022-11-06:
Thanks for opening this issue.

If you've seen my YouTube videos, you may know that I'm in the middle of a LIFEPO4 battery build, so I've now got suitable equipment to talk to in regards to charging/discharging rules.

I'm thinking of building a new "charging rules" page in the web interface, so you can specify what to tell the charger (inverter) to do at certain temperatures, voltages, SoC% etc.

The idea, is that this is a set of rules to apply to the battery, before the existing rules - those are for more critical error scenarios (over temperature, voltage etc.)

Would this help?

N1c084, 2022-11-06:
Great until this a real secure feature , the best will be to be fully
compatible with pylontech protocol over can bus , because 90% of inverter
are compatible with this protocol .
Maybe replace the Victron integration with pylontech integration ?
I am impatient to discover and test this new feature
Thanks

jnsaff, 2022-11-08:
Is there open source documentation for the Pylontech protocol?

Also adding few notes for a stretch goal for the victron logic:
I have a 18s 280Ah LFP battery bank with the EVE type prismatic cells.

The voltage curve of the cells is really flat between 10 and 90% charge. This also means that a coulomb counter (likely a shunt) is really useful. I have the Victron SmartShunt, integration with that would be very useful. It talks to the Cerbo but not via CAN. All the data is available via MQTT, so this is one option of using the MQTT integration for listening to those events.

Also as the voltage drop-off is very steep at the edges then both charging and discharging at the edges is kinda rough. I have a 210A chaging capability and it can shoot through very quickly at the end. Bottom side is about the same, voltage drops below the floor for bigger consumption, discharge gets cut-off, then voltage recovers and there will be a saw-tooth pattern. It would be helpful to gradually reduce CCL and DCL at some points or follow a curve to make it a bit smoother experience to utilize the full capacity.

laserranger3000, 2023-01-23:
> Would this help?

Yes, a separate set of rules for "soft" limits would be perfect.
Currently I limit the charge current to 1A once balancing starts, but balancing starts and stops every few seconds, so the allowed charge current changes from 1A to max (230A) all the time.
I would prefer soft limits using the Victron CAN-bus integration for additional SoC or voltage dependent rules.

Also, I noticed that the actual usable capacity can change based on temperature, discharge current and so on. Therefore I would like to see a dynamic lower limit independent of the calculated SoC by the shunt. For example, if one cell voltage is below a given limit, the discharge current limit should drop even if the calculated SoC is still >0%.
Disconnecting the battery by relay at high currents should be avoided whenever possible.

stuartpittaway, 2023-02-14:
The latest version has just been released - this should significantly improve the Victron control, please give it a go when you get chance.

stuartpittaway, 2023-02-15:
Going to close this issue, please open a new one if any problems with latest code releases.

closed 2023-02-15 by stuartpittaway