stuartpittaway/diyBMSv4ESP32 #155: No temperature displayed

Chris87-ux, 2022-12-20:
Hi @stuartpittaway,
I just flashed the latest Software (09. of Dec 2022) on a controller and also flashed the modules with the 5k version of the software.
after connecting everything, no temperature had been displayed. 
The rule (controller Temp over 60°C) is active.

[image]
[image]

In case, future informations are needed, nust ask.
/ Chris

stuartpittaway, 2022-12-21:
Ok, somethings not correct there - the module temps are also -40 ?

Did you solder the ATTINY chip yourself?  This sounds like a problem with pin 9.

Also check the JST power cables you have - I can see bare wire/metal in the photo - you don't want that to short out!

Chris87-ux, 2022-12-21:
Hello, thanks for the fast reply. 
I have the same controller and module board ( same Order) running already since over 1 year, without problems and updates. so i dont suspect any hardware issue.
Yes, i soldered the ATTiny myself.
Here some additional pictures, as it has been quiet late yesterday:
[image]
[image]

Thanks regarding the hint with the bare wire, I'm on it, but it is quiet difficult to crimp 0,75mm² wires to this JST.
New boards to come ;-)
[image]

stuartpittaway, 2022-12-21:
Did you have any problems with R19?  That part should be CMFB103F3950FANT - did you substitute it when ordering from JLCPCB?

If you put a multimeter across it (resistance) the value should change when you put your finger on the part (heat/cold).

Chris87-ux, 2022-12-21:
At roomtemperature it has 7,71kOhm, after heating up a bit it has 7,42kOhm.
I din't changed anything at the R19, as mantioned olderones from the same Order works great.

For the programming I used the AVR programming mode from the Controller.

stuartpittaway, 2022-12-21:
Ok.

Can you check that when you broke off the temp sensor PCB board, that the two connections (small holes/vias) on the edge of that "snap" point are not shorted together?

Pins 5, 9 and 13 on the ATTINY chip control the temperature sensor reading - check the soldering around those parts.

Chris87-ux, 2022-12-21:
Ok, 
what i did now:
I had an other controller here, with the Software from 26.Feb. 2022.
I flashed one other board  with the 5k .hex over this controller.
Thank I changed to only one module on this "older" controller and i can see temperaturs again.

stuartpittaway, 2022-12-21:
If you click on the "modules" page and "configure" against a working module, can you give me the version number please?

[image]

Chris87-ux, 2022-12-21:
Now I reflashed one of the "boards were no temperature is shown" with the 5k.hex from the "older controller" and both temperatures are shown.

Than I connected the new controller again and al temperatures are shown.

stuartpittaway, 2022-12-21:
Right - can you provide the module version from one of the modules that DOESNT work ?

Chris87-ux, 2022-12-21:
now its getting interesting...
[image]
 both modules have the same verison.
ID 0= programmed with new controller software
ID 1= programmed with older controller software

stuartpittaway, 2022-12-21:
Ok, now that's weird!

I can't see source code version d2984b7b in GITHUB - was this one of the pre-compiled releases or did you compile yourself?

Chris87-ux, 2022-12-21:
It was the precomiled one.
[image]

stuartpittaway, 2022-12-21:
Ok, did you actually use that version 2022-12-09-15-10 ?

If so, the modules should have a version number of "3bc74c95"

Chris87-ux, 2022-12-21:
after a restart of the controller it showed this:
[image]
could this be some kind of caching issue?

stuartpittaway, 2022-12-21:
Yes, that data does get cached.

Let me go away and test a few scenarios out.

Chris87-ux, 2022-12-21:
Thanks very much!!! 
Short-cut would be now, programm modules with the Software from Feb. 2022 and connect controller Dec. 2022 ;-)

stuartpittaway, 2022-12-21:
Good news is that I've repeated the issue you see - bad news I've now got to fix it!

Chris87-ux, 2022-12-21:
;-) Good news, I'm not just stupid.

Thanks again for your great work!!!

Chris87-ux, 2022-12-21:
[image]
ID 0 is the new flashed board, everything great and working.

Somehow i had an huge issue getting the Board connected to wifi, but that was probably nothing important, as it works finally.

Fixed!!  Thanks very much for your Work.

stuartpittaway, 2022-12-21:
Great!

closed 2022-12-21 by stuartpittaway